ipxe/ipxe #643: Optional autoexec.ipxe

lookfirst, 2022-05-07:
Any way to make this request optional? I read the source code and it doesn't appear to be something that can be turned off.

I've embedded my ipxe bootscript into the ipxe.efi binary and I don't need to make another request to autoexec.ipxe. 

Part of why I'm asking is because we are battling UDP storms on our network and there is no reason for our TFTP server to be responding to this traffic, even if it is relatively small.

NiKiZe, 2022-05-07:
I had some similar concerns, @mcb30 I think that there must be a config way to disable autoexec. And do we really want it to happen with embedded scripts?

scott-xu, 2023-02-05:
@mcb30 any chance to make this feature optional/configurable?

mcb30, 2023-02-05:
It should be one UDP packet each way per system boot.  Is that really a significant overhead?

scott-xu, 2023-02-05:
It is not a a significant overhead yet. But there will be a lot of error logs in our TFTP service because the file is not found. We can update the TFTP service to not log an error if the requested autoexec.ipxe is not found. But before that, I just want to check if there's any plan to make this feature optional from iPXE :-)

NiKiZe, 2023-02-05:
Thinking out loud, Could it maybe be made with dhcp option 175.x that disables this feature?

mcb30, 2023-02-05:
No, I wasn't planning to make it optional.  For a start, it's not always under the control of the iPXE code.  If you build a combined BIOS/UEFI bootable USB drive using e.g.
```
./util/genfsimg ./util/genfsimg -o usbkey.img -s myscript.ipxe \
          bin-x86_64-efi/ipxe.efi bin/ipxe.lkrn
```
then `myscript.ipxe` will be placed in the root of the USB drive with the name `autoexec.ipxe`.  For a UEFI boot, iPXE loads the file itself via `EFI_SIMPLE_FILE_SYSTEM_PROTOCOL`. For a BIOS boot from the same USB drive, `autoexec.ipxe` gets passed in by syslinux as the initrd: the code in `ipxe.lkrn` never knows where the script came from.

The idea of the feature is to provide a consistent user experience of "I can control the boot process by modifying a text file called `autoexec.ipxe` that is saved in a logical location", as an improvement upon "I can control the boot process by recompiling iPXE to update an embedded script".

mcb30, 2023-02-05:
> Thinking out loud, Could it maybe be made with dhcp option 175.x that disables this feature?

The loading of `autoexec.ipxe` happens early on, before loading our drivers or setting up any of the network stack.  With a lot of extra code (especially in the BIOS case, where this would have to happen in `pxeprefix.S`), it could be made to respect an option 175.x in one of the cached DHCP packets.  But that generally wouldn't help anyway: the underlying PXE stack won't have requested option 175 so it won't be in the cached replies unless the server has been configured to force it upon the client.  There's a lot of variety in configuring DHCP servers and so this would in turn introduce extra complexity, which is what the `autoexec.ipxe` feature is trying to get away from.

NiKiZe, 2023-02-05:
> The loading of `autoexec.ipxe` happens early on, before loading our drivers or setting up any of the network stack. With a lot of extra code (especially in the BIOS case, where this would have to [...]

My thought is exactly this, if you don't want autoexec.ipxe request, then you need to force 175 in your dhcp server response (the cached part). Note that autoexec.ipxe should still default to enabled, and only be an option to disable the request in a known network. (advanced feature)

This feature is also not available in pcbios yet, so that currently also "extra complexity"

scott-xu, 2023-02-05:
Thanks for the info. It is always good to be as consistant as possible. In our case, we use TFTP but not USB drive to support both BIOS/UEFI. To be consistant, we have to embed the script inside undionly.kpxe and ipxe.efi because pcbios does not support autoexec. We may also add support for HttpBoot which, as far as know, does not support autoexec.ipxe yet.

mcb30, 2023-02-05:
> Thanks for the info. It is always good to be as consistant as possible. In our case, we use TFTP but not USB drive to support both BIOS/UEFI. To be consistant, we have to embed the script inside undionly.kpxe and ipxe.efi because pcbios does not support autoexec. We may also add support for HttpBoot in the future, which as far as know, does not support autoexec.ipxe yet.

So, it sounds as though your ideal situation might be to have `autoexec.ipxe` supported for BIOS PXE boots, and potentially also for UEFI HTTP boot?  If those existed, would you be able to stop using the embedded script altogether?

There are definitely still valid use cases for embedded scripts, but my expectation is that most common use cases can be handled more simply by using `autoexec.ipxe` - it would be good to get field confirmation of this.

scott-xu, 2023-02-05:
It would be great if BIOS PXE Boot and UEFI HTTP Boot support autoexec.ipxe. I do agree that's the ideal solution and no embedded script is needed.

mcb30, 2023-02-05:
> It would be great if BIOS PXE Boot and UEFI HTTP Boot support autoexec.ipxe. I do agree that's the ideal solution and no embedded script is needed.

Great, thanks for confirming!  :slightly_smiling_face: It's always good to have input from people using iPXE at scale.

Those features **will** get implemented at some undefined future point.  If EMC would like to fund a couple of days' work to get them implemented sooner, please contact me directly.

lookfirst, 2023-02-05:
> It should be one UDP packet each way per system boot. Is that really a significant overhead?

When you're scaling to 20,000+ machines, yes.

When I filed this issue, it turned out that once we hit around 10k machines, dnsmasq fell over. It was fine before that. Once we replaced that with Kea, things started working better again. Basically the UDP buffer on that server was filling up and couldn't be processed quickly enough.

It might not be that this is the actual issue, but when you're trying to debug things and coming up with solutions to things under fire, all you want is simpler and simpler systems.

This one UDP packet adds something that isn't simple. It creates more logs to comb through. I can't tell for certain if it contributed to the cause of dnsmasq falling over or something else though.

All I'm saying is that what might seem simple at first glance, isn't simple when you are operating at scale.

mcb30, 2023-02-05:
> > It should be one UDP packet each way per system boot. Is that really a significant overhead?
> 
> When you're scaling to 20,000+ machines, yes.

But to get to the point of receiving that one UDP packet, you must have already successfully transferred ~300 packets per machine to download e.g. `snponly.efi`.  So if these machines are all booting at the same time then you have already just demonstrated the capacity of your network to handle around 6,000,000 UDP packets in a rapid burst (or around 25,000,000 if you're using `ipxe.efi` instead of `snponly.efi`).  You've also demonstrated the capacity of your infrastructure to handle 20,000 concurrent stateless TFTP transfers and 20,000 concurrent **stateful** DHCP lease transactions.

Compared to this, the overhead of an additional 20,000 "file not found" single-packet stateless TFTP requests is trivial.

lookfirst, 2023-02-05:
You're focusing on the wrong part of my message.

mcb30, 2023-02-05:
> You're focusing on the wrong part of my message.

I'm closing this (a) because there are no plans to implement it (as described above), and (b) because life is too short to deal with people being condescending, sorry.

stappersg, 2023-02-05:
On Sun, Feb 05, 2023 at 12:02:44PM -0800, Public Profile wrote:
> You're focusing on the wrong part of my message.

Please transmit "Who can we hire?" as "Who can we hire?".

In case the message was "We want to hire you", then transmit that clearly^Wpolitely.

Groeten
Geert Stappers
-- 
Silence is hard to parse

lookfirst, 2023-02-05:
@mcb30 I'm not trying to be condescending and I'm sorry you took it that way. 

The point of my previous comment that when things are on fire, seeing even 1 hit to the TFTP server looking for a file that won't ever exist, is not trivial. It one more thing to try to debug.

If anything, at least [document](https://ipxe.org/embed) the fact that this happens so that someone else isn't scratching their head wondering what's going on.

habetsm-xilinx, 2023-02-06:
If you put 'exit 1' in autoexec.ipxe it should become a no-op.
1 to return with an error, so the next boot mechanism is tried by the BIOS or the UEFI stack.

closed 2023-05-07 by zube[bot]

scott-xu, 2023-05-26:
> > It would be great if BIOS PXE Boot and UEFI HTTP Boot support autoexec.ipxe. I do agree that's the ideal solution and no embedded script is needed.
> 
> Great, thanks for confirming! 🙂 It's always good to have input from people using iPXE at scale.
> 
> Those features **will** get implemented at some undefined future point. If EMC would like to fund a couple of days' work to get them implemented sooner, please contact me directly.

Hi Michael, did you receive my email?

tonho911, 2024-06-09:
> If you put 'exit 1' in autoexec.ipxe it should become a no-op. 1 to return with an error, so the next boot mechanism is tried by the BIOS or the UEFI stack.

where is the autoexec.ipxe file located?

systemmonkey42, 2024-08-05:
> where is the autoexec.ipxe file located?

I have multiple machines, some waste time looking for "autoexec.ipxe" and use the wrong IP (dhcp server IP  instead of next-server), while others wait 5 seconds thinking about it, and dont find anything.  Even a network trace shows no tftp or http probes.

I've embedded a 'bootfile' into the ipxe binary, so autoexec is effectively an "escape hatch" (read "security vulnerability") for anyone wanting to mess with the hardware and setup, especially since its so unclear where it actually loads the autoexec file from. (disk or usb maybe?? who knows)

Its so inconsistent, i've had to go to the IPXE source and comment out `efi_autoexec_load();` in `efi_probe()` which now saves
5 seconds per machine, and eliminates the inconsistencies.

The only difference I can find between the ones which get autoexec from the DHCP server, and the ones which hang for 5 seconds and find nothing is the PXEClient UNDI version indicates level 3.10, and 3.00 respectively.

--
PS.

I realise this is a dead topic due to, well personalities and the long term "competitive enshittification" of almost every product out there (I expect to find AI support in the next release guys! get to it! /s) but I just wanted to add my pointless 2c.

NiKiZe, 2024-08-05:
`autoexec.ipxe` should load from the same place as your iPXE binary, it is up to firmware/SNP/NII/UNDI to find it for you. (It's done before the iPXE network stack takes over)

If it loads from some other location, that might be a bug. But essentially, someone should only be able to inject autoexec, if they are able to modify your iPXE binary.

If you see this kind of behaviour, a pcap of loading iPXE binary, and trying to load `autoexec.ipxe` would probably be appreciated.

blackknight36, 2024-11-19:
> Its so inconsistent, i've had to go to the IPXE source and comment out `efi_autoexec_load();` in `efi_probe()` which now saves
> 5 seconds per machine, and eliminates the inconsistencies.

So here I am a year later running into the same problem.  Is there a way to fix this error besides disabling the feature in the source code?  I would rather not have to do this every time that we update ipxe.

NiKiZe, 2024-11-19:
> So here I am a year later running into the same problem. Is there a way to fix this error besides disabling the feature in the source code? I would rather not have to do this every time that we [...]

If you are talking about "File not found" showing, I'd argue that isn't a error, or problem, just information.
If you have something more precise in what you define as problem, do let us know.