ipxe/ipxe #645: The documented example of make debug=.... fails

sandynomad, 2022-05-11:
In the file `src/include/compiler.h` is an example build to include debugging:

    make bin/rtl8139.dsk DEBUG=pci,buffer,heap

This doesn't work against the current master (`70995397e5bdfd3431e12971aa40630c7014785f`)

    $ make bin/rtl8139.dsk DEBUG=pci,buffer,heap
    ....
    [BUILD] bin/pci.dbg1.o
    make: *** No rule to make target `bin/buffer.dbg1.o', needed by `bin/blib.a'.   Stop.

NiKiZe, 2022-05-11:
There is no buffer.c or no dbg statements in that file.

stappersg, 2022-05-12:
https://github.com/ipxe/ipxe/blob/master/src/include/compiler.h#L250 is
```text
 *   make bin/rtl8139.dsk DEBUG=pci,buffer,heap
```

@sandynomad for reporting,  @NiKiZe thanks for _There is no `buffer.c` or no `dbg` statements in that file_.

Screenshot
```text
stappers@myhost:~/src/ipxe
$ find . -name pci.c
./src/drivers/bus/pci.c
stappers@myhost:~/src/ipxe
$ find . -name buffer.c
stappers@myhost:~/src/ipxe
$ find . -name heap.c
stappers@myhost:~/src/ipxe
$ 
```

That says _There is a `src/drivers/bus/pci.c` file, but no `buffer.c` neither `heap.c`._

https://github.com/ipxe/ipxe/blob/master/src/include/compiler.h#L257-L264
```text
 * @code
 *
 *   make bin/rtl8139.dsk DEBUG=pci,buffer:2,heap
 *
 * @endcode
 *
 * which would enable debugging for the PCI, buffer-handling and
 * heap-allocation code, with the buffer-handling code at level 2.
 ```

That says _you should be able to get level 2 debug messages from buffer-handling code_.

But now we have this stale issue.

NiKiZe, 2022-07-04:
Thanks for reporting this.
But closing as the example is an example and it still requires some consideration to use DEBUG= statement.
By keeping the example non working it is easier to point out how the DEBUG option is working rather then trying to explain this much further down the road.

closed 2022-07-04 by NiKiZe