ipxe/ipxe #646: Getting IP via DHCP fails on 82599-vf over ovmf UEFI

IriKaQ, 2022-05-16:


NiKiZe, 2022-05-16:
Exactly which binary? Which version? And what message? (Did you follow the guidence from the link in the message?)
What does ifstat show after the failure?
(Use Ctrl-B and run dhcp followed by ifstat)

IriKaQ, 2022-05-16:
Sorry for the late update, the main binary information has been updated, thanks.

IriKaQ, 2022-05-16:
My other virtual machines (spoof checking is on) that pass through the pf NIC can be used in the system normally.

My VMs include Windows 10, Ubuntu 20.04, RockyOS 8.5 and etc.

NiKiZe, 2022-05-16:
In the scenario where it does not work and you boot full OS, does ping work? (Please rule out all factors except for iPXE)

IriKaQ, 2022-05-16:
> In the scenario where it does not work and you boot full OS, does ping work? (Please rule out all factors except for iPXE)

Yes, it works.

stappersg, 2022-05-16:
FWIW   Issue has been seen. Also seen _passthrough 82599-vf to the virtual machine_.

Not seen is why _iPXE team_ should allocate further resources to this issue.

Advise for @IriKaQ and all others who are bitten by the _undesired behaviour_, provide ways to reproduce this _"feature"_.  That in other words: Make **your** problem **our** challenge. (Yes, you are all invited to join the _iPXE team_.)

NiKiZe, 2022-05-16:
Ok then, provide the information that was requested in my previous message.

IriKaQ, 2022-05-17:
Something unexpected happened when I tried to reproduce and I can't reproduce it for now, maybe I need to do more experiments to confirm it.

NiKiZe, 2022-05-17:
Closing until the requested information is documented.

closed 2022-05-17 by NiKiZe