ipxe/ipxe #653: Additional PCI ID for intelxl (aka i40e)?

danmcd, 2022-05-25:
A customer of ours was trying to iPXE boot (using the TritonDataCenter downstream `ipxe`) on a machine equipped with X710s with the PCI ID of 8086,15ff, and the machine would reboot before iPXE handed off control to illumos.  There is no entry for that in the intelxl.c file.  We need to be sure, but adding the line (per https://github.com/TritonDataCenter/ipxe/pull/11) may be all we needed.

danmcd, 2022-05-25:
PR  #654

stappersg, 2022-05-25:
On Wed, May 25, 2022 at 11:41:50AM -0700, Dan McDonald wrote:
> PR  #654 

Acknowledge.

Consider to close this issue.

NiKiZe, 2022-05-25:
> Consider to close this issue.

No, keep it open and close when PR is merged (done automatically)

NiKiZe, 2022-06-30:
Did we get this verified as working?

danmcd, 2022-06-30:
It's now public in our downstream Triton ipxe that just hit our servers today (announcement tomorrow).  It's been passing regression-tests internally (with other i40e/intelxl NICs).

Our customer who requested this in the first place is running the affected hardware in production with an iPXE with my change in it.

So to answer your question, yes, our customer verified it.

danmcd, 2022-08-08:
Note, the recent PR (and our own repo) had a typo (1599 vs. 15ff) that mismatched what we sent our customer (15ff).

danmcd, 2022-08-08:
ANd we can close this since it appears to be fixed upstream.

danmcd, 2022-08-08:
Closing as duplicate since it was fixed more properly by someone else.

closed 2022-08-08 by danmcd