ipxe/ipxe #655: FIT image format to support U-Boot PXE

benthetechguy, 2022-05-27:
The way U-Boot does PXE is that it looks for pxelinux.cfg files on a TFTP server and interprets them in a way that mimics PXELINUX. This is similar to the widely used extlinux.conf features of U-Boot.
The main issue with this is the fact that there is no way to chainload an EFI binary (currently the only build target for arm) from PXELINUX, so the only way to run iPXE from U-Boot on ARM would be to have an snp.efi laying around on the local drive for U-Boot to boot from.
My proposed method to solve this would be to include a FIT image or uImage target that U-Boot can understand and chainload from its PXE setup.

NiKiZe, 2022-05-27:
U-boot can boot ipxes .efi, but only if the efi is stored on disk? Could efirom be used instead?

benthetechguy, 2022-05-31:
U-Boot cannot boot EFI from the network due to the fact that its netbooting architecture is build around SYSLINUX/PXELINUX which never gained the ability to chainload EFI binaries. All that it supports is FIT images or Linux kernels in uImage format.
There is [this](https://u-boot.readthedocs.io/en/latest/develop/uefi/uefi.html#launching-a-uefi-binary-from-a-fit-image) section of the documentation that may be helpful specifically dealing with FIT images and EFI binaries.

NiKiZe, 2022-05-31:
But u-boot can boot efi from disk, so the issue is the network boot part?
If so embed ipxes efirom into the u-boot ROM instead of pxelinux.

benthetechguy, 2022-05-31:
What I'm saying is there is no such thing as a U-Boot ROM. No place to stick an efirom with U-Boot. U-Boot does not have a PXE ROM or use PXE at all. Instead, it natively uses TFTP to look for pxelinux.cfg files on the network, and parses them in a way that emulates the original PXELINUX. Alternatively, you can manually look for and boot your own files from the network via the U-Boot command line, but they have to be in FIT/uImage format.

stappersg, 2022-05-31:
> Your most likely path forward is to find a way to wrap efi binaries with a FIT header?

Yes, or build a binary in FIT image format.  (Which is what this issue is about.)

To @benthetechguy and all others in the same boat:
* Allow yourself the warm feeling that the problem is understood
* Find ways to make a solution possible

Now the nasty thing: Make it possible that more people have the same problem.
Actually: Make more people stakeholder in finding a solution.
How?: Provide access to a development environment with this issue, make it an awarding experience.

benthetechguy, 2022-05-31:
> Replace that pxelinux emulator with ipxe emulator ;)
You're telling me to rewrite U-Boot's entire network stack. No thanks.
Thank you @stappersg for understanding what I was trying to say, I was worried there.

NiKiZe, 2022-06-01:
> > Replace that pxelinux emulator with ipxe emulator ;)
> 
> You're telling me to rewrite U-Boot's entire network stack. No thanks. 

No that is not what I said, simply because pxelinux does not really have a network stack. It has no ethernet drivers, no DHCP client, limited UDP support, and minimal tftp client.

Again think of these parts as modules, (which is what NIC ROMs are) and see if that can get you there.

My question still remains on if u-boot supports efi calls from this FIT image format?
If not you are not only looking at implementing a new image format, but actually implementing a new platform (efi alternative) I hope that isn't the case. But you need to collect the information. And probably also understand the details here...

And to go back a bit, is the current implementation a copy of pxelinux? Or is it a rewritten pxelinux "emulator"?

NiKiZe, 2022-06-01:
U-boot s "pxelinux" is a partial reimplementation, mustly for parsing cfg files, tftp support is already in u-boot without it.
FIT image format explanation https://www.thegoodpenguin.co.uk/blog/u-boot-fit-image-overview/

Page on efi https://u-boot.readthedocs.io/en/latest/develop/uefi/uefi.html

What purpose does the pxelinux emulator fill? You should already be able to fetch data with the u-boot tftp command to fetch ipxes snp.efi and then run that with the indicated commands?

NiKiZe, 2022-06-01:
Can you modify u-boot, or are you thinking of u-boot as unchangeable? When I say modules, think of Linux modules, or Windows drivers. You are missunderstanding how NIC ROMs work and confusing it with hardware.

If you can wrap snp.efi in a FIT image, then do that (I did not see any earlier "yes" to that question) I don't see what iPXE needs to do here if you can generate the FIT image yourself?

closed 2026-03-04 by NiKiZe