ipxe/ipxe #656: No keyboard input on iPXE loaded from U-Boot

benthetechguy, 2022-05-31:
I've tried different combinations of booting methods on many different ARM boards and always get the same result. If I boot iPXE from U-Boot via the `bootefi` command, my keyboard doesn't work in iPXE (both USB keyboards and the built-in one of a laptop).
I booted the EFI shell binary via U-Boot and the keyboard worked there, so U-Boot isn't broken. However, booting iPXE from the Raspberry Pi 4 port of EDKII I could use my keyboard.
This means that the issue is specifically a combination of U-Boot and iPXE, but I can't seem to tell which side the issue lies on, if not both.

NiKiZe, 2022-05-31:
Does #605 explain and resolve this issue?

benthetechguy, 2022-05-31:
Yes, thank you for bringing that to my attention.

closed 2022-08-30 by zube[bot]