ipxe/ipxe #658: iPXE fails to compile on ARM with CONSOLE_SERIAL enabled

benthetechguy, 2022-06-01:
When I try to compile iPXE for arm64-efi or arm32-efi, it fails right at the very end:
```
aarch64-linux-gnu-ar: creating bin-arm64-efi/blib.a
  [LD] bin-arm64-efi/ipxe.efi.tmp
aarch64-linux-gnu-ld: bin-arm64-efi/blib.a(serial.o): in function `serial_init':
/home/ben/ipxe/src/core/serial.c:145: undefined reference to `uart_select'
aarch64-linux-gnu-ld: bin-arm64-efi/blib.a(serial.o): in function `uart_data_ready':
/home/ben/ipxe/src/include/ipxe/uart.h:112: undefined reference to `uart_read'
aarch64-linux-gnu-ld: /home/ben/ipxe/src/include/ipxe/uart.h:112: undefined reference to `uart_read'
aarch64-linux-gnu-ld: bin-arm64-efi/blib.a(serial.o): in function `uart_receive':
/home/ben/ipxe/src/include/ipxe/uart.h:124: undefined reference to `uart_read'
aarch64-linux-gnu-ld: bin-arm64-efi/blib.a(uart.o): in function `uart_transmit':
/home/ben/ipxe/src/core/uart.c:54: undefined reference to `uart_read'
aarch64-linux-gnu-ld: /home/ben/ipxe/src/core/uart.c:61: undefined reference to `uart_write'
aarch64-linux-gnu-ld: bin-arm64-efi/blib.a(uart.o): in function `uart_flush':
/home/ben/ipxe/src/core/uart.c:75: undefined reference to `uart_read'
aarch64-linux-gnu-ld: /home/ben/ipxe/src/core/uart.c:76: undefined reference to `uart_read'
aarch64-linux-gnu-ld: bin-arm64-efi/blib.a(uart.o): in function `uart_exists':
/home/ben/ipxe/src/core/uart.c:95: undefined reference to `uart_write'
aarch64-linux-gnu-ld: /home/ben/ipxe/src/core/uart.c:96: undefined reference to `uart_read'
aarch64-linux-gnu-ld: /home/ben/ipxe/src/core/uart.c:98: undefined reference to `uart_write'
aarch64-linux-gnu-ld: /home/ben/ipxe/src/core/uart.c:99: undefined reference to `uart_read'
aarch64-linux-gnu-ld: bin-arm64-efi/blib.a(uart.o): in function `uart_init':
/home/ben/ipxe/src/core/uart.c:124: undefined reference to `uart_read'
aarch64-linux-gnu-ld: /home/ben/ipxe/src/core/uart.c:126: undefined reference to `uart_write'
aarch64-linux-gnu-ld: /home/ben/ipxe/src/core/uart.c:131: undefined reference to `uart_write'
aarch64-linux-gnu-ld: /home/ben/ipxe/src/core/uart.c:132: undefined reference to `uart_write'
aarch64-linux-gnu-ld: /home/ben/ipxe/src/core/uart.c:138: undefined reference to `uart_write'
aarch64-linux-gnu-ld: /home/ben/ipxe/src/core/uart.c:141: undefined reference to `uart_write'
aarch64-linux-gnu-ld: bin-arm64-efi/blib.a(uart.o):/home/ben/ipxe/src/core/uart.c:144: more undefined references to `uart_write' follow
aarch64-linux-gnu-ld: bin-arm64-efi/blib.a(uart.o): in function `uart_init':
/home/ben/ipxe/src/core/uart.c:134: undefined reference to `uart_read'
aarch64-linux-gnu-ld: /home/ben/ipxe/src/core/uart.c:135: undefined reference to `uart_read'
aarch64-linux-gnu-ld: bin-arm64-efi/ipxe.efi.tmp: hidden symbol `uart_read' isn't defined
aarch64-linux-gnu-ld: final link failed: bad value
make: *** [Makefile.housekeeping:1220: bin-arm64-efi/ipxe.efi.tmp] Error 1
rm bin-arm64-efi/version.ipxe.efi.o
```
If I undefine CONSOLE_SERIAL, this doesn't happen. I want CONSOLE_SERIAL though, because UART output is very important on ARM and other "embedded" platforms, since it's not uncommon to have no display.

9001, 2022-06-01:
Looks like `uart_select`, `uart_read` and `uart_write` is only defined in [/arch/x86/core/x86_uart.c](https://github.com/ipxe/ipxe/blob/master/src/arch/x86/core/x86_uart.c) and 
[/arch/x86/include/bits/uart.h](https://github.com/ipxe/ipxe/blob/master/src/arch/x86/include/bits/uart.h), so guess nobody took a stab at adding arm support yet -- [/arch/arm/include/bits/uart.h](https://github.com/ipxe/ipxe/blob/master/src/arch/arm/include/bits/uart.h) is pretty stubby...

NiKiZe, 2022-06-01:
Closing as duplicate of #165 on the basis that this is up to efi firmware to handle.

closed 2022-06-01 by NiKiZe