ipxe/ipxe #659: compile error (Unrecognised relocation type 9) on latest and V1.21.1, but v1.20.1 works

hsteininger, 2022-06-07:
Hi,

today i tried to recompile iPXE after `git pull` to the latest version, but it failed with:

```
  [AR] bin-x86_64-efi/blib.a
ar: creating bin-x86_64-efi/blib.a
  [VERSION] bin-x86_64-efi/version.ipxe.efi.o
  [LD] bin-x86_64-efi/ipxe.efi.tmp
  [HOSTCC] util/elf2efi64
  [FINISH] bin-x86_64-efi/ipxe.efi
Unrecognised relocation type 9
Makefile.efi:33: recipe for target 'bin-x86_64-efi/ipxe.efi' failed
make: *** [bin-x86_64-efi/ipxe.efi] Error 1
rm bin-x86_64-efi/version.ipxe.efi.o
```

checking out `v1.21.1` and compile again, brings up the same error.

when checking out `v1.20.1` everything is building fine.

after that i tried `git bisect` to find the commit and found this one:

```
someuser@somehost:~/ipxe$ git bisect good
a61b27b97f572a83ede765a0e779694865950cf2 is the first bad commit
commit a61b27b97f572a83ede765a0e779694865950cf2
Author: Michael Brown <mcb30@ipxe.org>
Date:   Tue Jun 23 23:08:49 2020 +0100

    [efi] Enable stack protection where possible

    Enable -fstack-protector for EFI builds, where binary size is less
    critical than for BIOS builds.

    The stack cookie must be constructed immediately on entry, which
    prohibits the use of any viable entropy source.  Construct a cookie by
    XORing together various mildly random quantities to produce a value
    that will at least not be identical on each run.

    On detecting a stack corruption, attempt to call Exit() with an
    appropriate error.  If that fails, then lock up the machine since
    there is no other safe action that can be taken.

    The old conditional check for support of -fno-stack-protector is
    omitted since this flag dates back to GCC 4.1.

    Signed-off-by: Michael Brown <mcb30@ipxe.org>

:040000 040000 beb0f840b91ef33263c34a958f6f550e85868241 a424a724a5bdb3ce4d0c3dddb8040811e225bcfb M      src
```

Is there a way to fix that?
If you need further information, just get back to me.

TiA,
Herbert

NiKiZe, 2022-06-07:
That seems to have been working for a long time.
Which version of GCC are you using?

hsteininger, 2022-06-08:
Well, i'm sorry, after checking the build env i found out it was really outdated.

` 
gcc --version
gcc (Debian 4.9.2-10+deb8u2) 4.9.2
`

spinning up a new almalinux with newest gcc finished building ipxe without errors.

Sorry again.

closed 2022-06-08 by hsteininger