ipxe/ipxe #663: Console blanks out after booting iPXE

MischaBaars, 2022-06-25:
Hello,

I want to boot a cluster of network nodes using iPXE, because I then have to install and configure the operating system only once.

The DHCP server from the router had to be disabled and a static ip address assigned to the server, but I did get dnsmasq up and running. With an external tftp server because using the internal server downloads are systematically aborted for some reason.

The 'ipxe.iso' that can be downloaded from ipxe.org has a filesize of 4194304 bytes. The compiled version is only 1048576 bytes. When booting the first, I do get a screen and a command prompt. The second image does not boot up the system. The partitioning structure is also different from the downloaded version. The same holds for 'ipxe.usb'.

The 'undionly.kpxe' that can be downloaded from ipxe.org, has a filesize of 71717 bytes. The compiled version is only 68137 bytes. The client correctly downloads and executes both. The inifinite loop is visible from the server 'journalctl --boot --follow', but the first thing that happens is that the client's screen blanks out, no welcome message, nothing, for both the downloaded and the compiled version.

I've tried compiling with 'make' and with 'make everything'.

Any idea what might be the problem? I think that the CONSOLE_PCBIOS code is only compiled with 'make everyting', but I doubt that is the problem.

Best regards,
Mischa Baars.

stappersg, 2022-06-25:
> Any idea what might be the problem?

[Yes](http://www.catb.org/~esr/faqs/smart-questions.html)

MischaBaars, 2022-06-25:
And would you be willing to share your solution to this problem with me?

stappersg, 2022-06-25:
Depends on the fee.

MischaBaars, 2022-06-25:
I recommend putting your money on Netflix shares?!

stappersg, 2022-06-25:
_OK_

stappersg, 2022-06-25:
Welcome to this part of the _iPXE community_.
* We all started as beginners
* We made progress
* We are at what are at

For getting us further there _should be_ a **we**.

Advice for being part of _we_:
* Read and understand http://www.catb.org/~esr/faqs/smart-questions.html
* Ask open questions ( Avoid close ones like _Any idea what might be the problem?_ and _And would you be willing to share your solution to this problem with me?_)
* Take full ownership of your problem
* When sharing _"challenges"_, share also the _benefits_ for completing the challenge

NiKiZe, 2022-06-25:
> Welcome to this part of the _iPXE community_.
>
> * We all started as beginners
> * We made progress
> * We are at what are at
>
> For getting us further there _should be_ a **we**.
>
> Advice for being part of [...]

Stopp this bs! Either be helpful, or don't reply at all.

NiKiZe, 2022-06-25:
> Hello,
>
> I want to boot a cluster of network nodes using iPXE, because I then have to install and configure the operating system only once.
>
> The DHCP server from the router had to be disabled and a [...]

Depending on compiler or other options the filesizes will differ.
How are you actually trying to boot these machines, and is it in pcbios mode or efi? I assume pcbios? But on what kind of cluster/hardware/emulator is this?

MischaBaars, 2022-06-25:
Geert,

Als jij je reet graag afveegt met deze vissoort, heb je meer weg van een wijf dan ik, denk je niet?

De stekels mag je houden!

Mischa.

MischaBaars, 2022-06-25:
NiKiZe,

I'm booting them in Legacy mode. If I set Option ROM Execution -> Network to UEFI, the Realtek PXE doesn't show up as an Option under Boot Option Priorities. So... Legacy mode indeed.

I'm currently booting a maximum of 4 Beelink MII-V's, no emulator. They all run a separately installed Fedora Workstation or Fedora Server. This will change in the future to a minimum of 16 network nodes. That's why I'm trying PXE network booting, to keep the number of operating systems to install and configure under control.

What compiler options are you talking about? The README only refers to 'make' and 'make everything'. The default Makefile is the makefile compiling the Legacy target.

Best regards,
Mischa.

NiKiZe, 2022-06-25:
Ipxe.org has the documentation, but you don't have to care so much about that, undionly.kpxe should just work.

You could try to boot in qemu to get a working sample to compare to.

MischaBaars, 2022-06-26:
Hi Christian,

Putting it all back into the box? :) I might be trying that later on.

I found a solution to booting ipxe.iso:

With the default makefile, only 'bin/ipxe.lkrn' but not 'bin-x86_64-efi/ipxe.efi' are compiled when using 'make', and 'bin/ipxe.iso' will contain only 'bin/ipxe.lkrn'. This is what the README suggests doing.
With the default makefile, both 'bin/ipxe.lkrn' and 'bin-x86_64-efi/ipxe.efi' and compiled when using 'make everthing', and 'bin/ipxe.iso' will still contain only 'bin/ipxe.lkrn'.

When selecting either Legacy / UEFI as BIOS Option ROM Execution -> Storage, both USBs won't show up under BIOS Boot Options Priorities. That's why the screen blanked.

After manually generating an ISO with either: './util/genfsimg -o ipxe.iso bin/ipxe.lkrn bin-x86_64-efi/ipxe.efi' or './util/genfsimg -o ipxe.iso bin-x86_64-efi/ipxe.efi', the USB will show up and boot regardless of the BIOS Option ROM Execution -> Storage setting. I don't know what 'bin/ipxe.lkrn' is for, but it does not in any way contribute to booting up your system.

Cheers! That's half the story, now for the 'undionly.kpxe' part.

Mischa.

NiKiZe, 2022-06-26:
Well that is for building ipxe.efi with both boot paths, but using the iso in your scenario seems just wrong.

Instead of blindly guessing, have a look at the documentation? And as I said compare to a known working setup. Using qemu to know what works, and then transfer that to your VM.

MischaBaars, 2022-06-26:
Wrong? According to the documentation it is the best way to start: "The easiest way to start experimenting with iPXE is to use the [bootable ISO image](http://boot.ipxe.org/ipxe.iso).". Now I know it boots, has a working console, and I can chainload a helloworld.ipxe that outputs "Hello World!". What's so wrong about that?

When booting from network device instead of USB, helloworld.ipxe does execute, as can be seen from the server's 'journalctl --boot --follow', but I do not have the working console. That's the only difference between to two.

CONSOLE_PCBIOS should be enabled by default, AND, the console is working when booting from USB. Any ideas?

MischaBaars, 2022-06-26:
Hi Christian,

It works!!! The BIOS Option ROM Execution -> Video had to be put into Legacy Mode. Then the iPXE prompt appears on the second monitor, the one that is switched off by default :)

Thank you for you help!

Best regards,
Mischa Baars.

stappersg, 2022-06-26:
> It works

Thanks for reporting back.

NiKiZe, 2022-06-26:
Console_pcbios is enabled by default in pcbios builds.

This is not an issue with ipxe but rather with the bios implementation of you emulator. It works as expected on real hardware, and most emulators such as vbox an qemu.

closed 2022-06-26 by NiKiZe

stappersg, 2022-06-26:
As I understand it, on hindsight, _two video cards, one monitor connected, unexpected switch to other video card_.