ipxe/ipxe #664: VMware Workstation Triple CPU Fault

rumline, 2022-06-25:
Attempting to PXE boot either ipxe.efi or snponly.efi using a VMware Workstation 16.2.3 VM on Windows 10 Pro 21H2 build 19044.1766 with an Intel Core i7-6700k CPU causes the VM to crash with what is documented from VMware a CPU triple fault (https://kb.vmware.com/s/article/2058114).  Tried with both Intel VT-x/EPT and virtualzed IOMMU enabled and disabled.

Attempted with both a single vCPU and dual vCPU's with the same result.

Tried with compiling ipxe.efi and snponly.efi on a Rocky 8 VM running in this environment with the same result.

Please advise.

stappersg, 2022-06-25:
When does the crash happen?

Long version of the very same question again:
Which parts of the _network boot process_ do work and does it stop?

stappersg, 2022-06-25:
When does the crash happen?

Long version of the very same question again:
Which parts of the _network boot process_ do work and when does it stop?

NiKiZe, 2022-06-25:
Did any of the attempts to boot include using prebuilt binaries from boot.ipxe.org? (Some compilers might generate broken binaries)

You might have to dig further and find if this is an issue with specific version of VMware, or specific version of ipxe, or the compiler used. Or maybe even the host?
At this stage it sounds like anything could be the cause.

rumline, 2022-06-26:
Happens with prebuilt binaries and custom build binaries.  The crash happens after the initial "Initialising devices" step where a DHCP IP address is requested but at the "2nd initialising hardware" step where it would be "mounting" the tftp server.

More clarity:  I'm using a WDS server.  If you replace the wdsmgfw.efi file in WDS with your own efi boot loader it will begin the PXE process.  The replaced efi bootloader will perform the DHCP process then use DHDP boot option 67 to load the specified bootloader.  So using either "ipxe.efi" or "snponly.efi" for the replaced "wdsmgfw.efi" boot loader to get an IP address works fine.  But then ipxe.fi or snponly.efi is specified in DHCP option 67 and starts to actually load it crashes the VM at "initalising hardware">

stappersg, 2022-06-26:
That is odd.

Thing I'm willing to review are two  `.pcap` files.  One for the _`ipxe.efi` or `snponly.efi` for the replaced `wdsmgfw.efi`_ and the other one _when `ipxe.efi` or `snponly.efi` is specified in DHCP option 67_.  And please tell how the `.pcap`s were created. For that has Original Poster to translate
* Configured [port mirroring](https://en.wikipedia.org/wiki/Port_mirroring) for seeing the _netboot client_
* `tcpdump -w name.pcap ether host MA:Ca:dd:re:ss`

to what is at hand.

NiKiZe, 2022-06-26:
Have you tried using the VMware specific targets? But also do report this to VMware, a bug in their firmware is not unlikely.

NiKiZe, 2022-07-04:
@rumline did you have time to try the vmware specific target ?

Hammarskjold, 2023-06-08:
This is related to #883

stappersg, 2023-07-10:
> This is related to #883

And #883 is closed by #968 about a month ago.

Highlighting @rumline for seeing this feedback request.

stappersg, 2024-08-05:
@rumline a year has past.  Please close this issue, you as creator of the issue have that privilege. Closing this issue will prevent that further human energy is lost on this neglected issue.

closed 2024-11-04 by zube[bot]