ipxe/ipxe #665: IMAGE xxx.img format not recognized

jaepetto, 2022-06-28:
One of our researchers is currently building a custom kernel. In order to do that, I configured one of our machines to boot on ipxe with his image.

However, he keeps on getting the following error messages:
```
IMAGE xxx.img at [67ad5000,67b80600) registered
IMAGE xxx.img is not Multiboot: Exec format error...
IMAGE xxx.img is not ELF...
IMAGE xxx.img is not script...
IMAGE xxx.img is not bzImage...
IMAGE xxx.img is not PXE-NBP...
IMAGE xxx.img is not PXE-NBP (may be EFI?)...
Could not select: Exec format not recognized
```
Aside from the documentation here (https://ipxe.org/err/2e0080), what advice could I give him to build his image?

Thanks in advance,

Emmanuel

NiKiZe, 2022-06-28:
We can assume you are running under pcbios, but your image is not a valid binary. How do you run this outside of iPXE? Using file to check this can be helpful to explain what is going on.

One last suggestion, is this maybe an disk image? Then maybe sanboot could start this, however it is not likely to work all the way

NiKiZe, 2022-06-30:
More info still needed, if you have already resolved this please inform us about it and close the issue. Thanks!

jaepetto, 2022-06-30:
@mcb30 @NiKiZe Sorry guys for not getting back to you yet. I absolutely agree that you need a lot more information. Unfortunately, I could not get a hold of our researcher yet. I'll let you know as soon as I'll get something. In the meantime, please be assured that I really appreciate your help.

jaepetto, 2022-07-04:
@mcb30 @NiKiZe Our researcher finally managed to get his kernel to work. He actually completely bypassed his problem by loading his kernel instead of ipxe. Sorry for wasting your time.

closed 2022-10-02 by zube[bot]