ipxe/ipxe #669: HTTPS downloads fail on VMWare Fusion Player Virtual Machines

GeorgLippold, 2022-07-02:
Hi, thanks for making IPXE such a great software.

I am trying to use IPXE on a VM to install distributions from netboot.xyz.
When trying to use the HTTPS protocol, I always get the IPXE error message https://ipxe.org/err/1c25e0, see e.g. [image]. For reproducing, it seems to be enough to drop to the IPXE shell, then use

dhcp
imgfetch https://google.com

After I activated debugging, I got the following message during bootup: 
[image]

It seems that there is not enough entropy at startup for HTTPS connections. Is it possible to provide a patch that allows booting on VMWare despite the lack of entropy?

Thanks!

NiKiZe, 2022-07-02:
Please read that error URL, I do think it covers your scenario with proposal for fix as well. That is https://ipxe.org/err/1c25e0

closed 2022-07-02 by NiKiZe

GeorgLippold, 2022-07-02:
Thanks, disabling the Side Channel Mitigations under the advanced settings did the trick.