ipxe/ipxe #676: Intermittent Errors connecting via tls 0x420c6001

brokenjacobs, 2022-07-12:
I have a ipxe script that chains to a TLS endpoint that works intermittently. On high latency connections it fails every time. On lower latency connections it works inconsistently. It seems to occur either while retrieving the cross-cert from ipxe.org or during an OCSP validity check. I can't see any timers to adjust for these settings, and I'm debating removing TLS from this endpoint so I can get consistent ipxe boots at remote sites. Any ideas? The error page isn't particularly helpful.
Error 0x420c6001 (http://ipxe.org/420c6001)

The line in question shows that this is an http_close, but I believe it's due to a timeout of some sort in the http client.
https://github.com/ipxe/ipxe/blob/d3c8944d5c3ab262826ed33b16f3fc4dd43bc304/src/net/tcp/httpcore.c#L300

NiKiZe, 2022-07-12:
Compile with proper debug flags and grab a trace, first step is on that page.

brokenjacobs, 2022-07-12:
At this point I've just reverted to using http. The external OCSP dependency has bothered me enough to just not bother with this anymore. If I get some time I'll do a debug build, but as of right now we can't have the service listen on 443 and 80 at the same time so it's not easy to test.

NiKiZe, 2022-07-12:
Feel free to add more data and we can reopen if this shows to be something we can actually fix. But withou more data, there is nothing to show this being an issue with ipxe code.

closed 2022-07-12 by NiKiZe