ipxe/ipxe #677: Multipe Network I/Fs and DCHP

joakim-tjernlund, 2022-07-14:
When more that one I/F, ipxe probes them in sequence so one must wait for ipxe to get around to later I/Fs
Maybe ipxe can do what Linux ipconfig does, probe all I/F in parallel and use the first one that gets a link?

mcb30, 2022-07-14:
Another issue is that iPXE runs with very limited memory available for I/O buffers, and opening all network cards at once is likely to lead to memory exhaustion and large numbers of dropped packets.

joakim-tjernlund, 2022-07-14:
> Another issue is that iPXE runs with very limited memory available for I/O buffers, and opening all network cards at once is likely to lead to memory exhaustion and large numbers of dropped packets.

How so? ipxe would only poll all I/Fs for link up, when it finds one ipxe stops polling the other ones and continue with the with link up.

stappersg, 2022-08-09:
@joakim-tjernlund what about proper spelling of DHCP?

[image]

mcb30, 2022-08-11:
> How so? ipxe would only poll all I/Fs for link up, when it finds one ipxe stops polling the other ones and continue with the with link up.

There is no concept within iPXE of opening an interface without providing it with receive data buffers, and no way of polling for link state without opening the interface.

It could potentially be done, but it would have to be a substantial benefit to justify the amount of work.

Marking as closed on the basis that this is unlikely to be carried out.

closed 2022-08-11 by mcb30