ipxe/ipxe #682: Apparently, the NBI target still exists in Makefile, but it's not documented on the wiki. No need for mknbi. Nice. Let's try.

stappersg, 2022-07-18:
From #681

> Apparently, the NBI target still exists in Makefile, but it's not documented on the wiki. No need for mknbi.

Now there is this _wishlist issue_ for documenting the _NBI target_.

perkelix, 2022-07-18:
I stumbled upon this while investigating ways to generate NBI blobs to chainload onto an old Etherboot host. I noticed that gPXE had an NBI target e.g. 'make bin/natsemi.nbi". Tried the same on the iPXE Git and, lo and behold, it's still there.

NiKiZe, 2022-07-18:
nbi, pcirom, isarom etc is specific for the boot architecture, aka nbi is not rom so there is nothing to document in that regard.

We do not have to document nbi at all, since almost no one uses it. And now it is also documented in the issues if anyone searches for it. As such closing as won't fix.

perkelix, 2022-07-18:
What is the NBI target then, if not a ROM blob with an NBI header?

stappersg, 2022-07-18:
> What is the NBI target then, if not a ROM blob with an NBI header?

@perkelix feel free to email  stappers AT stappers DOT it

Do know that this comment will be marked as deleted.

stappersg, 2022-07-18:
Purposed text for https://ipxe.org/appnote/buildtargets#special_targets
Comma Separated Values
```text
.nbi,etherboot,Net Boot Image as used by etherboot
```

perkelix, 2022-07-19:
Btw, the buldtarget list has a mistake:

.pxe | pcbios | Headerless X86 assembly code, PXE- or NBP-booted, sometimes renamed to .0 to work on older DHCP/TFTP servers

Yet if someone throws that at an Etherboot device, one gets this error:

PXE -> EB: No PXE stack found!

Wasn't the point of this binary to _give_ PXE capability to the host?

NiKiZe, 2022-07-19:
> Btw, the buldtarget list has a mistake:
>
> `.pxe | pcbios | Headerless X86 assembly code, PXE- or NBP-booted, sometimes renamed to .0 to work on older DHCP/TFTP servers`
>
> Yet if someone throws that at [...]

That makes total sense if EB does not provide an PXE stack to work with.
All pcbios builds of iPXE can provide a PXE stack (compile time config) the .pxe target consumes a PXE stack to do it's initialization. (It can for example use a cached DHCP response)

perkelix, 2022-07-19:
> That makes total sense if EB does not provide an PXE stack to work with. All pcbios builds of iPXE can provide a PXE stack (compile time config) the .pxe target consumes a PXE stack to do it's [...]

This old host's EB indeed doesn't include PXE. That's why I need to chainload PXE capabilities.
Where is enabling the PXE stack via config options documented?

perkelix, 2022-07-19:
> Oh, I forgot to explain why having
> 
> ```
> .nbi | etherboot | Net Boot Image as used by etherboot
> ```
> 
> at https://ipxe.org/appnote/buildtargets#special_targets is a **good thing**.
> 
> It does document that the _iPXE toolchain_ can generate `.nbi` images. So we prevent that a community member wastes time with an additional `mknbi` step. With `ln -s ~/Projects/ipxe/src/bin/ipxe.nbi /srv/tftp/ipxe.nbi` can even a copy command be skipped to speed up their development cycle.

Adding this line to the table of available targets and pointing out that 'mknbi' probably won't be needed to prepare NBI blobs for netboot implementations that predate PXE would indeed be welcome.

stappersg, 2022-07-19:
On Tue, Jul 19, 2022 at 03:04:21AM -0700, perkelix wrote:
> 
> This old host's EB indeed doesn't include PXE. That's why I need to chainload PXE capabilities.
> Where is enabling the PXE stack via config options documented?
> 

Please keep this (closed) issue clean.  Keep this issue about
documententing that iPXE can provide `.nbi` images.

stappersg, 2022-07-19:
> Adding this line to the table of available targets and pointing out that 'mknbi' probably won't be needed to prepare NBI blobs for netboot implementations that predate PXE would indeed be welcome.

New proposal:
```text
.nbi | etherboot | Net Boot Image as used by etherboot, so no need to run mknbi
```

NiKiZe, 2022-07-19:
> This old host's EB indeed doesn't include PXE. That's why I need to chainload PXE capabilities. Where is enabling the PXE stack via config options documented?

Since it is enabled by default in every build of iPXE for pcbios there is not much need for that IMO, the only ones that finds it is those that need to minimize the size of iPXE and go looking in the config files ;)

closed 2022-10-17 by zube[bot]

perkelix, 2022-10-17:
> Since it is enabled by default in every build of iPXE for pcbios there is not much need for that IMO, the only ones that finds it is those that need to minimize the size of iPXE and go looking in the config files ;)

Which obviously fails, since your previous reply suggests that the blob needs the BIOS to provide a PXE stack to piggyback onto.

NiKiZe, 2022-10-17:
> Which obviously fails, since your previous reply suggests that the blob needs the BIOS to provide a PXE stack to piggyback onto.

PXE is only available in PCBIOS builds. You can still have a PCBIOS build without PXE stack support.