ipxe/ipxe #694: Last line of PXE menu repeats when further pressing DownArrow

perkelix, 2022-07-28:
In dnsmasq, I have the following PXE menu:

```
pxe-prompt="PXE boot options:"
pxe-service=x86PC,"Debian 11 installer (x86-64) Linux 5.10",/amd64/pxelinux.0
pxe-service=x86PC,"Debian 11 installer (x86-32) Linux 5.10",/i686/pxelinux.0
pxe-service=x86PC,"Debian  8 installer (x86-32) Linux 3.16",/i586/pxelinux.0
pxe-service=x86PC,"System Memory Tester (X86PC) Memtest86+",/memdisk/lpxelinux.0
```

Whenever I keep on pressing DownArrow, the last line repeats.

This issue doesn't happen with gPXE or with the NIC's built-in commercial PXE implementation.

NiKiZe, 2022-07-28:
The (mostly undocumented) PXE menu is not really something iPXE cares about.
But could you narrow it down further using git bisect https://ipxe.org/howto/bisect ?

stappersg, 2022-08-26:
This update is for for asking for further information about this issue.

[image]

Please state explicite that the _narrow down_ did happen (make clear wether or out the issue waits on outcome of `git bisect`).

[image]

Please tell more about _"last line fails to display"_.  Make explicite what you mean, like:
* Menu has four entries, but only first tree entries are shown
* The menu is at the bottom of my screen, no scroll up does happen

In
> last line fails to display. Whenever I keep on pressing DownArrow, the last line appears AND repeats.

do I read a problem (_last line fails to display_) and a solution (_pressing DownArrow_).
However I don't read what the expected outcome of this issue should be, so please elaborate desired goals.

mcb30, 2022-10-27:
@perkelix As suggested, could you try using https://ipxe.org/howto/bisect to identify the precise commit that introduced the issue?

RobMahn, 2023-01-07:
I found this researching the same problem and thought I would attempt the bisect.  Cloning V1.0.0 failed to compile, as did most of the bisect points.  I can compile the current version.

[root@Desktop src]# cd ~/ipxe/src ; make bin/undionly.kpxe
  [BUILD] bin/cwuri.o
In file included from core/cwuri.c:22:
include/gpxe/uri.h: In function ‘uri_get’:
include/gpxe/uri.h:152:12: error: taking address of packed member of ‘struct ur
’ may result in an unaligned pointer value [-Werror=address-of-packed-member]
  152 |  ref_get ( &uri->refcnt );
      |            ^~~~~~~~~~~~
include/gpxe/uri.h: In function ‘uri_put’:
include/gpxe/uri.h:163:12: error: taking address of packed member of ‘struct ur
’ may result in an unaligned pointer value [-Werror=address-of-packed-member]
  163 |  ref_put ( &uri->refcnt );
      |            ^~~~~~~~~~~~
cc1: all warnings being treated as errors
make: *** [bin/deps/core/cwuri.c.d:17: bin/cwuri.o] Error 1

mcb30, 2023-01-07:
@RobMahn Newer compilers tend to be more strict with warnings, meaning that old versions of the codebase will often trigger warnings that have since been fixed. You can work around this for bisection by building as `make bin/undionly.kpxe NO_WERROR=1`

stappersg, 2023-01-07:
> > attempt the bisect. Cloning V1.0.0 failed to compile, as did most of the bisect points. I can compile the current version.

> Newer compilers tend to be more strict with warnings

#620 is _"fixed"_ but still _"open"_.

RobMahn, 2023-01-07:
I 'm using gcc (Debian 10.2.1-6) 10.2.1 20210110.

Using make bin/undionly.kpxe NO_WERROR=1, I get this error.

Do I need to downgrade the gcc?

  [BUILD] bin/int13.o
arch/i386/interface/pcbios/int13.c: In function ‘register_int13_drive’:
arch/i386/interface/pcbios/int13.c:469:2: warning: ‘asm’ operand 0 probably does not match constraints
  469 |  __asm__  __volatile__ (
      |  ^~~~~~~
arch/i386/interface/pcbios/int13.c:469:2: error: impossible constraint in ‘asm’
make: *** [bin/deps/arch/i386/interface/pcbios/int13.c.d:53: bin/int13.o] Error 1

mcb30, 2023-01-07:
> [BUILD] bin/int13.o arch/i386/interface/pcbios/int13.c: In function ‘register_int13_drive’: arch/i386/interface/pcbios/int13.c:469:2: warning: ‘asm’ operand 0 probably does not match constraints 469 [...]

This is an incredibly old commit: that code was removed in 2010.  Try your bisection starting from a newer known-good commit.

RobMahn, 2023-01-07:
If I understand correctly, I can only clone from a tag, not a commit.
v1.0.0		2010-02-02
The next tag, which I compiled with no errors, but has the problem is:
v1.20.1		2020-01-02

mcb30, 2023-01-07:
> If I understand correctly, I can only clone from a tag, not a commit.

No.  For bisection, you clone the current, up-to-date tree, and then start bisecting.  You can use any commit as the starting point for bisection, not just tags.

There are instructions at https://ipxe.org/howto/bisect that walk you through the process.

stappersg, 2023-01-07:
> There are instructions at https://ipxe.org/howto/bisect that walk you through the process.

Replace the _good_ from that webpage with the _good_ from the top of this page, like:
```text
git bisect good c3d1e78
```

RobMahn, 2023-01-08:
I really appreciate all the support.  I was trying to clone from a tag to get to a working version that I could confirm.  Bisecting from "good c3d1e78" compiles at first, but fails to compile after a couple of iterations (HEAD detached at 79697c75) and I never made it to a "good" version.  Can't figure out how to compile ignoring errors, it that even makes sense.  Now, I'm working on how to get an older version of gcc installed.

mcb30, 2023-01-08:
What error did you get on  [79697c7](https://github.com/ipxe/ipxe/commit/79697c75eeba01f5068412e03bc5484152f98341) when building with `NO_WERROR=1`?

RobMahn, 2023-01-08:
[BUILD] bin/privkey.o
/tmp/cc9xXpcf.s: Assembler messages:
/tmp/cc9xXpcf.s:32: Error: missing or invalid displacement expression `private_key_len@GOTOFF'
make: *** [Makefile.housekeeping:852: bin/privkey.o] Error 1

mcb30, 2023-01-08:
```
> [BUILD] bin/privkey.o
/tmp/cc9xXpcf.s: Assembler messages:
/tmp/cc9xXpcf.s:32: Error: missing or invalid displacement expression `private_key_len@GOTOFF'
make: *** [Makefile.housekeeping:852: bin/privkey.o] Error 1
```

That's very odd.  The code in `crypto/privkey.c` has not changed in any meaningful way between that commit and now, so if you are able to compile the current code then there's no reason why it shouldn't also work for that older code.

stappersg, 2023-01-08:
> if you are able to compile

I suggest to do `make clean` and another `make NO_WERROR=1`.

RobMahn, 2023-01-23:
It was taking too much time to figure out how to compile an older version, so 40 years after my last C coding, I just went in to figure out the problem.  I'm just playing in my retirement.

In src/usr/pxemenu.c, change:
`row = ( LINES - menu->num_items + index );`
to
`row = ( LINES - 1 - menu->num_items + index ); // Place menu at bottom of screen`
or
`row = ( ( LINES / 2) - 1 - (menu->num_items / 2) + index ); // Place menu in center of screen`

Now I'm looking into the missing pxe-prompt and crashing with too many pxe-service lines.

mcb30, 2023-01-23:
> In src/usr/pxemenu.c, change: `row = ( LINES - menu->num_items + index );` to `row = ( LINES - 1 - menu->num_items + index ); // Place menu at bottom of screen` or `row = ( ( LINES / 2) - 1 - (menu->num_items / 2) + index ); // Place menu in center of screen`

Thanks.  I suspect this quirk may have been introduced in commit https://github.com/ipxe/ipxe/commit/03401f9d2 which happened to correct the default console size from 80x24 to 80x25.

I've created PR #858 (based on your suggestions) which should fix this - could you please test and confirm that it works as expected?  Also, if you want the `Debugged-by:` credit in the commit log, please let me know your email address.

Thanks!

RobMahn, 2023-01-23:
I cloned your branch and compiled your change.  It worked as expected.

I'm not worried about credit, thank you.  I do wonder why you like the menu positioned at the bottom of the screen.  I would have selected either top or center.

stappersg, 2023-01-23:
On Mon, Jan 23, 2023 at 12:20:42PM -0800, Rob Mahn wrote:
> I do wonder why you like the menu positioned at the bottom of the screen.

Bottom of screen is where TTY output goes.

> I would have selected either top or center.

a Serial line is one of those things what you mis when it is gone.

Groeten
Geert Stappers
-- 
Silence is hard to parse

mcb30, 2023-01-23:
> I cloned your branch and compiled your change. It worked as expected.
> 
> I'm not worried about credit, thank you. I do wonder why you like the menu positioned at the bottom of the screen. I would have selected either top or center.

Thank you for testing!

The appearance of legacy PXE menus is not something that we really think about that much.  For pretty menus (and the ability to generate content dynamically, choose colours, use a background image, create submenus, define shortcut keys, etc) there is iPXE's native [menu capability](https://ipxe.org/cmd/choose).

[image]

perkelix, 2023-01-24:
Pulled up to commit e72670ad7ba6f6d8058359c990375e36e71a1cc4. This indeed works. Thanks!

tonho911, 2024-06-21:
> I cloned your branch and compiled your change. It worked as expected.
> 
> I'm not worried about credit, thank you. I do wonder why you like the menu positioned at the bottom of the screen. I would have selected either top or center.

@RobMahn 
can you share ipxe v1.0.0 2010-02-02
which no longer has an error in crypto/privkey.c?
I have tried but it doesn't work by changing,thanks

In src/usr/pxemenu.c, change:
row = ( LINES - menu->num_items + index );
to
row = ( LINES - 1 - menu->num_items + index ); // Place menu at bottom of screen
or
row = ( ( LINES / 2) - 1 - (menu->num_items / 2) + index ); // Place menu in center of screen