ipxe/ipxe #699: Input/output error on some ENA-enabled EC2 servers

magnetarnix, 2022-08-05:
Hello, when trying to boot some AWS EC2 servers (least on servers type `c6a.large`, `c6a.16xlarge` and `c6i.32xlarge`) using the latest iPXE AMI I get the error:
```
iPXE 1.21.1+ (gd3c89) -- Open Source Network Boot Firmware -- https://ipxe.org
Features: DNS HTTP HTTPS iSCSI TFTP SRP AoE ELF MBOOT PXE bzImage Menu PXEXT
Amazon EC2 - iPXE boot via user-data
CPU: AuthenticAMD AMD EPYC 7R13 Processor
net0: 0e:3e:1b:32:0b:a9 using ena-vf on 0000:00:05.0 (Ethernet) [closed]
  [Link:up, TX:0 TXE:0 RX:0 RXE:0]
DHCP attempt 1
Could not open net0: Input/output error (https://ipxe.org/1dc94039)
[Same error for 9 attempts]
DHCP failed - rebooting
```

I recompiled the latest version of `master` with debug support to get more info (with `make CONFIG=cloud EMBED=config/cloud/aws.ipxe DEBUG=pci:3,ena:3 bin/ipxe.usb`, then uploaded it using `./contrib/cloud/aws-import`) and I get:
```
iPXE initialising devices...0000:00:00.0 (8086:1237 class 060000) has no driver
0000:00:01.0 (8086:7000 class 060100) has no driver
0000:00:01.3 (8086:7113 class 000000) has no driver
0000:00:03.0 (1d0f:1111 class 030000) has no driver
0000:00:04.0 (1d0f:8061 class 010802) has no driver
0000:00:05.0 (1d0f:ec20) has driver "ena-vf"
0000:00:05.0 has mem febf8000 io 0 irq 0
0000:00:05.0 device not enabled by BIOS! Updating PCI command 0103->0107
0000:00:05.0 latency timer is unreasonably low at 0. Setting to 32.
ENA 0xec278 AQ [bff52400,bff52480) ACQ [bff52380,bff52400)
ENA 0xec278 admin request 0x0:
bff52400 : 00 00 08 01 00 00 00 00-00 00 00 00 00 00 00 00 : ................
bff52410 : 00 01 00 00 00 00 00 00-00 00 00 00 00 00 00 00 : ................
bff52420 : 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00 : ................
bff52430 : 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00 : ................
ENA 0xec278 admin response 0x0:
bff52380 : 00 00 00 01 00 00 00 00-01 00 00 00 0a 00 00 00 : ................
bff52390 : 86 58 20 14 01 00 00 00-30 00 00 00 40 00 00 00 : .X .....0...@...
bff523a0 : 0e 3e 1b 32 0b a9 00 00-00 24 00 00 00 00 00 00 : .>.2.....$......
bff523b0 : 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00 : ................
ENA 0xec278 MAC 0e:3e:1b:32:0b:a9 MTU 9216
0000:00:1f.0 (1d0f:8061 class 010802) has no driver
ok

iPXE 1.21.1+ (gd3c89) -- Open Source Network Boot Firmware -- https://ipxe.org
Features: DNS HTTP HTTPS iSCSI TFTP SRP AoE ELF MBOOT PXE bzImage Menu PXEXT
Amazon EC2 - iPXE boot via user-data
CPU: AuthenticAMD AMD EPYC 7R13 Processor
net0: 0e:3e:1b:32:0b:a9 using ena-vf on 0000:00:05.0 (Ethernet) [closed]
  [Link:up, TX:0 TXE:0 RX:0 RXE:0]
DHCP attempt 1
ENA 0xec278 admin request 0x1:
bff52440 : 01 00 03 01 00 02 10 00-00 00 00 00 00 30 f5 bf : .............0..
bff52450 : 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00 : ................
bff52460 : 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00 : ................
bff52470 : 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00 : ................
ENA 0xec278 admin response 0x1:
bff523c0 : 01 00 06 01 00 00 00 00-00 00 00 00 00 00 00 00 : ................
bff523d0 : 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00 : ................
bff523e0 : 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00 : ................
bff523f0 : 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00 : ................
ENA 0xec278 admin response 0x1 status 6:
bff52440 : 01 00 03 01 00 02 10 00-00 00 00 00 00 30 f5 bf : .............0..
bff52450 : 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00 : ................
bff52460 : 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00 : ................
bff52470 : 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00 : ................
bff523c0 : 01 00 06 01 00 00 00 00-00 00 00 00 00 00 00 00 : ................
bff523d0 : 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00 : ................
bff523e0 : 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00 : ................
bff523f0 : 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00 : ................
Could not open net0: Input/output error (https://ipxe.org/1dc94039)
net0: 0e:3e:1b:32:0b:a9 using ena-vf on 0000:00:05.0 (Ethernet) [closed]
  [Link:up, TX:0 TXE:0 RX:0 RXE:0]
DHCP attempt 2
[...]
```

@mcb30 I see that you worked on the ENA driver, maybe you have some ideas of what could be going wrong here?
I am available to do more tests if needed, just let me know what you need.

mcb30, 2022-08-05:
@magnetarnix Thanks for the bug report.  The NIC is being fairly unhelpful: the only information in the response is the error code 6, which is `ENA_ADMIN_UNKNOWN_ERROR` according to the Linux kernel driver.

Is this error 100% reproducible on those instance types, or is it intermittent?

If 100% reproducible, you could try tweaking the request parameters constructed by `ena_create_cq()` to see if the hardware can be be mollified with some alternative values.  It's unfortunately reasonably common with PCI virtual functions that newer versions of the PF driver will impose additional undocumented constraints on parameter values.  Based on experience with e.g. the Intel 40Gig/100Gig family, you might want to try (separately):

- Increase `ENA_TX_COUNT` to a larger power of 2
- Set `req->vector = 1` to request a non-zero MSI-X vector
- Set `req->intr = 0x20` to request that interrupts are enabled

Good luck!

magnetarnix, 2022-08-05:
Thanks for the quick answer! For those instance types I had this issue 100% of the time, so unless I was very unlucky it seems 100% reproducible.

Following your advice:
- Increasing `ENA_TX_COUNT` to 32: same issue
- Increasing `ENA_TX_COUNT` to 64: same issue
- In `ena_create_cq()`, `req->create_cq.vector = 1;`: same issue
- In `ena_create_cq()`, `req->create_cq.intr = 0x20;`: same issue
- In `ena_create_cq()`, both `req->create_cq.vector = 1;` and `req->create_cq.intr = 0x20;`: same issue
- Increasing `ENA_TX_COUNT` to 32 and both `req->create_cq.vector = 1;` and `req->create_cq.intr = 0x20;`: same issue

Note than when I say "same issue", I compared the hex dumps to verify that the bytes were indeed modified for each request, but I still got the error message (which stayed exactly identical for each attempt).

Do you have any other idea I could try by chance?

stappersg, 2022-08-05:
> Do you have any other idea I could try by chance?

I have an idea that is cheap to verify:  Increase `ENA_TX_COUNT` to 128.  So eight times the original value.  I'm not sure if the original value was 16, but the 32 and 64 made me think so.

Reason for eight times is the MTU of  9216 ( as seen in `ENA 0xec278 MAC 0e:3e:1b:32:0b:a9 MTU 9216` )  and  a more normal MTU of 1500.    9216/1500 is about six.

And while writing down that idea, can another idea:    MTU smaller as the maximum of 9216.    Do know that I have no idea how to accomplise that.   Reason for this message is that I could not resist to ignore the _"any other idea I could try"_.

magnetarnix, 2022-08-05:
> I have an idea that is cheap to verify: Increase `ENA_TX_COUNT` to 128. So eight times the original value. I'm not sure if the original value was 16, but the 32 and 64 made me think so.

Yes it was set to 16, I tried to increase it to 128 but same issue again! It was a good idea though!

For the MTU it seems like it queries it from the network adapter but doesn't seem to try to set it at some point, but I might be mistaken.

stappersg, 2022-08-05:
Observation:
```text
$ grep ^ENA /tmp/text_file_copy_and_paste_debug_attempt_from_issue_openings_message
ENA 0xec278 AQ [bff52400,bff52480) ACQ [bff52380,bff52400)
ENA 0xec278 admin request 0x0:
ENA 0xec278 admin response 0x0:
ENA 0xec278 MAC 0e:3e:1b:32:0b:a9 MTU 9216
ENA 0xec278 admin request 0x1:
ENA 0xec278 admin response 0x1:
ENA 0xec278 admin response 0x1 status 6:
$
```
Should that be read as _admin request 0x1_ got **two** responses?

mcb30, 2022-08-05:
> Note than when I say "same issue", I compared the hex dumps to verify that the bytes were indeed modified for each request, but I still got the error message (which stayed exactly identical for each attempt).
> 
> Do you have any other idea I could try by chance?

Thanks for testing.  I would probably next add some debug code to the Linux kernel driver to dump out the exact admin queue commands that it submits, and compare those against the (existing) admin queue command dumps from iPXE to see if there are any significant differences.

magnetarnix, 2022-08-08:
> Should that be read as _admin request 0x1_ got **two** responses?

It only got 1 response, but if this response is an error, it prints back the request that triggered the erroneous response and the erroneous response again! So that's why you see 2 `admin response`, and the hex dump of the second one is twice the size, because it contains the request and the response concatenated. See https://github.com/ipxe/ipxe/blob/0965cec53c9e421f900bb050b1c83196a0a5687b/src/drivers/net/ena.c#L344-L346

> Thanks for testing. I would probably next add some debug code to the Linux kernel driver to dump out the exact admin queue commands that it submits, and compare those against the (existing) admin queue command dumps from iPXE to see if there are any significant differences.

As I was using iPXE only for a very basic setup, I found out a way to work around the limitation by not using the iPXE image but by having the server boot by default on a Linux distro with a working ENA driver, automatically download the required `initrd`/`bzImage` files, setup the server bootloader to boot from those downloaded files only for the next boot, and reboot the server. This way each time the server reboots it updates the files and reboots on them, simulating what iPXE does.

This is a "good enough" setup that works for me right now so I'll stop investigating on my side for now until I have the use for more advanced iPXE features.

stappersg, 2022-08-08:
Thanks for reporting back.

> right now so I'll stop investigating

Please prevent that this issue keeps dangling open,  close it.

ech68, 2022-08-12:
I'd like to try to track down or help find a fix for this (I'm sure others will probably come across this issue as time goes by)...

Were there any other potential fixes you'd suggest (I see the possible addition of debug statements to the linux kernel driver mentioned above)?

stappersg, 2022-08-19:
I still think this is issue should be closed. Either by the _original poster_ or by _iPXE community members_ that have the privilege to close issues.
Other option I see that would help the _iPXE project_ is that this issue gets an _assignee_.
Goal is getting us beyond _"dangling issues are fine"_.

Regarding 
> I'd like to try to track down or help find a fix for this. Were there any other potential fixes you'd suggest?

https://ipxe.org/contact#developers

mcb30, 2022-08-27:
This should be fixed as of the latest series of commits:

  https://github.com/ipxe/ipxe/commit/856ffe000
  https://github.com/ipxe/ipxe/commit/6d2cead46
  https://github.com/ipxe/ipxe/commit/9f81e97af
  https://github.com/ipxe/ipxe/commit/3b81a4e25
  https://github.com/ipxe/ipxe/commit/a80124456

closed 2022-08-27 by mcb30

stappersg, 2022-08-27:
> This should be fixed as of the latest series of commits:

```text
stappers@myhost:~/src/mailinglists/ipxe
$ git log --oneline | head -n 5
a8012445 [ena] Increase receive ring size to 128 entries
3b81a4e2 [ena] Provide a host information page
9f81e97a [ena] Specify the unused completion queue MSI-X vector as 0xffffffff
6d2cead4 [ena] Allow for out-of-order completions
856ffe00 [ena] Limit submission queue fill level to completion queue size
stappers@myhost:~/src/mailinglists/ipxe
$ git log 856ffe00...6d2cead4
commit 6d2cead461db7243330f3275ff9ea7ff4607c4f8
Author: Michael Brown <mcb30@ipxe.org>
Date:   Fri Aug 26 15:48:52 2022 +0100

    [ena] Allow for out-of-order completions
    
    The ENA data path design has separate submission and completion
    queues.  Submission queues must be refilled in strict order (since
    there is only a single linear tail pointer used to communicate the
    existence of new entries to the hardware), and completion queue
    entries include a request identifier copied verbatim from the
    submission queue entry.  Once the submission queue doorbell has been
    rung, software never again reads from the submission queue entry and
    nothing ever needs to write back to the submission queue entry since
    completions are reported via the separate completion queue.
    
    This design allows the hardware to complete submission queue entries
    out of order, provided that it internally caches at least as many
    entries as it leaves gaps.
    
    Record and identify I/O buffers by request identifier (using a
    circular ring buffer of unique request identifiers), and remove the
    assumption that submission queue entries will be completed in order.
    
    Signed-off-by: Michael Brown <mcb30@ipxe.org>
stappers@myhost:~/src/mailinglists/ipxe
$ 
```

That is nice to see.  Thanks.

magnetarnix, 2022-08-29:
I can confirm that it is working on `c6a`/`c6i` servers, thanks a lot for the fix it's much appreciated :)