ipxe/ipxe #706: Trying to build on Debian 11

hradec, 2022-08-16:
I'm getting this error during build using `make NO_WERROR=1 bin/undionly.kpxe`:
```
  [BUILD] bin/ath5k.ids.o
  [BUILD] bin/ath9k.ids.o
  [HOSTCC] util/zbin
  [VERSION] bin/version.undionly.kpxe.o
In file included from include/ipxe/features.h:6,
                 from core/version.c:33:
<command-line>: error: 'cad1' undeclared here (not in a function)
include/ipxe/dhcp.h:529:58: note: in definition of macro 'DHCP_OPTION'
  529 | #define DHCP_OPTION( ... ) VA_ARG_COUNT ( __VA_ARGS__ ), __VA_ARGS__
      |                                                          ^~~~~~~~~~~
include/ipxe/features.h:71:2: note: in expansion of macro '__DHCP_FEATURE'
   71 |  __DHCP_FEATURE ( _name, feature_opt, __VA_ARGS__ )
      |  ^~~~~~~~~~~~~~
include/ipxe/features.h:69:2: note: in expansion of macro '_DHCP_FEATURE'
   69 |  _DHCP_FEATURE ( OBJECT, feature_opt, __VA_ARGS__ )
      |  ^~~~~~~~~~~~~
include/ipxe/features.h:106:2: note: in expansion of macro 'DHCP_FEATURE'
  106 |  DHCP_FEATURE ( DHCP_ENCAPSULATED ( DHCP_EB_VERSION ), __VA_ARGS__ )
      |  ^~~~~~~~~~~~
core/version.c:49:1: note: in expansion of macro 'FEATURE_VERSION'
   49 | FEATURE_VERSION ( VERSION_MAJOR, VERSION_MINOR, VERSION_PATCH );
      | ^~~~~~~~~~~~~~~
core/version.c:49:19: note: in expansion of macro 'VERSION_MAJOR'
   49 | FEATURE_VERSION ( VERSION_MAJOR, VERSION_MINOR, VERSION_PATCH );
      |                   ^~~~~~~~~~~~~
core/version.c:49:47: error: expected expression before ',' token
   49 | FEATURE_VERSION ( VERSION_MAJOR, VERSION_MINOR, VERSION_PATCH );
      |                                               ^
include/ipxe/dhcp.h:529:58: note: in definition of macro 'DHCP_OPTION'
  529 | #define DHCP_OPTION( ... ) VA_ARG_COUNT ( __VA_ARGS__ ), __VA_ARGS__
      |                                                          ^~~~~~~~~~~
include/ipxe/features.h:71:2: note: in expansion of macro '__DHCP_FEATURE'
   71 |  __DHCP_FEATURE ( _name, feature_opt, __VA_ARGS__ )
      |  ^~~~~~~~~~~~~~
include/ipxe/features.h:69:2: note: in expansion of macro '_DHCP_FEATURE'
   69 |  _DHCP_FEATURE ( OBJECT, feature_opt, __VA_ARGS__ )
      |  ^~~~~~~~~~~~~
include/ipxe/features.h:106:2: note: in expansion of macro 'DHCP_FEATURE'
  106 |  DHCP_FEATURE ( DHCP_ENCAPSULATED ( DHCP_EB_VERSION ), __VA_ARGS__ )
      |  ^~~~~~~~~~~~
core/version.c:49:1: note: in expansion of macro 'FEATURE_VERSION'
   49 | FEATURE_VERSION ( VERSION_MAJOR, VERSION_MINOR, VERSION_PATCH );
      | ^~~~~~~~~~~~~~~
<command-line>: warning: initialization of 'int' from 'uint8_t *' {aka 'unsigned char *'} makes integer from pointer without a cast [-Wint-conversion]
core/version.c:64:35: note: in expansion of macro 'VERSION_MAJOR'
   64 | const int product_major_version = VERSION_MAJOR;
      |                                   ^~~~~~~~~~~~~
<command-line>: error: initializer element is not constant
core/version.c:64:35: note: in expansion of macro 'VERSION_MAJOR'
   64 | const int product_major_version = VERSION_MAJOR;
      |                                   ^~~~~~~~~~~~~
core/version.c:67:48: error: expected expression before ';' token
   67 | const int product_minor_version = VERSION_MINOR;
      |                                                ^
core/version.c:67:48: warning: initialization of 'int' from 'uint8_t *' {aka 'unsigned char *'} makes integer from pointer without a cast [-Wint-conversion]
core/version.c:67:48: error: initializer element is not constant
make[1]: *** [Makefile.housekeeping:1209: bin/version.undionly.kpxe.o] Error 1
make[1]: *** Waiting for unfinished jobs....
make[1]: Leaving directory '/root/docker/pxe-manager/tftp/grub-netboot-git/ipxe/git/src'
make: *** [Makefile:104: docker_build_ipxe] Error 2
make: *** [Makefile:93: ipxe] Error 2
```

any ideas?

hradec, 2022-08-16:
btw, I'm building using a docker container. The Dockerfile is this one for reference: 
```
FROM debian:11

RUN apt update ;\
    apt install -y \
        zip \
        dosfstools \
        syslinux-utils \
        genisoimage \
        build-essential \
        wget

RUN apt install -y \
        squashfs-tools \
        cpio \
        curl \
        util-linux

RUN apt install -y \
        grub2 \
        sudo

RUN apt install -y \
        autoconf \
        automake \
        git

RUN apt install -y \
        autopoint \
        pkg-config \
        bison \
        flex

RUN apt install -y \
        python \
        libdevmapper-dev \
        libfreetype-dev \
        ipxe \
        extlinux \
        isolinux

RUN apt install -y \
        libfuse-dev \
        fonts-dejavu \
        fonts-dejavu-core \
        fonts-dejavu-extra

RUN apt install -y \
        lzma-dev \
        liblzma-dev
```
Maybe I'm missing some package?

hradec, 2022-08-16:
Just because I've noticed the github action uses Ubuntu 20.04, I gave it a try and got an error at the same place: 
```
  [HOSTCC] util/zbin
  [BUILD] bin/ath9k.ids.o
  [VERSION] bin/version.undionly.kpxe.o
  [AR] bin/blib.a
ar: creating bin/blib.a
In file included from include/ipxe/features.h:6,
                 from core/version.c:33:
<command-line>: error: 'cad1' undeclared here (not in a function)
include/ipxe/dhcp.h:529:58: note: in definition of macro 'DHCP_OPTION'
  529 | #define DHCP_OPTION( ... ) VA_ARG_COUNT ( __VA_ARGS__ ), __VA_ARGS__
      |                                                          ^~~~~~~~~~~
include/ipxe/features.h:71:2: note: in expansion of macro '__DHCP_FEATURE'
   71 |  __DHCP_FEATURE ( _name, feature_opt, __VA_ARGS__ )
      |  ^~~~~~~~~~~~~~
include/ipxe/features.h:69:2: note: in expansion of macro '_DHCP_FEATURE'
   69 |  _DHCP_FEATURE ( OBJECT, feature_opt, __VA_ARGS__ )
      |  ^~~~~~~~~~~~~
include/ipxe/features.h:106:2: note: in expansion of macro 'DHCP_FEATURE'
  106 |  DHCP_FEATURE ( DHCP_ENCAPSULATED ( DHCP_EB_VERSION ), __VA_ARGS__ )
      |  ^~~~~~~~~~~~
core/version.c:49:1: note: in expansion of macro 'FEATURE_VERSION'
   49 | FEATURE_VERSION ( VERSION_MAJOR, VERSION_MINOR, VERSION_PATCH );
      | ^~~~~~~~~~~~~~~
core/version.c:49:19: note: in expansion of macro 'VERSION_MAJOR'
   49 | FEATURE_VERSION ( VERSION_MAJOR, VERSION_MINOR, VERSION_PATCH );
      |                   ^~~~~~~~~~~~~
core/version.c:49:47: error: expected expression before ',' token
   49 | FEATURE_VERSION ( VERSION_MAJOR, VERSION_MINOR, VERSION_PATCH );
      |                                               ^
include/ipxe/dhcp.h:529:58: note: in definition of macro 'DHCP_OPTION'
  529 | #define DHCP_OPTION( ... ) VA_ARG_COUNT ( __VA_ARGS__ ), __VA_ARGS__
      |                                                          ^~~~~~~~~~~
include/ipxe/features.h:71:2: note: in expansion of macro '__DHCP_FEATURE'
   71 |  __DHCP_FEATURE ( _name, feature_opt, __VA_ARGS__ )
      |  ^~~~~~~~~~~~~~
include/ipxe/features.h:69:2: note: in expansion of macro '_DHCP_FEATURE'
   69 |  _DHCP_FEATURE ( OBJECT, feature_opt, __VA_ARGS__ )
      |  ^~~~~~~~~~~~~
include/ipxe/features.h:106:2: note: in expansion of macro 'DHCP_FEATURE'
  106 |  DHCP_FEATURE ( DHCP_ENCAPSULATED ( DHCP_EB_VERSION ), __VA_ARGS__ )
      |  ^~~~~~~~~~~~
core/version.c:49:1: note: in expansion of macro 'FEATURE_VERSION'
   49 | FEATURE_VERSION ( VERSION_MAJOR, VERSION_MINOR, VERSION_PATCH );
      | ^~~~~~~~~~~~~~~
core/version.c:67:48: error: expected expression before ';' token
   67 | const int product_minor_version = VERSION_MINOR;
      |                                                ^
make[1]: *** [Makefile.housekeeping:1209: bin/version.undionly.kpxe.o] Error 1
make[1]: *** Waiting for unfinished jobs....
make[1]: Leaving directory '/root/docker/pxe-manager/tftp/grub-netboot-git/ipxe/git/src'
make: *** [Makefile:107: docker_build_ipxe] Error 2
make: *** [Makefile:96: ipxe] Error 2
```

the Dockerfile is as follow:
```
FROM ubuntu:20.04

ENV DEBIAN_FRONTEND=noninteractive

RUN apt update ;\
    apt install -y -q \
        zip \
        dosfstools \
        syslinux-utils \
        genisoimage \
        build-essential \
        wget

RUN apt install -y -q\
        squashfs-tools \
        cpio \
        curl \
        util-linux

RUN apt install -y -q\
        grub2 \
        sudo

RUN apt install -y -q\
        autoconf \
        automake \
        git

RUN apt install -y -q\
        autopoint \
        pkg-config \
        bison \
        flex

RUN apt install -y -q\
        python \
        libdevmapper-dev \
        libfreetype-dev \
        ipxe \
        extlinux \
        isolinux

RUN apt install -y -q\
        libfuse-dev \
        fonts-dejavu \
        fonts-dejavu-core \
        fonts-dejavu-extra

RUN apt install -y -q\
        lzma-dev \
        liblzma-dev

RUN dpkg --add-architecture i386 ;\
    apt update ;\
    apt install -y -o Acquire::Retries=50 \
        mtools syslinux isolinux \
        libc6-dev-i386 libc6-dbg:i386 valgrind
```

hradec, 2022-08-16:
Just to make sure my setup was working, I tried to build the latest tag (v1.21.1) and it indeed builds successfully. 

So it must be the main master branch is broken... but then I had a look at the github actions, and they are building fine: 
[image]

Sooo... any idea what I could be doing wrong to build the latest master branch, considering I'm building with the same ubuntu 20.04 as the github action does? (maybe the github ubuntu 20.04 has something else installed by default that I'm missing?)

NiKiZe, 2022-08-16:
Do you have a full git checkout?
Any local config modifications?
Which version of gcc?

There is some other issues reported due to broken GCC.

9001, 2022-08-16:
Seems like something's wrong regarding version number detection. Could you try running `make version`? It should say `1.21.1+ (gcad1)` on master currently

stappersg, 2022-08-18:
Today, timezone UTC+2, in the _iPXE IRC channel_:
```text
19:27 -!- FroMaster [~FroMaster@redacted] has joined #ipxe
19:28 < FroMaster> How do I get a version number into iPXE when I compile from source? It always ends up 'iPXE 1.0.0+'
19:37 < NiKiZe> FroMaster: do you have a full git checkout?
20:05 -!- FroMaster [~FroMaster@redacted] has quit [Quit: Client closed]
```
@hradec it was the _"do you have a full git checkout?"_ that triggered me to sent this reminder.

hradec, 2022-08-23:
> @NiKiZe  Do you have a full git checkout? 
I think so. I'm checking out using `it clone --recursive --depth=1 https://github.com/ipxe/ipxe.git` just to be safe. 

> @NiKiZe  Any local config modifications? 
no local configs, apart from a custom ipxe script I'm passing to `make` via `EMBED` argument. Ex: `make bin/undionly.kpxe EMBED=/tmp/script.ipxe`

> @NiKiZe  Which version of gcc?
running with the debian docker, I get gcc 10.2.1:
```
gcc (Debian 10.2.1-6) 10.2.1 20210110
Copyright (C) 2020 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```
3b. running with the ubuntu 20.04 docker, I get gcc 9.4.0:
```
gcc (Ubuntu 9.4.0-1ubuntu1~20.04.1) 9.4.0
Copyright (C) 2019 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

As I mentioned previously, boot docker containers (debian and ubuntu 20.04) give the same error. 
I'll try again today, by doing a clean checkout.

hradec, 2022-08-23:
Just tried now to do a clean checkout of the master branch, same error building. :(

hradec, 2022-08-23:
I've made this one liner for anyone to give it a try and see the error: 
```
docker run --rm hradec/ipxe-buid-ubuntu:20.04 /bin/bash -c 'cd /root && git clone --depth=1 --recursive https://github.com/ipxe/ipxe.git ipxe && cd ipxe/src && LANG=C make -j $(nproc) NO_WERROR=1 bin/undionly.kpxe'
```

This will run the same ubuntu 20.04 image I'm using (built fromthe ubuntu dockerfile above), checkout the latest ipxe master branch the same way I'm doing it, and do a make. 
I've tried on 3 different computers and I get the error on all of then.  (as it should be since the build is running inside a docker container)

hradec, 2022-08-23:
> @ 9001 Seems like something's wrong regarding version number detection. Could you try running `make version`? It should say `1.21.1+ (gcad1)` on master currently
make version give me this: 
```
...
  [DEPS] core/ansiesc.c
  [DEPS] core/ansicoldef.c
  [DEPS] core/ansicol.c
  [DEPS] core/acpimac.c
  [DEPS] core/acpi_settings.c
  [DEPS] core/acpi.c
  [DEPS] libgcc/implicit.c
  [DEPS] libgcc/icc.c
  [DEPS] libgcc/__umoddi3.c
  [DEPS] libgcc/__udivmoddi4.c
  [DEPS] libgcc/__udivdi3.c
  [DEPS] libgcc/__moddi3.c
  [DEPS] libgcc/__divmoddi4.c
  [DEPS] libgcc/__divdi3.c
c5af..+
```
I've used this one liner to run `make version` right after cloning the main master branch: 
debian 11: 
```
docker run --rm hradec/ipxe-buid-debian:11 /bin/bash -c 'cd /root && git clone --depth=1 --recursive https://github.com/ipxe/ipxe.git ipxe && cd ipxe/src && make version'
```
ubuntu 20:04
```
docker run --rm hradec/ipxe-buid-ubuntu:20.04 /bin/bash -c 'cd /root && git clone --depth=1 --recursive https://github.com/ipxe/ipxe.git ipxe && cd ipxe/src && make version'
```
If you ran booth lines above, you should see the same result.

hradec, 2022-08-23:
I think I've found the problem!!!

I've removed the `--depth=1` from the `git clone` before running `make version`, and now I'm seeing what looks like the correct result: `1.21.1+ (gc5af4)`

running: 
```
docker run --rm  hradec/ipxe-buid-ubuntu:20.04 /bin/bash -c 'cd /root && git clone --recursive https://github.com/ipxe/ipxe.git ipxe && cd ipxe/src && make version'
```
now returns 
```
...
  [DEPS] libgcc/__divmoddi4.c
  [DEPS] libgcc/__moddi3.c
  [DEPS] libgcc/__udivdi3.c
1.21.1+ (gc5af4)
```

I did the same for the `make bin/undionly.kpxe' and it finished successfully now, without any errors! 
now, this one liner:
```
docker run --rm hradec/ipxe-buid-ubuntu:20.04 /bin/bash -c 'cd /root && git clone --recursive https://github.com/ipxe/ipxe.git ipxe && cd ipxe/src && LANG=C make -j $(nproc) NO_WERROR=1 bin/undionly.kpxe'
```
finishes successfully!!!:D

It seems `--depth=1` during clone was causing the issue to detect the version (as @9001 mentioned), which caused the build error.

Thanks a lot for the help, guys! Really appreciate!

NiKiZe, 2022-08-23:
> > @NiKiZe  Do you have a full git checkout?

> I think so. I'm checking out using `git clone --recursive --depth=1 https://github.com/ipxe/ipxe.git` just to be safe.
> 

That is called a shallow checkout, not full.
Closing as duplicate of #491 and #454

closed 2022-08-23 by NiKiZe

stappersg, 2022-08-23:
> That is called a shallow checkout, not full.
> Closing as duplicate of #491 and #454

That does remind me of merge request #534 _"Better experience for shallow repository_".