ipxe/ipxe #707: UEFI BIOS Halt - Mellanox ConnectX-5 & Dell C6420

dhekimian, 2022-08-18:
Built ipxe.efi from source and got this strange UEFI BIOS error message before even printing the iPXE boot string. I am able to use a older version of ipxe.efi v1.20.1 (built from [git tag 1.20.1](https://github.com/ipxe/ipxe/archive/refs/tags/v1.20.1.tar.gz)) and v1.21.1+ (gc5af4) downloaded April 1, 2022 on this exact host so it doesn't appear to be a hardware issue but something changed in iPXE recently. 

I'm not certain on how to get better debug information but I tried with `DEBUG=init` to see if I could get more details. If there's another DEBUG option to try, please advise.

**iPXE 1.21.1+** - Downloaded from GitHub [c5af41a](https://github.com/ipxe/ipxe/commit/c5af41a6f5b5f4a420b3e539f9e3a8dc9f8dd03e)  August 18, 2022

[image]

[image]

### Build Steps
```
wget https://github.com/ipxe/ipxe/archive/c5af41a6f5b5f4a420b3e539f9e3a8dc9f8dd03e.tar.gz
tar xzvf v1.20.1.tar.gz
cd ipxe-1.20.1/src
make bin-x86_64-efi/ipxe.efi DEBUG=init
cp bin-x86_64-efi/ipxe.efi /var/lib/tftpboot/
```

Mellanox Version Info: 
```
/opt/mellanox/bin/mlxfwmanager --query
Querying Mellanox devices firmware ...

Device #1:
----------

  Device Type:      ConnectX5
  Part Number:      MCX516A-CCA_Ax
  Description:      ConnectX-5 EN network interface card; 100GbE dual-port QSFP28; PCIe3.0 x16; tall bracket; ROHS R6
  PSID:             MT_0000000012
  PCI Device Name:  mt4119_pciconf0
  Base GUID:        0c42a103008dcef0
  Base MAC:         0c42a18dcef0
  Versions:         Current        Available
     FW             16.32.1010     N/A
     PXE            3.6.0502       N/A
     UEFI           14.25.0017     N/A
```

mcb30, 2022-08-18:
You could try using https://ipxe.org/howto/bisect to identify the commit that first exhibits the problem.

stappersg, 2022-08-26:
An attempt to get this stalled issue forward:

```text
stappers@myhost:~/src/mailinglists/ipxe/src
$ git bisect start
You need to run this command from the toplevel of the working tree.
stappers@myhost:~/src/mailinglists/ipxe/src
$ cd ..
stappers@myhost:~/src/mailinglists/ipxe
$ git bisect start
stappers@myhost:~/src/mailinglists/ipxe
$ git bisect bad c5af41a6f
stappers@myhost:~/src/mailinglists/ipxe
$ git bisect good v1.20.1
Bisecting: 224 revisions left to test after this (roughly 8 steps)
[5e260c73f17d9d2ccbe89244d322c0cb1586d2d8] [build] Avoid shell brace expansion in "make clean"
stappers@myhost:~/src/mailinglists/ipxe
$ 
```
@dhekimian you probably want to copy and paste this
```text
cd ..
git bisect start
git bisect bad c5af41a6f
git bisect good v1.20.1
```

edgood1, 2022-12-12:
Also seeing this exact problem on ConnectX-6DX NICs.

stappersg, 2022-12-13:
Time will tell when outcome of _git bisect_ is reported.

edgood1, 2022-12-13:
I'll try to run through a bisect if I can find time. The commit we were previously using that was working fine is `e09e1142a3bd8bdb702efc92994c419a53e9933b`. That will be my starting point.