ipxe/ipxe #730: Retry on different port

stappersg, 2022-08-29:
Hi,

Seen in a packet capture:

```text
iPXE:  Hey Webserver, I want a TCP stream at port 15117
Webserver: Hi iPXE, we have already a stream at port 15117
iPXE: Okay Webserver, let us tear down our port 15117 stream
```
follow-uped by:
```text
iPXE:  Hey Webserver, I want a TCP stream at port 15117
Webserver: Hi iPXE, we have already a stream at port 15117
iPXE: Okay Webserver, let us tear down our port 15117 stream
```
plus **three** more:

```text
iPXE:  Hey Webserver, I want a TCP stream at port 15117
Webserver: Hi iPXE, we have already a stream at port 15117
iPXE: Okay Webserver, let us tear down our port 15117 stream
```
which resulted in a https://ipxe.org/err/4c0a60 _Connection timed out_.  ( https://github.com/ipxe/ipxe/discussions/704 )

This issue is for asking to improve _iPXE_ by doing _retries on different ports_.

stappersg, 2022-08-29:
For your information, this issue is a retry of #710. Difference is that `Okay Webserver, we lets tear down our port 15117 stream` is being changed into `Okay Webserver, let us tear down our port 15117 stream`.

[image]

NiKiZe, 2022-08-29:
Your spam is not acceptable

stappersg, 2022-08-29:
[image]

> Your spam is not acceptable

For the record:  I'm ignoring the personal attack.  At least for now. This issue is about getting retries on different ports. Improving iPXE.

Things that we have already records of:
* Issue #710 being closed with _"converted into a discussion"_ as **only** comment
* The silence in discussion #711

closed 2022-11-28 by zube[bot]