ipxe/ipxe #731: Intel X710-TM4 nics not found

loood, 2022-08-31:
I'm using ipxe.iso to boot a [Supermicro A+ Server 1124US-TNRP](https://www.supermicro.com/en/Aplus/system/1U/1124/AS-1124US-TNRP.cfm) which uses a Intel X710-TM4. ipxe doesn't seem to see the intel nics. Wanted to find out if those nics are suppose to be supported or what I can do to help get those supported?

mcb30, 2022-08-31:
What are the PCI vendor:device IDs for this NIC?

NiKiZe, 2022-08-31:
Also verify that you are using the latest version of iPXE. (And post the version number to verify)

loood, 2022-09-01:
Here is the lspci output with vendor:device info for the nics
```
~ # lspci -nn | grep -i intel
44:00.0 Ethernet controller [0200]: Intel Corporation Device [8086:15ff] (rev 02)
44:00.1 Ethernet controller [0200]: Intel Corporation Device [8086:15ff] (rev 02)
44:00.2 Ethernet controller [0200]: Intel Corporation Device [8086:104e] (rev 02)
44:00.3 Ethernet controller [0200]: Intel Corporation Device [8086:104e] (rev 02)
~ #
```

Output of ipxe version
```
iPXE 1.21.1+ (gc5af4) -- Open Source Network Boot Firmware -- https://ipxe.org
Features: DNS HTTP iSCSI TFTP SRP AoE EFI Menu
```

stappersg, 2022-09-01:
Newer version of exist and that does have those PCI-IDs supported.
```text
$ cd ../../mailinglists/ipxe/
stappers@myhost:~/src/mailinglists/ipxe
$ git branch
  gcc12
  getpciids
  main
  opensuse-7.1
  temp
* upstream
stappers@myhost:~/src/mailinglists/ipxe
$ git pull
Already up to date.
stappers@myhost:~/src/mailinglists/ipxe
$ git log --oneline | head -n 3
a8012445 [ena] Increase receive ring size to 128 entries
3b81a4e2 [ena] Provide a host information page
9f81e97a [ena] Specify the unused completion queue MSI-X vector as 0xffffffff
stappers@myhost:~/src/mailinglists/ipxe
$ git grep 8086:104
stappers@myhost:~/src/mailinglists/ipxe
$ git grep -l 8086
src/Makefile
src/arch/x86/drivers/net/undinet.c
src/drivers/net/eepro100.c
src/drivers/net/ice.c
src/drivers/net/igbvf/igbvf_main.c
src/drivers/net/intel.c
src/drivers/net/intelx.c
src/drivers/net/intelxl.c
src/drivers/net/intelxlvf.c
src/drivers/net/intelxvf.c
src/drivers/net/tulip.c
src/drivers/usb/xhci.c
src/include/ipxe/efi/Library/BaseLib.h
src/util/parserom.pl
stappers@myhost:~/src/mailinglists/ipxe
$ git grep 8086
src/Makefile:              bin/8086100e.mrom bin/80861209.rom bin/10500940.rom \
src/Makefile:              bin/8086100f.mrom bin/808610d3.mrom bin/15ad07b0.rom
src/Makefile:           bin-x86_64-pcbios/8086100e.mrom bin-x86_64-pcbios/intel.rom \
src/Makefile:vmware : bin/8086100f.mrom bin/808610d3.mrom bin/10222000.rom bin/15ad07b0.rom
src/Makefile:   @$(ECHO) '    bin/8086100f.mrom -- intel/e1000'
src/Makefile:   @$(ECHO) '    bin/808610d3.mrom -- intel/e1000e'
src/arch/x86/drivers/net/undinet.c:     { 0x8086, 0x1502, PCI_ANY_ID, PCI_ANY_ID },
src/arch/x86/drivers/net/undinet.c:     { 0x8086, 0x1503, PCI_ANY_ID, PCI_ANY_ID },
src/drivers/net/eepro100.c:PCI_ROM(0x8086, 0x1029, "id1029",        "Intel EtherExpressPro100 ID1029", 0),
src/drivers/net/eepro100.c:PCI_ROM(0x8086, 0x1030, "id1030",        "Intel EtherExpressPro100 ID1030", 0),
src/drivers/net/eepro100.c:PCI_ROM(0x8086, 0x1031, "82801cam",      "Intel 82801CAM (ICH3) Chipset Ethernet Controller", 0),
src/drivers/net/eepro100.c:PCI_ROM(0x8086, 0x1032, "eepro100-1032", "Intel PRO/100 VE Network Connection", 0),
src/drivers/net/eepro100.c:PCI_ROM(0x8086, 0x1033, "eepro100-1033", "Intel PRO/100 VM Network Connection", 0),
src/drivers/net/eepro100.c:PCI_ROM(0x8086, 0x1034, "eepro100-1034", "Intel PRO/100 VM Network Connection", 0),
src/drivers/net/eepro100.c:PCI_ROM(0x8086, 0x1035, "eepro100-1035", "Intel 82801CAM (ICH3) Chipset Ethernet Controller", 0),
src/drivers/net/eepro100.c:PCI_ROM(0x8086, 0x1036, "eepro100-1036", "Intel 82801CAM (ICH3) Chipset Ethernet Controller", 0),
src/drivers/net/eepro100.c:PCI_ROM(0x8086, 0x1037, "eepro100-1037", "Intel 82801CAM (ICH3) Chipset Ethernet Controller", 0),
src/drivers/net/eepro100.c:PCI_ROM(0x8086, 0x1038, "id1038",        "Intel PRO/100 VM Network Connection", 0),
src/drivers/net/eepro100.c:PCI_ROM(0x8086, 0x1039, "82562et",       "Intel PRO100 VE 82562ET", 0),
src/drivers/net/eepro100.c:PCI_ROM(0x8086, 0x103a, "id103a",        "Intel Corporation 82559 InBusiness 10/100", 0),
src/drivers/net/eepro100.c:PCI_ROM(0x8086, 0x103b, "82562etb",      "Intel PRO100 VE 82562ETB", 0),
src/drivers/net/eepro100.c:PCI_ROM(0x8086, 0x103c, "eepro100-103c", "Intel PRO/100 VM Network Connection", 0),
src/drivers/net/eepro100.c:PCI_ROM(0x8086, 0x103d, "eepro100-103d", "Intel PRO/100 VE Network Connection", 0),
src/drivers/net/eepro100.c:PCI_ROM(0x8086, 0x103e, "eepro100-103e", "Intel PRO/100 VM Network Connection", 0),
src/drivers/net/eepro100.c:PCI_ROM(0x8086, 0x1051, "prove",         "Intel PRO/100 VE Network Connection", 0),
src/drivers/net/eepro100.c:PCI_ROM(0x8086, 0x1059, "82551qm",       "Intel PRO/100 M Mobile Connection", 0),
src/drivers/net/eepro100.c:PCI_ROM(0x8086, 0x1209, "82559er",       "Intel EtherExpressPro100 82559ER", 0),
src/drivers/net/eepro100.c:PCI_ROM(0x8086, 0x1227, "82865",         "Intel 82865 EtherExpress PRO/100A", 0),
stappers@myhost:~/src/mailinglists/ipxe
$ git grep -l 8086
src/Makefile
src/arch/x86/drivers/net/undinet.c
src/drivers/net/eepro100.c
src/drivers/net/ice.c
src/drivers/net/igbvf/igbvf_main.c
src/drivers/net/intel.c
src/drivers/net/intelx.c
src/drivers/net/intelxl.c
src/drivers/net/intelxlvf.c
src/drivers/net/intelxvf.c
src/drivers/net/tulip.c
src/drivers/usb/xhci.c
src/include/ipxe/efi/Library/BaseLib.h
src/util/parserom.pl
stappers@myhost:~/src/mailinglists/ipxe
$ git grep 8086 | wc
    278    2641   24738
stappers@myhost:~/src/mailinglists/ipxe
$ git grep 0x8086 | wc
    269    2583   24017
stappers@myhost:~/src/mailinglists/ipxe
$ git grep 0x8086 | grep 0x104 | wc
      7      67     649
stappers@myhost:~/src/mailinglists/ipxe
$ git grep 0x8086 | grep 0x104
src/drivers/net/intel.c:	PCI_ROM ( 0x8086, 0x1049, "82566mm", "82566MM", INTEL_PBS_ERRATA ),
src/drivers/net/intel.c:	PCI_ROM ( 0x8086, 0x104a, "82566dm", "82566DM", INTEL_PBS_ERRATA ),
src/drivers/net/intel.c:	PCI_ROM ( 0x8086, 0x104b, "82566dc", "82566DC", INTEL_PBS_ERRATA ),
src/drivers/net/intel.c:	PCI_ROM ( 0x8086, 0x104c, "82562v", "82562V", INTEL_PBS_ERRATA ),
src/drivers/net/intel.c:	PCI_ROM ( 0x8086, 0x104d, "82566mc", "82566MC", INTEL_PBS_ERRATA ),
src/drivers/net/intelxl.c:	PCI_ROM ( 0x8086, 0x104e, "x710-sfp-b", "X710 10GbE SFP+", 0 ),
src/drivers/net/intelxl.c:	PCI_ROM ( 0x8086, 0x104f, "x710-kx-b", "X710 10GbE backplane", 0 ),
stappers@myhost:~/src/mailinglists/ipxe
$ git grep 0x8086 | grep 0x15ff
src/drivers/net/intelxl.c:	PCI_ROM ( 0x8086, 0x15ff, "x710-10gt-b", "X710 10GBASE-T", 0 ),
stappers@myhost:~/src/mailinglists/ipxe
$ 
```

mcb30, 2022-09-01:
@loood Thanks for confirming the IDs and version.  Could you please try building with `DEBUG=intelxl:3` and report the complete output that you get?

loood, 2022-09-02:
Recompiled today with the DEBUG=intelxl:3. This is the output:

```
iPXE initialising devices...INTELXL 0x9e922910 PF 7 using port 3 queues [0x07ff-0x07ff]
INTELXL 0x9e922910 could not enable MSI-X: No such file or directory (https://ipxe.org/2dcc408e)
INTELXL 0x9e922910 PF 7 using port 3 queues [0x07ff-0x07ff]
INTELXL 0x9e922910 could not enable MSI-X: No such file or directory (https://ipxe.org/2dcc408e)
INTELXL 0x9e922910 PF 7 using port 3 queues [0x07ff-0x07ff]
INTELXL 0x9e922910 could not enable MSI-X: No such file or directory (https://ipxe.org/2dcc408e)
INTELXL 0x9e922910 PF 7 using port 3 queues [0x07ff-0x07ff]
INTELXL 0x9e922910 could not enable MSI-X: No such file or directory (https://ipxe.org/2dcc408e)
ok

iPXE 1.21.1+ (ga8012) -- Open Source Network Boot Firmware -- https://ipxe.org
Features: DNS HTTP iSCSI TFTP SRP AoE EFI Menu

```
Then it continued on unsuccessfully with the rest of the ipxe script

stappersg, 2022-09-02:
Yes, `a0812` is the latest commit.

Some research:
```text
$ git blame src/drivers/net/intelxl.c | grep -e '0x8086, 0x15ff' -e '0x8086, 0x104e'
9dfcdc04c (Michael Brown 2022-08-08 14:13:34 +0100 1875) 	PCI_ROM ( 0x8086, 0x104e, "x710-sfp-b", "X710 10GbE SFP+", 0 ),
9dfcdc04c (Michael Brown 2022-08-08 14:13:34 +0100 1890) 	PCI_ROM ( 0x8086, 0x15ff, "x710-10gt-b", "X710 10GBASE-T", 0 ),
$
```

Please confirm that your branch contains commit `9dfcdc0`, share with us the output of
```text
git log --oneline | grep 9dfcdc0
```

Back to the `could not enable MSI-X` and https://ipxe.org/2dcc408e from the
```text
INTELXL 0x9e922910 could not enable MSI-X: No such file or directory (https://ipxe.org/2dcc408e)
```
lines.

For getting us further is a build with `DEBUG=intelxl:3,pcimsix`, another run and sharing the output needed.

loood, 2022-09-02:
Hi! The output from `git log --oneline | grep 9dfcdc0` is:
```
9dfcdc04 [intelxl] Update list of PCI IDs
```

The build script I have does a `git clone` each time in a clean space.

I rebuilt it with `DEBUG=intelxl:3,pcimsix` and this is the output:
```
iPXE initialising devices...INTELXL 0x9e921910 PF 7 using port 3 queues [0x07ff-0x07ff]
MSI-X 0x9e921950 has 129 vectors for 0000:44:00.0
MSI-X 0x9e921950 table could not find BAR3
INTELXL 0x9e921910 could not enable MSI-X: No such file or directory (https://ipxe.org/2dcc408e)
INTELXL 0x9e921910 PF 7 using port 3 queues [0x07ff-0x07ff]
MSI-X 0x9e921950 has 129 vectors for 0000:44:00.1
MSI-X 0x9e921950 table could not find BAR3
INTELXL 0x9e921910 could not enable MSI-X: No such file or directory (https://ipxe.org/2dcc408e)
INTELXL 0x9e921910 PF 7 using port 3 queues [0x07ff-0x07ff]
MSI-X 0x9e921950 has 129 vectors for 0000:44:00.2
MSI-X 0x9e921950 table could not find BAR3
INTELXL 0x9e921910 could not enable MSI-X: No such file or directory (https://ipxe.org/2dcc408e)
INTELXL 0x9e921910 PF 7 using port 3 queues [0x07ff-0x07ff]
MSI-X 0x9e921950 has 129 vectors for 0000:44:00.3
MSI-X 0x9e921950 table could not find BAR3
INTELXL 0x9e921910 could not enable MSI-X: No such file or directory (https://ipxe.org/2dcc408e)
ok

iPXE 1.21.1+ (ga8012) -- Open Source Network Boot Firmware -- https://ipxe.org
Features: DNS HTTP iSCSI TFTP SRP AoE EFI Menu
```

stappersg, 2022-09-02:
Thanks for confirming that the recent commit with additional PCI-IDs is present.

From https://duckduckgo.com/?q=MSI-X+table+could+not+find+BAR3 I did learn that `BAR` is _Base Address Register_.  I don't know how `BAR3` came in play.  I do hope that `DEBUG=intelxl:3,pcimsix:3` does provide further clues.

loood, 2022-09-02:
Tried `DEBUG=intelxl:3,pcimsix:3` but there was no difference in output.

mcb30, 2022-09-02:
@loood Thank you for the debug output. The values being read from the card are very strange: PF 7, port 3, queues 0x7ff-0x7ff, BAR 3.  All of these values represent all bits being set (within the bit width of the relevant field). It looks as though every single MMIO read from the card is returning 0xffffffff.

You are using UEFI, so it's possible that the platform firmware is providing a broken translation table or similar.  As a quick test: if the server supports booting in legacy BIOS mode, could you try that?

mcb30, 2022-09-02:
@loood Could you please also try (in UEFI mode) `DEBUG=intelxl:3,pcimsix,efi_pci:3`?

loood, 2022-09-05:
Here is the output for `DEBUG=intelxl:3,pcimsix,efi_pci:3`

```
iPXE initialising devices...EFIPCI 0000:00:00.0 (1022:1480 class 060000) has no driver
EFIPCI 0000:00:00.2 (1022:164f class 080600) has no driver
EFIPCI 0000:00:01.0 (1022:1482 class 060000) has no driver
EFIPCI 0000:00:02.0 (1022:1482 class 060000) has no driver
EFIPCI 0000:00:03.0 (1022:1482 class 060000) has no driver
EFIPCI 0000:00:04.0 (1022:1482 class 060000) has no driver
EFIPCI 0000:00:05.0 (1022:1482 class 060000) has no driver
EFIPCI 0000:00:07.0 (1022:1482 class 060000) has no driver
EFIPCI 0000:00:07.1 (1022:1484 class 060400) has no driver
EFIPCI 0000:01:00.0 (1022:148a class 130000) has no driver
EFIPCI 0000:01:00.2 (1022:1498 class 108000) has no driver
EFIPCI 0000:00:08.0 (1022:1482 class 060000) has no driver
EFIPCI 0000:00:08.1 (1022:1484 class 060400) has no driver
EFIPCI 0000:02:00.0 (1022:1485 class 130000) has no driver
EFIPCI 0000:02:00.2 (1022:1498 class 108000) has no driver
EFIPCI 0000:02:00.3 (1022:148c class 0c0330) has no driver
EFIPCI 0000:00:14.0 (1022:790b class 0c0500) has no driver
EFIPCI 0000:00:14.3 (1022:790e class 060100) has no driver
EFIPCI 0000:00:18.0 (1022:1650 class 060000) has no driver
EFIPCI 0000:00:18.1 (1022:1651 class 060000) has no driver
EFIPCI 0000:00:18.2 (1022:1652 class 060000) has no driver
EFIPCI 0000:00:18.3 (1022:1653 class 060000) has no driver
EFIPCI 0000:00:18.4 (1022:1654 class 060000) has no driver
EFIPCI 0000:00:18.5 (1022:1655 class 060000) has no driver
EFIPCI 0000:00:18.6 (1022:1656 class 060000) has no driver
EFIPCI 0000:00:18.7 (1022:1657 class 060000) has no driver
EFIPCI 0000:00:19.0 (1022:1650 class 060000) has no driver
EFIPCI 0000:00:19.1 (1022:1651 class 060000) has no driver
EFIPCI 0000:00:19.2 (1022:1652 class 060000) has no driver
EFIPCI 0000:00:19.3 (1022:1653 class 060000) has no driver
EFIPCI 0000:00:19.4 (1022:1654 class 060000) has no driver
EFIPCI 0000:00:19.5 (1022:1655 class 060000) has no driver
EFIPCI 0000:00:19.6 (1022:1656 class 060000) has no driver
EFIPCI 0000:00:19.7 (1022:1657 class 060000) has no driver
EFIPCI 0000:20:00.0 (1022:1480 class 060000) has no driver
EFIPCI 0000:20:00.2 (1022:164f class 080600) has no driver
EFIPCI 0000:20:01.0 (1022:1482 class 060000) has no driver
EFIPCI 0000:20:02.0 (1022:1482 class 060000) has no driver
EFIPCI 0000:20:03.0 (1022:1482 class 060000) has no driver
EFIPCI 0000:20:03.1 (1022:1483 class 060400) has no driver
EFIPCI 0000:21:00.0 (1e0f:0007 class 010802) has no driver
EFIPCI 0000:20:03.2 (1022:1483 class 060400) has no driver
EFIPCI 0000:22:00.0 (1e0f:0007 class 010802) has no driver
EFIPCI 0000:20:03.3 (1022:1483 class 060400) has no driver
EFIPCI 0000:23:00.0 (1e0f:0007 class 010802) has no driver
EFIPCI 0000:20:03.4 (1022:1483 class 060400) has no driver
EFIPCI 0000:24:00.0 (1e0f:0007 class 010802) has no driver
EFIPCI 0000:20:04.0 (1022:1482 class 060000) has no driver
EFIPCI 0000:20:05.0 (1022:1482 class 060000) has no driver
EFIPCI 0000:20:07.0 (1022:1482 class 060000) has no driver
EFIPCI 0000:20:07.1 (1022:1484 class 060400) has no driver
EFIPCI 0000:25:00.0 (1022:148a class 130000) has no driver
EFIPCI 0000:25:00.2 (1022:1498 class 108000) has no driver
EFIPCI 0000:20:08.0 (1022:1482 class 060000) has no driver
EFIPCI 0000:20:08.1 (1022:1484 class 060400) has no driver
EFIPCI 0000:26:00.0 (1022:1485 class 130000) has no driver
EFIPCI 0000:26:00.1 (1022:1486 class 108000) has no driver
EFIPCI 0000:26:00.2 (1022:1498 class 108000) has no driver
EFIPCI 0000:26:00.3 (1022:148c class 0c0330) has no driver
EFIPCI 0000:40:00.0 (1022:1480 class 060000) has no driver
EFIPCI 0000:40:00.2 (1022:164f class 080600) has no driver
EFIPCI 0000:40:01.0 (1022:1482 class 060000) has no driver
EFIPCI 0000:40:01.1 (1022:1483 class 060400) has no driver
EFIPCI 0000:41:00.0 (1a03:1150 class 060400) has no driver
EFIPCI 0000:42:00.0 (1a03:2000 class 030000) has no driver
EFIPCI 0000:40:01.3 (1022:1483 class 060400) has no driver
EFIPCI 0000:43:00.0 (1b21:1142 class 0c0330) has no driver
[... 178 lines cut ...]
EFIPCI 0000:e0:00.2 (1022:164f class 080600) has no driver
EFIPCI 0000:e0:01.0 (1022:1482 class 060000) has no driver
EFIPCI 0000:e0:02.0 (1022:1482 class 060000) has no driver
EFIPCI 0000:e0:03.0 (1022:1482 class 060000) has no driver
EFIPCI 0000:e0:03.1 (1022:1483 class 060400) has no driver
EFIPCI 0000:e0:03.2 (1022:1483 class 060400) has no driver
EFIPCI 0000:e0:03.3 (1022:1483 class 060400) has no driver
EFIPCI 0000:e0:03.4 (1022:1483 class 060400) has no driver
EFIPCI 0000:e0:04.0 (1022:1482 class 060000) has no driver
EFIPCI 0000:e0:05.0 (1022:1482 class 060000) has no driver
EFIPCI 0000:e0:07.0 (1022:1482 class 060000) has no driver
EFIPCI 0000:e0:07.1 (1022:1484 class 060400) has no driver
EFIPCI 0000:e5:00.0 (1022:148a class 130000) has no driver
EFIPCI 0000:e5:00.2 (1022:1498 class 108000) has no driver
EFIPCI 0000:e0:08.0 (1022:1482 class 060000) has no driver
EFIPCI 0000:e0:08.1 (1022:1484 class 060400) has no driver
EFIPCI 0000:e6:00.0 (1022:1485 class 130000) has no driver
EFIPCI 0000:e6:00.2 (1022:1498 class 108000) has no driver
ok
iPXE 1.21.1+ (ga8012) -- Open Source Network Boot Firmware -- https://ipxe.org
Features: DNS HTTP iSCSI TFTP SRP AoE EFI Menu
iPXE>
```

Haven't tried legacy boot but we're trying to convert to all uefi. Also like to add that I just tried the snp.efi (didn't know that was an option until a few days ago) and that works without issue.

stappersg, 2022-09-05:
Looking only at PCI slot `0000:44:00.2`:
```text
EFIPCI 0000:44:00.2 (8086:104e class 020000) has driver "x710-sfp-b"
EFIPCI 0000:44:00.2 (8086:104e class 020000) has driver "x710-sfp-b"
EFIPCI 0000:44:00.2 (8086:104e class 020000) has driver "x710-sfp-b"
EFIPCI 0000:44:00.2 found range [c9000000,ca900000) -> [c9000000,ca900000)
EFIPCI 0000:44:00.2 found range [c4000000,c8100000) -> [c4000000,c8100000)
EFIPCI 0000:44:00.2 c5000000+200000 -> c5000000
EFIPCI 0000:44:00.2 could not enable DAC: Error 0x7f10e083 (https://ipxe.org/7f10e083)
INTELXL 0x9ea1b910 PF 7 using port 3 queues [0x07ff-0x07ff]
MSI-X 0x9ea1b950 has 129 vectors for 0000:44:00.2
MSI-X 0x9ea1b950 table could not find BAR3
INTELXL 0x9ea1b910 could not enable MSI-X: No such file or directory (https://ipxe.org/2dcc408e)
EFIPCI 0000:44:00.2 could not probe driver "x710-sfp-b": No such file or directory (https://ipxe.org/2dcc408e)
EFIPCI 0000:44:00.2 (8086:104e class 020000) has driver "x710-sfp-b"
EFIPCI 0000:44:00.2 found range [c9000000,ca900000) -> [c9000000,ca900000)
EFIPCI 0000:44:00.2 found range [c4000000,c8100000) -> [c4000000,c8100000)
EFIPCI 0000:44:00.2 00000000+200000 is not within root bridge address space
EFIPCI 0000:44:00.2 could not probe driver "x710-sfp-b": No such device (https://ipxe.org/2ccb4087)
```

The lines with _http_ in it, to make them _clickable_.
> EFIPCI 0000:44:00.2 could not enable DAC: Error 0x7f10e083 (https://ipxe.org/7f10e083)
> INTELXL 0x9ea1b910 could not enable MSI-X: No such file or directory (https://ipxe.org/2dcc408e)
> EFIPCI 0000:44:00.2 could not probe driver "x710-sfp-b": No such file or directory (https://ipxe.org/2dcc408e)
> EFIPCI 0000:44:00.2 could not probe driver "x710-sfp-b": No such device (https://ipxe.org/2ccb4087)

I did visit those three different URLs. Their `DEBUG=` advice had nothing new to add to the current `DEBUG=intelxl:3,pcimsix,efi_pci:3`. I do hope other ideas do show up to solve this interesting problem.

mcb30, 2022-09-05:
Commentary inline below:

> ```
> EFIPCI 0000:44:00.0 (8086:15ff class 020000) has driver "x710-10gt-b"
> EFIPCI 0000:44:00.0 found range [c9000000,ca900000) -> [c9000000,ca900000)
> EFIPCI 0000:44:00.0 found range [...]

PF 0 is detected with memory BAR at 0xc7000000.  The address 0xc7000000 is within the root bridge range, and is apparently identity mapped.  So far, so good.

The "could not enable DAC" message is potentially harmless: from the pointers observed within the other debug messages (e.g. 0x9ea1b910), it appears that iPXE is running within the low 4GB of RAM and so dual address cycles will not be needed anyway.

For some unknown reason, all subsequent MMIO reads from the memory BAR appear to be returning all ones (0xffffffff), as described in my earlier message.

> ```
> EFIPCI 0000:44:00.0 (8086:15ff class 020000) has driver "x710-10gt-b"
> EFIPCI 0000:44:00.0 found range [c9000000,ca900000) -> [c9000000,ca900000)
> EFIPCI 0000:44:00.0 found range [...]

This is the same device (PF 0) being discovered again.  This is not entirely unsurprising, since other UEFI platforms have been observed to retry device probes several times in the event of a driver `Start()` method failing.

What is interesting is that the memory BAR now appears to have been zeroed (and hence disabled).

> ```
> EFIPCI 0000:44:00.1 (8086:15ff class 020000) has driver "x710-10gt-b"
> EFIPCI 0000:44:00.1 found range [c9000000,ca900000) -> [c9000000,ca900000)
> EFIPCI 0000:44:00.1 found range [...]

Same sequence of events happens for PF 1 ...

> ```
> EFIPCI 0000:44:00.2 (8086:104e class 020000) has driver "x710-sfp-b"
> EFIPCI 0000:44:00.2 (8086:104e class 020000) has driver "x710-sfp-b"
> EFIPCI 0000:44:00.2 (8086:104e class 020000) has driver [...]

... and again for PF 2 ...

> ```
> EFIPCI 0000:44:00.3 (8086:104e class 020000) has driver "x710-sfp-b"
> EFIPCI 0000:44:00.3 found range [c9000000,ca900000) -> [c9000000,ca900000)
> EFIPCI 0000:44:00.3 found range [...]

...and again for PF 3.

So, it's clear that something is zeroing the memory BAR, which would fully explain the fact that all reads seem to return 0xffffffff (since that is the defined behaviour when no PCI target responds to the read transaction).

It is highly suspicious that this seems to happen around the same time that the failed attempt to enable DAC is made, so the first experiment I'd want to try would be to comment out the call to `dma_set_mask_64bit ( intelxl->dma );` in `drivers/net/intelxl.c`.

If that doesn't change anything, then I'd try adding debug statements such as:

```
DBGC ( intelxl, "*** BAR0 = %08x\n", pci_bar ( pci, PCI_BASE_ADDRESS_0 ) );
```

at various points within `intelxl_probe()` to find the point at which BAR 0 (the memory BAR) gets zeroed.

loood, 2022-09-06:
Tried commenting out the `dma_set_mask_64bit ( intelxl->dma );` but still nothing. Will continue with adding DBGC lines.
```
iPXE initialising devices...EFIPCI 0000:00:00.0 (1022:1480 class 060000) has no driver
EFIPCI 0000:00:00.2 (1022:164f class 080600) has no driver
EFIPCI 0000:00:01.0 (1022:1482 class 060000) has no driver
EFIPCI 0000:00:02.0 (1022:1482 class 060000) has no driver
EFIPCI 0000:00:03.0 (1022:1482 class 060000) has no driver
EFIPCI 0000:00:04.0 (1022:1482 class 060000) has no driver
EFIPCI 0000:00:05.0 (1022:1482 class 060000) has no driver
EFIPCI 0000:00:07.0 (1022:1482 class 060000) has no driver
EFIPCI 0000:00:07.1 (1022:1484 class 060400) has no driver
EFIPCI 0000:01:00.0 (1022:148a class 130000) has no driver
EFIPCI 0000:01:00.2 (1022:1498 class 108000) has no driver
EFIPCI 0000:00:08.0 (1022:1482 class 060000) has no driver
EFIPCI 0000:00:08.1 (1022:1484 class 060400) has no driver
EFIPCI 0000:02:00.0 (1022:1485 class 130000) has no driver
EFIPCI 0000:02:00.2 (1022:1498 class 108000) has no driver
EFIPCI 0000:02:00.3 (1022:148c class 0c0330) has no driver
EFIPCI 0000:00:14.0 (1022:790b class 0c0500) has no driver
EFIPCI 0000:00:14.3 (1022:790e class 060100) has no driver
EFIPCI 0000:00:18.0 (1022:1650 class 060000) has no driver
EFIPCI 0000:00:18.1 (1022:1651 class 060000) has no driver
EFIPCI 0000:00:18.2 (1022:1652 class 060000) has no driver
EFIPCI 0000:00:18.3 (1022:1653 class 060000) has no driver
EFIPCI 0000:00:18.4 (1022:1654 class 060000) has no driver
EFIPCI 0000:00:18.5 (1022:1655 class 060000) has no driver
EFIPCI 0000:00:18.6 (1022:1656 class 060000) has no driver
EFIPCI 0000:00:18.7 (1022:1657 class 060000) has no driver
EFIPCI 0000:00:19.0 (1022:1650 class 060000) has no driver
EFIPCI 0000:00:19.1 (1022:1651 class 060000) has no driver
EFIPCI 0000:00:19.2 (1022:1652 class 060000) has no driver
EFIPCI 0000:00:19.3 (1022:1653 class 060000) has no driver
EFIPCI 0000:00:19.4 (1022:1654 class 060000) has no driver
EFIPCI 0000:00:19.5 (1022:1655 class 060000) has no driver
EFIPCI 0000:00:19.6 (1022:1656 class 060000) has no driver
EFIPCI 0000:00:19.7 (1022:1657 class 060000) has no driver
EFIPCI 0000:20:00.0 (1022:1480 class 060000) has no driver
EFIPCI 0000:20:00.2 (1022:164f class 080600) has no driver
EFIPCI 0000:20:01.0 (1022:1482 class 060000) has no driver
EFIPCI 0000:20:02.0 (1022:1482 class 060000) has no driver
EFIPCI 0000:20:03.0 (1022:1482 class 060000) has no driver
EFIPCI 0000:20:03.1 (1022:1483 class 060400) has no driver
EFIPCI 0000:21:00.0 (1e0f:0007 class 010802) has no driver
EFIPCI 0000:20:03.2 (1022:1483 class 060400) has no driver
EFIPCI 0000:22:00.0 (1e0f:0007 class 010802) has no driver
EFIPCI 0000:20:03.3 (1022:1483 class 060400) has no driver
EFIPCI 0000:23:00.0 (1e0f:0007 class 010802) has no driver
EFIPCI 0000:20:03.4 (1022:1483 class 060400) has no driver
EFIPCI 0000:24:00.0 (1e0f:0007 class 010802) has no driver
EFIPCI 0000:20:04.0 (1022:1482 class 060000) has no driver
EFIPCI 0000:20:05.0 (1022:1482 class 060000) has no driver
EFIPCI 0000:20:07.0 (1022:1482 class 060000) has no driver
EFIPCI 0000:20:07.1 (1022:1484 class 060400) has no driver
EFIPCI 0000:25:00.0 (1022:148a class 130000) has no driver
EFIPCI 0000:25:00.2 (1022:1498 class 108000) has no driver
EFIPCI 0000:20:08.0 (1022:1482 class 060000) has no driver
EFIPCI 0000:20:08.1 (1022:1484 class 060400) has no driver
EFIPCI 0000:26:00.0 (1022:1485 class 130000) has no driver
EFIPCI 0000:26:00.1 (1022:1486 class 108000) has no driver
EFIPCI 0000:26:00.2 (1022:1498 class 108000) has no driver
EFIPCI 0000:26:00.3 (1022:148c class 0c0330) has no driver
EFIPCI 0000:40:00.0 (1022:1480 class 060000) has no driver
EFIPCI 0000:40:00.2 (1022:164f class 080600) has no driver
EFIPCI 0000:40:01.0 (1022:1482 class 060000) has no driver
EFIPCI 0000:40:01.1 (1022:1483 class 060400) has no driver
EFIPCI 0000:41:00.0 (1a03:1150 class 060400) has no driver
EFIPCI 0000:42:00.0 (1a03:2000 class 030000) has no driver
EFIPCI 0000:40:01.3 (1022:1483 class 060400) has no driver
EFIPCI 0000:43:00.0 (1b21:1142 class 0c0330) has no driver
EFIPCI 0000:40:01.4 (1022:1483 class 060400) has no driver
[... 170 lines cut ...]
EFIPCI 0000:c0:08.2 (1022:1484 class 060400) has no driver
EFIPCI 0000:c4:00.0 (1022:7901 class 010601) has no driver
EFIPCI 0000:e0:00.0 (1022:1480 class 060000) has no driver
EFIPCI 0000:e0:00.2 (1022:164f class 080600) has no driver
EFIPCI 0000:e0:01.0 (1022:1482 class 060000) has no driver
EFIPCI 0000:e0:02.0 (1022:1482 class 060000) has no driver
EFIPCI 0000:e0:03.0 (1022:1482 class 060000) has no driver
EFIPCI 0000:e0:03.1 (1022:1483 class 060400) has no driver
EFIPCI 0000:e0:03.2 (1022:1483 class 060400) has no driver
EFIPCI 0000:e0:03.3 (1022:1483 class 060400) has no driver
EFIPCI 0000:e0:03.4 (1022:1483 class 060400) has no driver
EFIPCI 0000:e0:04.0 (1022:1482 class 060000) has no driver
EFIPCI 0000:e0:05.0 (1022:1482 class 060000) has no driver
EFIPCI 0000:e0:07.0 (1022:1482 class 060000) has no driver
EFIPCI 0000:e0:07.1 (1022:1484 class 060400) has no driver
EFIPCI 0000:e5:00.0 (1022:148a class 130000) has no driver
EFIPCI 0000:e5:00.2 (1022:1498 class 108000) has no driver
EFIPCI 0000:e0:08.0 (1022:1482 class 060000) has no driver
EFIPCI 0000:e0:08.1 (1022:1484 class 060400) has no driver
EFIPCI 0000:e6:00.0 (1022:1485 class 130000) has no driver
EFIPCI 0000:e6:00.2 (1022:1498 class 108000) has no driver
ok
iPXE 1.21.1+ (ga8012) -- Open Source Network Boot Firmware -- https://ipxe.org
Features: DNS HTTP iSCSI TFTP SRP AoE EFI Menu
```

stappersg, 2022-09-06:
Please reduce the amount of lines with `has no driver`.

alexdepalex, 2022-09-20:
Same here. We're using `master` (I know), and it worked till a couple of weeks ago. So some commits in between must have broken it.

mcb30, 2022-09-20:
> Same here. We're using `master` (I know), and it worked till a couple of weeks ago. So some commits in between must have broken it.

In that case, https://ipxe.org/howto/bisect may be useful.

alexdepalex, 2022-09-20:
Commit a202de385dcb9c15a1b2b3fd339abbe721ed5c86 seems to break things for me.

byvns, 2022-10-03:
Hi We have the same problem with 

	PCI_ROM ( 0x8086, 0x37d1, "x722-1gt", "X722 1GBASE-T", 0 ),

while compiled with DEBUG=intelxl:3,pcimsix:3 we found this:

`>>Media Present
Downloading NBP file...
NBP file dotmioaded successfully.
IPXE init iaiising devices...
INTEL XL Ox5de6b890 PF 7 using port 3 queues [Ox07ff-Ox07ff]
MSI-X OxSde6b8dO has 129 vectors for 0000: 1b:00.O
MSI—X OxSde6b8dO table could not find BAR3
INTEL XL OxSde6bB90 could not enable MSI—X: No such file or directory (https://ipxe.org/2dcc408e)
INTEL XL OxSde6b830 PF 7 using port 3 queues [Ox07ff-Ox07ff]
MSI—X OxSde6b8dO has 123 vectors for 0000: 1b:OO.1
MSI—X OxSde6b8dO table could not find BAR3
INTEL XL OxSde6b830 could not enable MSI—X: No such file or directory (https://ipxe.org/2dcc408e)
INTEL XL OxSde6b830 PF 7 using port 3 queues [Ox07ff—Ox07ff]
MSI—X OxSde6b8dO has 129 vectors for 0000: 1b:OO.2
MSI-X OxSde6b8dO table could not find BAR3
INTEL XL OxSde6b890 could not enable MSI—X: No such file or directory (https://ipxe.org/2dcc408e)
INTEL XL OxSde6b830 PF 7 using port 3 queues [Ox07ff-Ox07ff]
MSI—X OxSde6b8dO has 123 vectors for 0000: 1b:OO.3
MSI—X OxSde6b8dO table could not find BAR3
INTEL XL OxSde6b830 COU1d not enable MSI—X: No such file or directory (https://ipxe.org/2dcc408e)`

sasharozenson, 2022-11-09:
> Commit [a202de3](https://github.com/ipxe/ipxe/commit/a202de385dcb9c15a1b2b3fd339abbe721ed5c86) seems to break things for me.

Same here.

mcb30, 2022-11-13:
> Commit [a202de3](https://github.com/ipxe/ipxe/commit/a202de385dcb9c15a1b2b3fd339abbe721ed5c86) seems to break things for me.

@alexdepalex Thanks for the bisection result.  Should be fixed by #801

closed 2022-11-13 by mcb30

alexdepalex, 2023-02-06:
Just a quick question to the poepl that are following this before I open a new issue. Did some of you try the fixed code? We're using master, with which the adapters get discovered, but we can't get packets flowing over the network. We're booting into ipxe, configuring networking as we previously did on the commit that did work, but we can't ping the gateway or our dns servers.

Was somebody able to get this to work with X710 after the fix was introduced?

sasharozenson, 2023-02-06:
I can confirm, the newest commit fixed the problem for us.

alexdepalex, 2023-02-06:
Thanks. I'll give that a shot.

PR #801 (2022-11-13): Backup and restore standard config space across PCIe FLR
Fixes: #731