ipxe/ipxe #748: Compile on "aarch64-unknown-linux-gnu-gcc 12.2.0"

stappersg, 2022-09-16:
The compiler in use is "aarch64-unknown-linux-gnu-gcc (Gentoo 12.2.0 p1) 12.2.0" and to compile with no error three files had to be patched to suppress some errors that I suspect it's caused compiler version.

These files are:

1. src/arch/arm/include/ipxe/arm_io.h (#pragma GCC diagnostic ignored "-Warray-bounds")
2. src/crypto/deflate.c (#pragma GCC diagnostic ignored "-Wdangling-pointer")
3. src/net/fc.c (#pragma GCC diagnostic ignored "-Warray-bounds")

Hopefully the patch doesn't cause any problem.

_Originally posted by @jorgeventura in https://github.com/ipxe/ipxe/discussions/745#discussioncomment-3664142_

NiKiZe, 2022-09-19:
Closing as duplicate of #620

closed 2022-09-19 by NiKiZe

stappersg, 2022-09-19:
> Closing as duplicate of #620

Merging the patches provide in #620 will prevent future developer time being wasted on _Hey, my recent GCC gives me compile errors_.

@jorgventura your focus is most likely currently on #744. So I didn't asked for a merge request about _compile for aarch64 GCC 12_. Neither did I ask if #620 also fixes _aarch64_ builds.  I did create an issue with
```text
    src/arch/arm/include/ipxe/arm_io.h (#pragma GCC diagnostic ignored "-Warray-bounds")
```

I'm following the progress of #745 and when that is completed, I'll ask you if #620 also fixes building on _aarch64_. Because you has access to an _aarch64_ system.

stappersg, 2022-09-23:
I feel sorry for @jorgeventura that he had to deal with a known build error.

For the good of the iPXE project would it help how _Compile on "aarch64-unknow-linux-gnu-gcc 12.2.0"_ was fixed.  Documenting it is a small task for Jorge  and a huge benefit for future iPXE users who have to take the hurdle.

Extra reason for preventing loss of knowledge is that _aarch64_ is some what rare.

jorgeventura, 2022-09-23:
Not a problem at all. Part of the problem was because I tried to use
infrastructure without know how it works. My fault.

Anyway I learned a lot, believe me.

stappersg, 2022-09-24:
> Anyway I learned a lot, believe me.

Yes, it is the reason that I am begging for sharing notes.    :smiley: