ipxe/ipxe #749: console command --keep argument is misleading

DimanKuskov, 2022-09-19:
**Problem:**
Description of the `console` command states that:
```
The background picture will be discarded after configuring the console unless the --keep option is specified.
...
You can use the --keep option if you intend to reuse the same background image in a future console command, to avoid downloading the image a second time.
```

However, if you use a full url for an image, like `http://boot.ipxe.org/ipxe.png`, the image is stored in memory using it's basename, like `ipxe.png`: 
https://github.com/ipxe/ipxe/blob/5d22307c4161dde453d50e8dc7bef8b3a2f6c9b3/src/core/image.c#L125 
So in order to reuse a short name must be specified:
```
console --picture http://boot.ipxe.org/ipxe.png --keep #This will download image and keep it
console --picture http://boot.ipxe.org/ipxe.png #This will re-download image
console --picture ipxe.png #This will use kept image
```

**Possible solutions:**

1. Update `--keep` argument description and add information that image must be reused by its base name.
2. Update `imgacquire` to search for image by basename, not uri
3. Make `find_image` search both `image->name` and `image->uri`.

stappersg, 2022-09-19:
It is https://ipxe.org/cmd/console where solution option 1, _"Update `--keep` description, add information that image must be reused by its base name"_ can be applied.  @mcb30 please enable me to do that.

NiKiZe, 2022-09-19:
try imgstat after console --picture without --keep
Then try it with keep. I think you will find it does exactly what it says it does.

But the issue might exist for imgfetch as well?

mcb30, 2022-10-27:
@DimanKuskov Thank you for the detailed description.

>     1. Update `--keep` argument description and add information that image must be reused by its base name.

This feels user-unfriendly, and I'd prefer to avoid it if possible.

>     2. Update `imgacquire` to search for image by basename, not uri

This is not viable since it would create the surprising situation that e.g. an attempt to download `http://server1/path/to/script.ipxe` followed by an attempt to download `http://server2/unrelated/path/to/different/script.ipxe` would result in the same file being downloaded twice.

>     3. Make `find_image` search both `image->name` and `image->uri`.

This seems as though it may be the cleanest approach.  There are some subtleties to consider, though.  iPXE will resolve the URI to an absolute URI before storing it in `image->uri`.  We could plausibly update `find_image()` to attempt to parse its argument to a URI, then resolve the URI to an absolute URI, then compare the absolute URI against `image->uri`.  This would be a relatively large amount of binary code for a relatively obscure feature (not least because there is no code at present to compare URIs for equality).

From a user expectations perspective, it may be simpler to treat the `name_uri` argument to `imgacquire` as being the first match for:

1. the name of an existing image, or
2. the original `imgacquire` argument of an existing image, or
3. the URI string of a new image to download

This pattern would probably match most user expectations: an attempt to download an image using the exact same URI string would use the already-downloaded version (if present).  It would be arguably incorrect in the case of a relative URI if the current working URI has changed between first and second download attempts.

One possible failure mode of any approach would be a script that expects to be able to perform a fresh download by specifying a URI, even if the URI has already been previously downloaded.  Such scripts should arguably be using `imgfree`, but there is still the potential for breaking existing scripts, which we try hard to avoid.

Comments welcome!

NiKiZe, 2022-10-27:
Could we recommend that any use of console is used with imgfetch to verify that the correct image is used. (And deprecate keep?)

Of course we should avoid breaking changes, but maybe it is fine since it is "only" background images? As long as it isn't causing errors? And also maybe an announcement on mailinglist about this change?

stappersg, 2022-10-28:
> > 1. Update `--keep` argument description and add information that image must be reused by its base name.
> 
> 
> This feels user-unfriendly, and I'd prefer to avoid it if possible.

On the other hand:
Having documentation describing what is implemented, **is** user-friendly.