ipxe/ipxe #760: Possible leak from strndup: Missing Release of Memory after Effective Lifetime

darnuria, 2022-09-28:
How to reproduce?
compile with a gcc (version 12 or before) with `-fanalyzer` [gcc-doc](https://gcc.gnu.org/onlinedocs/gcc/Static-Analyzer-Options.html)
```

```
`export NO_WERROR=1; CLAGS="-fanalyzer"; make`

[CWE-401: Missing Release of Memory after Effective Lifetime](https://cwe.mitre.org/data/definitions/401.html)

```
  [BUILD] bin/string.o
core/string.c: In function ‘strndup’:
core/string.c:401:16: warning: leak of ‘strndup(src, 4294967295)’ [CWE-401] [-Wanalyzer-malloc-leak]
  401 |         return dup;
      |                ^~~
  ‘strdup’: events 1-3
    |
    |  380 | char * strdup ( const char *src ) {
    |      |        ^~~~~~
    |      |        |
    |      |        (1) entry to ‘strdup’
    |  381 | 
    |  382 |         return strndup ( src, ~( ( size_t ) 0 ) );
    |      |                ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    |      |                |
    |      |                (2) allocated here
    |      |                (3) calling ‘strndup’ from ‘strdup’
    |
    +--> ‘strndup’: events 4-5
           |
           |  392 | char * strndup ( const char *src, size_t max ) {
           |      |        ^~~~~~~
           |      |        |
           |      |        (4) entry to ‘strndup’
           |  393 |         size_t len = strnlen ( src, max );
           |      |                      ~~~~~~~~~~~~~~~~~~~~
           |      |                      |
           |      |                      (5) calling ‘strnlen’ from ‘strndup’
           |
           +--> ‘strnlen’: events 6-8
                  |
                  |  255 | size_t strnlen ( const char *src, size_t max ) {
                  |      |        ^~~~~~~
                  |      |        |
                  |      |        (6) entry to ‘strnlen’
                  |......
                  |  259 |         while ( max-- && *(src_bytes++) )
                  |      |                 ~~~~~~~~~~~~~~~~~~~~~~~
                  |      |                       |             |
                  |      |                       |             (8) ...to here
                  |      |                       (7) following ‘true’ branch (when ‘max != 0’)...
                  |
           <------+
           |
         ‘strndup’: events 9-12
           |
           |  393 |         size_t len = strnlen ( src, max );
           |      |                      ^~~~~~~~~~~~~~~~~~~~
           |      |                      |
           |      |                      (9) returning to ‘strndup’ from ‘strnlen’
           |......
           |  397 |         if ( dup ) {
           |      |            ~          
           |      |            |
           |      |            (10) following ‘false’ branch (when ‘dup’ is NULL)...
           |......
           |  401 |         return dup;
           |      |         ~~~~~~ ~~~    
           |      |         |      |
           |      |         |      (12) ‘strndup(src, 4294967295)’ leaks here; was allocated at (2)
           |      |         (11) ...to here
           |

```

mcb30, 2022-10-27:
This is very obviously a false positive: the whole point of `strndup()` is to return a newly allocated string.

closed 2022-10-27 by mcb30

stappersg, 2022-10-27:
> obviously a false positive

Could, should?, have closed as  _not planned_.  (Now it is closed as _completed_)

And we, _iPXE project as a whole_, should find a way how we deal with disagreement of compilers.  Example given #620.