ipxe/ipxe #762: ipxe freezes on GA- motherboard after been loaded by nic efi PXE rom.

hradec, 2022-09-28:
I've built ipxe.efi with `DEBUG=efi_driver,snpnet,pci,efi_pxe,efi_pci,ipv4`, and got this log when trying to netboot ipxe.efi:
[image]
I've added a `sleep(5)` at line 191 of src/interface/efi/efi_driver.c, so it would pause and I could capture a screenshot, since it stays looping the "refusing to start" message indefinitely.

NiKiZe, 2022-09-28:
iPXE has a number of workarounds for broken firmware, but some firmware bugs is not reasonable to deal with.

Ensure you have the latest firmware, if it still doesn't work then contact Gigabyte.

Grab the pciid, build ipxe for that NIC (if we have native drivers) without NII driver. But most likely the release of firmware nio driver is the issue.

hradec, 2022-09-28:
> Ensure you have the latest firmware, if it still doesn't work then contact Gigabyte.

If by firmware, you mean the motherboard uefi bios, it is running the latest version (F5 from 2013/11/14). There's nothing newer than F5 at the [Gigabyte website](https://www.gigabyte.com/Motherboard/GA-B75N-rev-10/support#support-dl-bios). 

> Grab the pciid, build ipxe for that NIC (if we have native drivers) without NII driver. But most likely the release of firmware nio driver is the issue.

I need to look how to do that. Not sure how to build ipxe with native drivers for a given pciid. 

> But most likely the release of firmware nio driver is the issue.
Do you think the problem may be caused by loading ipxe from the pxe boot rom? if I boot ipxe from a usb efi drive  it may be able to start the driver correctly? 

Thanks a lot for the quick feedback.

hradec, 2022-09-29:
A quick update. The first screen was boot using `UEFI Only` boot mode. 
I've changed it to `UEFI and Legacy` boot mode, which shows up the network card 2 as boot option: the uefi network card and a bios legacy network card. 
[image]
I've selected the UEFI Realtek network card as first boot device, and now the iPXE log is showing something a little different: 
[image]
The driver iPXE is trying to load now is `NII`, instead of `NIIONLY`.
Also, there's a extra EFIPCI 02:00 that was identified with rtl8168. 
I'm guessing with this mix of UEFI and Legacy boot modes, iPXE is able to detect the actual network card correctly, but it's still failing to apply the driver for the EFIDRV?

hradec, 2022-09-29:
just for reference, I've also captured a screenshot of the debug log without the `sleep()` I added, and I've noticed error log messages for the same `EFIDRV` device with 2 different drivers - `NII` and `SNP`: 
[image]

stappersg, 2022-09-29:
What I would like to known (concerning this issue :-) :
* The amount of ethernet connectors ([RJ45 connectors](https://en.wikipedia.org/wiki/Modular_connector#8P8C))
* The _pci ids_ of the ethernet interfaces

For getting _pci ids_ of ethernet interfaces is booting into the operating system the best path to walk. The _how to_ and _what to expect_ on a Linux system:
```text
$ lspci -n | awk '$2 ~ /0200:/ { print $0 }'
08:00.0 0200: 1969:1083 (rev c0)
$
```
(that is on a Linux laptop with one ethernet interface (I can't remember if I have used _iPXE_ on that laptop))

In case only graphical screenshots are available from the operating system, add the text version of the _pci id_.

GrayGuide, 2026-03-10:
Dear all,
     Pls give a help！！I found that the latest iPXE open source in https://github.com/ipxe/ipxe has no driver for 8168  network card,  maybe the git address is not correct. But where can I find one supporting it?  Pls give a help ...

NiKiZe, 2026-04-07:
Closing as stale, Happy to have this reopened if new details comes to light.
But in summary, this is a realtek card which is not uncommon to have issues with.
If someone else has similar issues it is a good idea to put shell.efi, ipxe.efi and the latest realtek efi drivers on disk and boot into the shell. try starting ipxe from there and verify any hang or issue to persist, then try again, but before starting ipxe, unload the old efi network driver, and load the new one from realtek, and then start ipxe and check if that works.

closed 2026-04-07 by NiKiZe