ipxe/ipxe #767: Invalid HTTP Range header using sanboot

Kebab11noel, 2022-10-07:
I tried to sanboot a Zorin iso image, but it fails because an invalid range header is sent. Below is an image of the intercepted web traffic.
[image]

NiKiZe, 2022-10-07:
iPXE 081b3eefc48953795122c4b1a38430498e2bb2b6 which at this time is the latest.
What is the total size of file? I think there was a fix for range requests of large files, but it might be that it was not merged.

However booting Isos with sanboot over http seldom works. Once the kernel boots the iso will not be found. You are better of extracting relevant files and boot the correct kernel and initrds from iPXE.

Kebab11noel, 2022-10-07:
It's 2 587 770 880 bytes

Kebab11noel, 2022-10-07:
Actually I found a [mailing list](https://lists.ipxe.org/pipermail/ipxe-devel/2017-May/005628.html) about this issue and after building with `make bin-x86_64-pcbios/undionly.kpxe EMBED=.ipxe` (instead of `make bin/undionly.kpxe EMBED=.ipxe`) it works.