ipxe/ipxe #775: Issues when try generate embed iso (iso with custom script).

Mateushrr, 2022-10-15:
Hi, when i try execute make command to create the iso (make bin/ipxe.iso EMBED=myscript.ipxe), i'm receving the message: "util/genfsimg: could not find isolinux.bin".

How to fix it? i got the syslinux installed on my ubuntu 22.04.
Thanks for help and for maintain great this software.

My comand order to create embed iso.

$ git clone https://github.com/ipxe/ipxe.git
$ cd ipxe/src
$ make bin/ipxe.iso EMBED=myscript.ipxe (this is failing)

stappersg, 2022-10-15:
At https://ipxe.org/download is below the line
> You will need to have at least the following packages installed in order to build iPXE: 

a list of packages. How much of that list is installed?

My actual question: What needs to be updated at https://ipxe.org/download ?

NiKiZe, 2022-10-15:
Do you have isolinux package installed on your system? (Required if making Isos, and documented on the download page)

If you do have it installed, please tell us the path to the file, and which system it is.

Mateushrr, 2022-10-15:
> At https://ipxe.org/download is below the line
> 
> > You will need to have at least the following packages installed in order to build iPXE:
> 
> a list of packages. How much of that list is installed?
> 
> My actual question: What needs to be updated at https://ipxe.org/download ?

Thx for response. Its all packages installed.

Mateushrr, 2022-10-15:
> Do you have isolinux package installed on your system? (Required if making Isos, and documented on the download page)
> 
> If you do have it installed, please tell us the path to the file, and which system it is.

Thx for response too. Hm, checking now i realized i was missing the package isolinux, so I installed using the command: apt install isolinux. Now its working and got done the embed iso creation!!! thx all for reply.

closed 2022-10-15 by Mateushrr

stappersg, 2022-10-15:
@Mateushrr or any other user of Ubuntu 22.04, please provide the output of
```text
dpkg -L syslinux | grep isolinux.bin
```
and
```text
dpkg -L isolinux | grep isolinux.bin
```

Right now I have the idea that the `syslinux` package doesn't contain anymore a `isolinux.bin`,
that https://ipxe.org/download should be updated. (Euh yes, I think this issue was closed too early.)