ipxe/ipxe #777: vga=ask/normal/ext not working if not last argument in KCL

srett, 2022-10-19:
In
https://github.com/ipxe/ipxe/blob/6b2c94d3a7d93a8fc47fcb0b895477d4dafca5f0/src/arch/x86/image/bzimage.c#L253-L263
the check for "special values" to `vga=` uses `strcmp()`, which would fail if the argument is not the last one in the kernel command line.

The easy fix would be to use `strncmp(vga, "ask", 3)` instead, but it would also trigger on things like `vga=askiicode`, which imo would be acceptable, but obviously not correct.

stappersg, 2022-10-19:
On Wed, Oct 19, 2022 at 08:06:10AM -0700, Simon Rettberg wrote:
> In
> https://github.com/ipxe/ipxe/blob/6b2c94d3a7d93a8fc47fcb0b895477d4dafca5f0/src/arch/x86/image/bzimage.c#L253-L263
> the check for "special values" to `vga=` uses `strcmp()`, which would
> fail if the [...]
 

Thanks for reporting.

Please consider to convert the observation into a merge request.

Groeten
Geert Stappers
-- 
Silence is hard to parse

srett, 2022-10-20:
> Please consider to convert the observation into a merge request.

Thought I should be waiting for @mcb30 on the `strncmp` solution with the aforementioned gotcha. Otherwise some more fiddling would be required, at least I didn't immediately come up with a clean solution. But yes, otherwise that would be a pretty simple PR I can create.

stappersg, 2022-10-22:
> But yes, otherwise that would be a pretty simple PR I can create.

_Just do it_

> Thought I should be waiting for @mcb30

That waiting can be done in a _merge request_.

PR #785 (2022-10-27): [bzimage] Fix parsing of "vga=..." when not at end of command line
bzimage_parse_cmdline() uses strcmp() to identify the named "vga=..." kernel command line option values, which will give a false negative if the option is not last on the command line.

Fix by temporarily changing the relevant command line separator (if any) to a NUL terminator.

Fixes: #777 

Debugged-by: Simon Rettberg <simon.rettberg@rz.uni-freiburg.de>
closed 2022-10-27 by mcb30