ipxe/ipxe #780: Tell about  ipxe.efi

stappersg, 2022-10-24:
Current output of `make` after a succesfull `make`:
```text
===========================================================

To create a bootable floppy, type
    cat bin/ipxe.dsk > /dev/fd0
where /dev/fd0 is your floppy drive.  This will erase any
data already on the disk.

To create a bootable USB key, type
    cat bin/ipxe.usb > /dev/sdX
where /dev/sdX is your USB key, and is *not* a real hard
disk on your system.  This will erase any data already on
the USB key.

To create a bootable CD-ROM, burn the ISO image 
bin/ipxe.iso to a blank CD-ROM.

These images contain drivers for all supported cards.  You
can build more customised images, and ROM images, using
    make bin/<rom-name>.<output-format>

===========================================================
```

There should be something about `ipxe.efi`
and that it can be build by `make bin-x86_64-efi/ipxe.efi`

stappersg, 2023-01-24:
And `snponly.efi` can be build by [`make bin-x86_64-efi/snponly.efi`](https://github.com/ipxe/ipxe/issues/853#issuecomment-1401606848).

mcb30, 2023-01-24:
There is a potentially endless list of things that "should" be included in the output of a plain `make`.  The existing online documentation seems a better home for this than the terminal output from running `make`.

closed 2023-01-24 by mcb30