ipxe/ipxe #782: IMAGE_BZIMAGE cause compilation failure

pietrushnic, 2022-10-26:
Issue discovered while testing #300 

OS: Debian 11
SHA: 8fce26730c4df7a9792bb144c75c2c5b998c91af

Reproduction steps:

```bash
git clone https://github.com/ipxe/ipxe.git

cd ipxe/src
sed -i 's|.*DOWNLOAD_PROTO_HTTPS|#define DOWNLOAD_PROTO_HTTPS|g' config/general.h
sed -i 's|//#define\s*IMAGE_BZIMAGE.*|#define IMAGE_BZIMAGE|' config/general.h
sed -i 's|//#define\s*IMAGE_GZIP.*|#define IMAGE_GZIP|' config/general.h
sed -i 's|//#define\s*NSLOOKUP_CMD.*|#define NSLOOKUP_CMD|' config/general.h
sed -i 's|//#define\s*IMAGE_TRUST_CMD.*|#define IMAGE_TRUST_CMD|' config/general.h
sed -i 's|//#define\s*PING_CMD.*|#define PING_CMD|' config/general.h

make bin-x86_64-efi/ipxe.efi
```

As result I get compilation error:

```bash
ar: creating bin-x86_64-efi/blib.a
  [HOSTCC] util/elf2efi64
  [VERSION] bin-x86_64-efi/version.ipxe.efi.o
  [LD] bin-x86_64-efi/ipxe.efi.tmp
bin-x86_64-efi/blib.a(librm.o): in function `pm_esp':
(.data.pm_esp+0x0): relocation truncated to fit: R_X86_64_32 against symbol `_estack' defined in .stack section in bin-x86_64-efi/blib.a(stack.o)
bin-x86_64-efi/blib.a(librm.o): in function `real_to_prot':
(.text16.real_to_prot+0x1): relocation truncated to fit: R_X86_64_PC16 against symbol `enable_a20' defined in .text16.early section in bin-x86_64-efi/blib.a(liba20.o)
(.text16.real_to_prot+0x8): relocation truncated to fit: R_X86_64_16 against symbol `rm_ds' defined in .text16.data.rm_ds section in bin-x86_64-efi/blib.a(librm.o)
(.text16.real_to_prot+0x22): relocation truncated to fit: R_X86_64_16 against `.bss16.rm_virt_addrs'
(.text16.real_to_prot+0x2e): relocation truncated to fit: R_X86_64_16 against `.bss16.rm_virt_addrs'
(.text16.real_to_prot+0x33): relocation truncated to fit: R_X86_64_16 against `.bss16.rm_virt_addrs'
(.text16.real_to_prot+0x39): relocation truncated to fit: R_X86_64_16 against `.data16.gdt'
(.text16.real_to_prot+0x58): relocation truncated to fit: R_X86_64_32 against `.text.real_to_prot'
bin-x86_64-efi/blib.a(librm.o): in function `r2p_pmode':
(.text.real_to_prot+0x10): relocation truncated to fit: R_X86_64_32 against `.data.pm_esp'
(.text.real_to_prot+0x17): relocation truncated to fit: R_X86_64_32 against symbol `idtr32' defined in .data section in bin-x86_64-efi/blib.a(librm_mgmt.o)
(.text.real_to_prot+0x1f): additional relocation overflows omitted from the output
ld: bin-x86_64-efi/blib.a(hidemem.o): in function `hide_region':
/tmp/ipxe/src/arch/x86/interface/pcbios/hidemem.c:97: undefined reference to `_textdata'
ld: /tmp/ipxe/src/arch/x86/interface/pcbios/hidemem.c:98: undefined reference to `_etextdata'
ld: bin-x86_64-efi/blib.a(hidemem.o): in function `hide_etherboot':
/tmp/ipxe/src/arch/x86/interface/pcbios/hidemem.c:158: undefined reference to `_textdata'
ld: /tmp/ipxe/src/arch/x86/interface/pcbios/hidemem.c:172: undefined reference to `_data16_memsz'
ld: /tmp/ipxe/src/arch/x86/interface/pcbios/hidemem.c:171: undefined reference to `_text16_memsz'
ld: bin-x86_64-efi/ipxe.efi.tmp: hidden symbol `_textdata' isn't defined
ld: final link failed: bad value
make: *** [Makefile.housekeeping:1220: bin-x86_64-efi/ipxe.efi.tmp] Error 1
rm bin-x86_64-efi/version.ipxe.efi.o
/tmp
cp: cannot stat 'ipxe/src/bin-x86_64-efi/ipxe.efi': No such file or directory
```

NiKiZe, 2022-10-26:
BZIMAGE can not be enabled in efi builds.

closed 2022-10-26 by NiKiZe

pietrushnic, 2022-10-26:
@NiKiZe can you elaborate why? Are you consider it not valid use case?

NiKiZe, 2022-10-26:
https://github.com/ipxe/ipxe/pull/300#issuecomment-803475090

In efi mode iPXE will only tell firmware to execute any binary. It is up to the usual firmware to perform validation and execution of the binary. (Secure boot)
Since the Linux kernel has proper efi support, there is no reason for iPXE to support anything other than the efi booting mechanism.
So correct, not a valid usecase. Even so without checking for signing if secure boot is enabled it is an unacceptable usecase ;) and the complexy is really really hard to motivate. That is without the part of even booting/handover to BZIMAGE in efi mode.

stappersg, 2022-10-26:
This feels wrong.  That a _fails to build_ issue is closed as _"completed"_.

iPXE has build option  `EMBED=`, which makes it possible that the iPXE executable is carrier for an iPXE script. In a simular way should it be possible that iPXE is carrier for a BZIMAGE.

My hope is that the next person that has a use case (or is in need) for getting BZIMAGE in EFI environment opens a fresh issue and that that issue gets a threatment as #152: Mostly being ignored and a nice polite hint to come with a merge request.

pietrushnic, 2022-10-26:
@NiKiZe Secure Boot-based justification seem to come from Microsoft code review and signing iPXE [1] - at some point there was word document describing review in detail, but now we can't find it. Do you have link to it?

Assuming UEFI compatible implementation is the only way to go is far fetched and it may happen to be wrong in long run. I guess Ventoy popularity proves what headaches this kind of approach cause and that community need and value solutions, which can get rid of those. But I understand the message - anything what can harm "Secure Boot" would be rejected.

I understand your argument about complexity, it was used many times, but Microsoft itself heavily use advanced hardware security features like D-RTM [2]. Chain of trust based on root of trust for verification (like "Secure Boot" family)  are not simpler and maintaining trust in those solutions is very hard e.g. BootHole [3] - I can also tell from BIOS developer perspective. Please note that in #300 we trying to enable the same mechanisms that Microsoft already use but in open-source way. There is very long discussion on LKML justifying need for D-RTM [4], when patches will be merged upstream there would be at least two reasons to consider accepting open-source D-RTM solution like TrenchBoot.

But this issue is about something different. If you not allow given configuration you should tell it in normal way with clear message to user (I really like links to error messages explanations that iPXE show runtime). User/developer should not wait whole compilation process and then ask on GH to figure out that while building efi target `IMAGE_BZIMAGE` is not allowed. This is waste of time.

Coworkers from Dasharo Community explained to me that bzImage needs 16bit what cannot fly with UEFI compatible implementation and I'm completely fine with that. From my side let's treat this bug report as developer experience improvement suggestion not as request for enabling ability to boot bzImage in UEFI compatible environment.

[1]: https://techcommunity.microsoft.com/t5/hardware-dev-center/ipxe-security-assurance-review/ba-p/1062943
[2]: https://www.microsoft.com/en-us/security/blog/2020/09/01/force-firmware-code-to-be-measured-and-attested-by-secure-launch-on-windows-10/
[3]: https://eclypsium.com/2021/04/14/boothole-how-it-started-how-its-going/
[4]: https://lkml.org/lkml/2022/2/18/699

NiKiZe, 2022-10-26:
Any PR that adds support for booting any binary in efi mode will need to have proper secure boot considerations addressed to have any chance of being accepted.

What I can recommend here instead is to create a separate efi binary that acts as a compability layer, maybe as shim.

I hear you on the build error part. I have raised similar issues myself, but the answer we end up with is that we can't guard for every type of bad configuration that users try to compile. If you modify build configurations you should do it one at a time and only if you need to. And enabling PCBIOS features in efi builds, or EFI features in PCBIOS builds is of course not possible. I tried to find this but failed to find that discussion atm.

So in principle I agree that build errors could be better. But it doesn't scale in a good way.
Another thing is to always check for existing (including closed) issues before creating new ones. ;)

#340 #577