ipxe/ipxe #786: Where should DHE be added to https://ipxe.org/crypto

stappersg, 2022-10-27:
> > Someone update the docs here to list DHE. Tq! https://ipxe.org/crypto

>  Part of current source of that wiki page.
```
===== Supported configurations =====

^ Protocol versions        | TLSv1.0 TLSv1.1 TLSv1.2    |
^ Public-key algorithms    | RSA                        |
^ Block cipher algorithms  | AES-128-CBC AES-256-CBC    |
^ Hash algorithms          | MD5 SHA-1 SHA-224 SHA-256 SHA-384 SHA-512 SHA-512/224 SHA-512/256  |
```

> Where should DHE be added?

New line, "Key exchange algorithms"?

_Originally posted by @joernheissler in https://github.com/ipxe/ipxe/issues/246#issuecomment-1294069683_

jerrywoo96, 2022-10-28:
Also if possible, use the official full IANA names for the cipher suites.
https://wiki.mozilla.org/Security/Cipher_Suites

jerrywoo96, 2022-10-28:
@NiKiZe

do help edit the documentation tq!

mcb30, 2022-11-13:
Already updated

closed 2022-11-13 by mcb30