ipxe/ipxe #787: support TLS 1.3, modern cipher, and ECDSA certificates

leeyc0, 2022-10-27:
(continue from #246)

Cryptography security policy is now more stringent and server may no longer support older and less secure ciphers. CBC ciphers are being faded out in favour to AEAD ciphers like GCM.

To be exact, iPXE need to support these cryptography algorithms:

- TLS 1.3 and related ciphers (these ciphers needed to be supported in TLS 1.2 too)
  - cipher suites
    - TLS_AES_128_GCM_SHA256 (RFC 8446: MUST)
    - TLS_AES_256_GCM_SHA384 (RFC 8446: SHOULD, but I believe this cipher can be supported without any extra memory space if GCM AEAD is implemented for TLS_AES_128_GCM_SHA256)
    - TLS_CHACHA20_POLY1305_SHA256 (RFC 8446: SHOULD. Since this is a new cipher that requires extra memory space, if space is a concern this may be omitted.)
  - ECDHE key exchange
    - NIST P-256  (RFC 8446: MUST)
    - x25519 (RFC 8446: SHOULD, but this curve have superior properties compared to P-256)
  - ECDSA certificates
    - While RFC 8446 only specifies NIST P-256 is a MUST, [Mozilla Root Store Policy](https://www.mozilla.org/en-US/about/governance/policies/security-group/certs/policy/) (only) accepts P-256 or P-384.
    - Not sure if P-521 or EdDSA certificates exists in private environments or not. (As a side note, ed25519 certificates uses SHA-512, and ed448 certificates uses SHAKE256 as hashing algorithm.)

(Refer to rfc2119 for interpretation of MUST and SHOULD)

Since TLS 1.2 is pretty much mandatory now, which practically means ECDHE key exchange is always supported, I don't think integer DH key exchange is needed, to have smaller boot image.

stappersg, 2022-10-28:
Link to ["There is a cost"](https://github.com/ipxe/ipxe/pull/574#issuecomment-1022380888) note.

jerrywoo96, 2022-10-28:
Here's a list of Cloudflare's recommendations of cipher suites. Hopefully the modern TLS 1.2 can be implemented.

TLS 1.3
TLS_AES_256_GCM_SHA384
TLS_CHACHA20_POLY1305_SHA256
TLS_AES_128_GCM_SHA256

TLS 1.2 (Modern)
TLS_ECDHE_ECDSA_WITH_CHACHA20_POLY1305_SHA256
TLS_ECDHE_RSA_WITH_CHACHA20_POLY1305_SHA256
TLS_ECDHE_RSA_WITH_AES_128_GCM_SHA256
TLS_ECDHE_ECDSA_WITH_AES_128_GCM_SHA256
TLS_ECDHE_RSA_WITH_AES_256_GCM_SHA384
TLS_ECDHE_ECDSA_WITH_AES_256_GCM_SHA384

https://developers.cloudflare.com/ssl/reference/cipher-suites/supported-cipher-suites/

mcb30, 2022-10-28:
DHE (non-ECDHE) is already implemented, and GCM is in progress.

There is no plan at this point to implement TLSv1.3, any of the elliptic curve algorithms, or any of the more obscure ciphers such as CHACHA20.

Marking this as closed unless someone is willing to fund the (very significant) development costs.

closed 2022-10-28 by mcb30

stappersg, 2022-10-28:
> Marking this as closed unless someone is willing to fund the (very significant) development costs.

Idea:

* Re-open this issue
* Express that this issue depends on funding for development and maintiance.

(Actual message of this comment: _Telling mankind that libre software does have a price._)

NiKiZe, 2024-01-31:
Some updates, see https://github.com/ipxe/ipxe/discussions/1135#discussioncomment-8319766