ipxe/ipxe #792: ISA driver(cs89x0.c) always in/out to IO-port 0x22A

peterliu134679, 2022-11-03:
Recently we discovered iPXE ISA NIC drivers always in/out to several IO-port address.
In our system, IO address 0x22A is map to a chip's control register.
### The Problem
ISA driver cs89x0.c would destroy port 0x22A due to following code:
```c
static int cs89x0_probe_addr ( isa_probe_addr_t ioaddr ) {
	/* if they give us an odd I/O address, then do ONE write to
	   the address port, to get it back to address zero, where we
	   expect to find the EISA signature word. */
	if (ioaddr & 1) {
		ioaddr &= ~1;
		if ((inw(ioaddr + ADD_PORT) & ADD_MASK) != ADD_SIG)
			return 0;
		outw(PP_ChipID, ioaddr + ADD_PORT);   /* when ioaddr is 0x221, this writes to port 0x22A */
	}
	
	if (inw(ioaddr + DATA_PORT) != CHIP_EISA_ID_SIG)
		return 0;

	return 1;
}
```

The array of cs89x0's  `ioaddr` are shown below. Since `ADD_PORT` is 0xA (ADD_MASK=ADD_SIG=0x3000), causing 0x22A being destoryed.
```c
static isa_probe_addr_t cs89x0_probe_addrs[] = { 
#ifndef EMBEDDED
	/* use "conservative" default values for autoprobing */
	0x300, 0x320, 0x340, 0x200, 0x220, 0x240,
	0x260, 0x280, 0x2a0, 0x2c0, 0x2e0,
	/* if that did not work, then be more aggressive */
	0x301, 0x321, 0x341, 0x201, 0x221, 0x241,
	0x261, 0x281, 0x2a1, 0x2c1, 0x2e1,
#else
	0x01000300,
#endif
};
```
### The Question(Proposal/Wishlist)

1. Since vast majority of modern server/notebook/desktop no longer uses ISA NIC, can iPXE disable ISA NIC support by default so that it has less chance to interfere with other hardware that uses IO-port? eg: in some config file has a `#undef ISA_BUS_SUPPORT` ?
2. As a possible workaround, does any of iPXE drivers (eg. ipxe.pxe, undionly.efi, snponly.efi, snp.efi) skip the loading of ISA port probing?

stappersg, 2022-11-03:
_Patches welcome_

NiKiZe, 2022-11-03:
The probing of ISA devices is causing issues for you on modern hardware?

Undionly and SNP variants only has the undi/SNP/NII driver, so there should be no ISA in there.

peterliu134679, 2022-11-04:
> The probing of ISA devices is causing issues for you on modern hardware?
Hi @NiKiZe , we use IO-port 0x200~0x240 for other purposes. 
Yes, I understand nothing is wrong with iPXE ISA NIC detection code so this is just a wishlist since I also wonder how many machines still uses ISA NIC? 

> Undionly and SNP variants only has the undi/SNP/NII driver, so there should be no ISA in there.
Will check this. 

Thanks you very much.

stappersg, 2023-02-10:
```patch
--- a/src/core/device.c
+++ b/src/core/device.c
@@ -83,6 +83,9 @@ static void probe_devices ( void ) {
        struct root_device *rootdev;
        int rc;
                     
+       // skip ISA device probing
+       return;
+                                                     
        for_each_table_entry ( rootdev, ROOT_DEVICES ) {
                list_add ( &rootdev->dev.siblings, &devices );                                            
                INIT_LIST_HEAD ( &rootdev->dev.children );                                                      
```

See also #877

stappersg, 2023-04-27:
For what it is worth: #888 made it worth to check again with recent version and update this issue.