ipxe/ipxe #793: libre versus gratuit

stappersg, 2022-11-03:
> Ok, the email address that Intel provides in the Linux kernel drivers does not exist. I have contacted support, but this is going to take a while I suppose.

Intel is unlikely to support iPXE directly, however for their new devices they **seem** to use paid services from @mcb30 

_Originally posted by @NiKiZe in https://github.com/ipxe/ipxe/issues/613#issuecomment-1301848778_

This issue for getting beyond rumours of _it seems a paid service_.  We know that iPXE is free software and that it has value.  But the english word _free_ is too often translated to the french word _gratuit_.  I would like to see that we,  iPXE stake holders, find a way to promote iPXE as _libre_ software.  That iPXE users are aware that _iPXE_ has a price.

NiKiZe, 2022-11-03:
Not an issue with iPXE code, closing.

Please refrain from spaming and keep the noise level down.

closed 2022-11-03 by NiKiZe

stappersg, 2022-11-03:
For the record, a snippet from the iPXE  IRC channel:
```text
13:26  * stappers is aiming in https://github.com/ipxe/ipxe/issues/793 for a good discussion
13:27 < stappers> and when the discussion is over, we move on and close the issue.
13:38 < stappers> And NiKiZe killed the discussion ...
```
for time reference:
```text
$ date ; date --rfc-2822
Thu Nov  3 13:57:19 CET 2022
Thu, 03 Nov 2022 13:57:19 +0100
$
```

Regarding _keep the noise level down_:
This issue start with
> however for their new devices they seem to use paid services

P.S.
Qualifing something that you don't like as spam, does not help getting beyond the situation that you don't like.  (Ans spamming is spelled with double m)