ipxe/ipxe #813: Cannot `set` variables finishing by `/${index}` when index is corresponding to a DHCP option

whois-marvin-42, 2022-11-22:
Hi,

First, thanks for this fabulous tool :bow:

I'm actually developing a service using a lot of iPXE scripts (and generating some of them based on some _patterns_). It uses -intensively- the fact that the `set` command supports '/' in variables names (per documentation), which works great ... most of the time, as I painfully discovered recently :smiling_face_with_tear: 

While it works flawlessly when the index is "in the middle", like `set pre/fixed/${index}/key value`, it doesn't when the index is at the end, no matter what prefix/variable name you're using (you can start iterating though ... since `/0` doesn't seem to be used internally ... unlike `/1` :smiling_face_with_tear: ) ... finding that one sent me on a path paved with some craziness :smirk: 

So after some _investigation_, it seems that _capability_ is colliding with an internal mechanism storing these (dhcp) options values under `/${dhcp-numbered-option}` without any specific (aka reserved prefix) which leads to `set foo/1 bar` raising an error (`Could not store "netmask" ...`) or `set bar/3 baz` raising a similar error -probably- for the same reason (`Could not store "gateway"...`). `/1` (`netmask`), `/3` (`gateway`), `/6` (`dns`), `/26` (`mtu`), `/54` (`dhcp-server`), `/119` (`dnssl`) are the "problematic" _indexes_/values.

Since the use of `set pre/fixed/${index}/key value` is so helping to (greatly) simplify the logic and the loading of the corresponding resources (and since the variable name can be re-used -partially or fully- as a path too :heart_eyes: ) may I suggest (if they cannot be hidden ... since their corresponding named variables are already _reserved_) you prefix your internal "indexes" with some reserved keyword (like `reserved/1` or `whatever/1`) and add this information/reserved keyword to the [`Notes` part of the webpage for the `set` command](https://ipxe.org/cmd/set#notes) ... to make everybody aware of this _trick_?!

Thanks for your help.

A little script showing the problem:
```shell
#!ipxe
goto main

:loop_ok
iseq ${pause} true && prompt --key t [DEBUG] ${index}/${index/last}: Press 't' to test with this new index ||
set foo/${index}/bar ok || goto error
echo [DEBUG] foo/${index}/bar: '${foo/${index}/bar}'
iseq ${index} ${index/last} && goto main_after_loop_ok || inc index && goto loop_ok

:loop_nok
iseq ${pause} true && prompt --key t [DEBUG] ${index}/${index/last}: Press 't' to test with this new index ||
set foo/${index} ok || goto error
echo [DEBUG] foo/${index}: '${foo/${index}}'
iseq ${index} ${index/last} && goto main_after_loop_nok || inc index && goto loop_nok

:error
echo [ERROR] variable name ending with ${index} is provoking an error!
# exit
goto main_after_loop_nok

:main
# set pause false
set pause true
# Expected: `set` command allows variables containing an index "in the middle" of their name. Got: OK
iseq ${pause} true && prompt --key p [DEBUG] Press 'p' to prove that variables names like .../{index}/... are ok! ||
set index:int32 0
set index/last:int32 9
goto loop_ok
:main_after_loop_ok

# set pause false
set pause true
# Expected: `set` command allows variables names ENDING with an index. Got: errors for values (1, 3, 6, ...) corresponding to (some) DHCP options (known) indexes
iseq ${pause} true && prompt --key t [DEBUG] Press 't' to test that variables ending with .../{index} = nok = BUG ?! ||
set index:int32 0 # Result: ok (1st failure is when index == 1)
# set index:int32 1 # Result: BUG => Could not store "netmask": Invalid argument (https://ipxe.org/1c056002)
# set index:int32 2 # Result: ok
# set index:int32 3 # Result: BUG => Could not store "gateway": Invalid argument (https://ipxe.org/1c056002)
# set index:int32 4 # Result: ok
# set index:int32 5 # Result: ok
# set index:int32 6 # Result: BUG => Could not store "dns": Invalid argument (https://ipxe.org/1c056002)
# set index:int32 7 # Result: ok
# ...
# set index:int32 26 # Result: BUG => Could not store "mtu": Invalid argument (https://ipxe.org/1c0c2002)
# set index:int32 27 # Result: ok
# ...
# set index:int32 54 # Result: BUG => Could not store "dhcp-server": Invalid argument (https://ipxe.org/1c056002)
# set index:int32 119 # Result: BUG => Could not store "dnssl": Invalid argument (https://ipxe.org/3c0c2003)
# set index:int32 120 # Result: ok ... tested up to 255 (highest DHCP options I found in RFCs)
# set index/last:int32 255
# set index/last:int32 125
set index/last:int32 10
goto loop_nok
:main_after_loop_nok

:debug_problematic_values
echo [DEBUG]   /1: ${/1} => netmask: ${netmask}
echo [DEBUG]   /3: ${/3} => gateway: ${gateway}
echo [DEBUG]   /6: ${/6} => dns: ${dns}
echo [DEBUG]  /26: ${/26} =>  mtu: ${mtu}
echo [DEBUG]  /54: ${/54} =>  dhcp-server: ${dhcp-server}
echo [DEBUG] /119: ${/119} =>  dnssl: ${dnssl}

prompt --key e Press 'e' to exit now ||
```

NiKiZe, 2022-11-22:
Awesome to see that you are using config scopes!
It is unlikely that the internal structure would change.
lets put it like this, like in many languages "variables can not start with a number"

Might also be good to know (for anyone that might not know about scopes) that any scoped setting can be accessed in any upper level without using the scope, which is a great feature, but at the same time it can be a bit confusing.

whois-marvin-42, 2022-11-23:
Hi,

Thanks for your reply :bow:  Are you sure nobody on the team can fix it?! :pray: Christmas is coming :christmas_tree: :santa: Don't you have a few elves to spare in Sweden?! :wink:

For now, I'm using a _workaround_ (far from the expected intellectual satisfaction of being able to produce something _clean :tm:_ ... as always with _workarounds_) but it puts a dent in what I was expecting to keep _as simple as possible_ ... still, iPXE is great (has been for years, as I said: immense respect for that :bow: :heart_eyes: )

Btw, based on the published **iPXE scripting grammar** (aka [`Command`](https://ipxe.org/cmd) + [`Settings`](https://ipxe.org/cfg) reference pages from ipxe.org, showing the reserved _labels_), **no limitation is being introduced anywhere**, not even in `set` and/or `inc` pages ... so, yeah ... I agree, there are things that can be _a bit confusing_ :wink:

So, with all due respect, as a great believer in the RTFM method (probably because I was raised during an era where the basics were _"Don't know how it works?! Use `man`! Don't know what `man` is?! Try `man man`!!!"_ :smirk: ), I think the **unexpected behavior** is still an issue, starting with the fact it's not documented! Of course, adding _limitations_ could be the way to handle it (momentarily?! :pray: ). In that case, if nothing else, you should update the documentation (at least the footnotes of the `set` command page?), so the people _crazy enough_ to play around with these _features_ will not get into trouble anymore :wink: (not talking about the time-consuming process to pin the problem itself, because everyone one of us has to deal with similar constraints ... but the doc ... please, at least, update it :pray:)

> (...) in many languages "variables can not start with a number" (...)

Concerning that "many languages" part, you're absolutely right (opinionated: mostly for historical/bad reasons, if I may :wink: ) ... **BUT** ... **iPXE interpreter DOES allow/implement/support it**! In fact, the parser resolves variables labeled with strings AND/OR **(only) numbers without trouble** (with such a limited instructions set, it could be a _killing feature_, if fully functional). That's actually the reason why (EXCEPT for the 6 undocumented values identified before) it worked flawlessly in many of my use cases (aka variables labeled only with numbers), allowing me to _play_ with `inc` in the middle of that, even _crazier stuff :tm:_ with nested variable substitution, etc.

But, since documentation / a lot of tutorials / etc. are limited to "basic" capabilities (yep, you have to _dig_ a little to see the _true potential_ of iPXE :smirk: ), maybe you consider that most users are not interested in those capabilities?! But it's only making it harder for those who'd like to go a little further ... and share knowledge about it :man_shrugging: 

> (...) It is unlikely that the internal structure would change (...)

Since we're talking about a "_presentation_" level problem, I don't think the _internal_ (aka C/core) _struct_ would have to change (and hope not, since it shouldn't be its _concern_ ... my 2 cents :man_shrugging: ). But still, the interpreter, in fact probably just the parser, in charge of _consuming_ iPXE scripts would have to be fixed since it's mainly a grammar definition/handling issue (aka not the gathering part, the one _talking_ with the hardware or any "low level" part / struct / whatever), leading to unexpected behavior.

In fact, it's "just" the _interface_ with _us_ (the ones who shouldn't know anything about what's happening "inside" :wink:), aka the _scripting_ part, which needs some fixing. Honestly, I would love to help but digging in the source code is way too outside of our scope (not to mention that _eating_ C code is bad for my health ... and my memory :smirk:) but I can propose an update of the documentation  (I searched `ipxe` repos, here on GitHub, without success ... please point me to the repository and I'll do a PR right away) and/or a tutorial (as one of the purposes was to publish a "howto push the envelope of iPXE without too much fuss" ... showing that great things can be done with it ... and, for those interested, even greater things with some _extra_ help :smirk: )

So ... unless there is a higher purpose / ulterior motive, historical/compatibility reason, may I suggest a -very- quick fix = remove (or make it a compile option) the conditional branching which is _somewhere_ forcing the interpreter/parser to do something different for those cases :man_shrugging:  And **since getting/setting these variables is documented/available ONLY when labeled with strings** (aka `${gateway}`, `${netmask}`, `${dns}`, `${mtu}`, `${dhcp-server}` and `${dnssl}`) ... why keep other undocumented variables doing the same thing (but with numbers only) ... and messing with possible killing features in the process (aka _locking_ `${/1}`, `${/3}`, `${/6}`, `${/26}`, `${/54}` and `${/119}` which, as a side effect, thanks to `scoping`, _locks_ `${foo/1}`, `${bar/3}`, etc.)

Btw, knowing that **your parser is already capable of handling them correctly** (numbers as/in variables, ex.: `${foo/0}` or `${foo/1/bar}` can be used), there is some extra work done somewhere, for nothing.! :man_shrugging:  Just **keep the main behavior**, aka evaluating them like other variables (named only with numbers: `${foo/0}`, `${bar/4}`, etc. = **it already works like a charm**) = most probably it means removing/disabling code inste [... cut ...] t/code :wink: :zipper_mouth_face:

Of course, everything is written here with a real/huge respect for the great work you've already put into this project and for making it available to the community. Obviously, since I don't have time to (really) read the source code, I can be completely wrong and it could be a huge deal to fix that for some reason :man_shrugging: 

No matter what, thanks for everything ... and let's start writing our letters to Santa ... and his elves :wink: :zipper_mouth_face:

whois-marvin-42, 2022-11-23:
I gave it another try ... to allow more _logic_ directly inside iPXE scripts (nothing _crazy_ though). While adding some _interesting_ features is still possible (ex.: localized menus) directly/only with iPXE _tricks_ ... between the limitations in the naming, the (pseudo) _scoping_ "side effects" (confusing:tm: by all means, as you pointed out) ... I probably reached the limit of what iPXE was designed for?! (only the scripting part, of course :wink: ... on purpose, probably, to _keep it tight_?!)

Since very few people would probably be interested in these capabilities, I'm closing this issue. Anyway, thanks for your feedback. Keep up the good work :bow:

I'll publish an article somewhere if/when I'll have time (isn't it what the Christmas period is for?! :smirk:) ... in the meantime, the _envelope_ of iPXE is still allowing pretty nifty stuff ... even if I was hoping to be able to push it a little further :zipper_mouth_face: 

 Best regards.

closed 2022-11-23 by whois-marvin-42

stappersg, 2022-11-23:
> I'll publish an article somewhere if/when I'll have time

:heart: