ipxe/ipxe #821: Error on 1st compile

DennisGlobal, 2022-11-30:
I notiched on the 1st ru, when i edit settings on general.h, i also have to make them on branding.h (seems to be solved now)

But then this error popped-up:
I havent made any alterations to zbin, so whats wrong?

info: its also missing from the Github location

  [BUILD] bin/ath9k.ids.o
  [AR] bin/blib.a
ar: creating bin/blib.a
  [HOSTCC] util/zbin
util/zbin.c:7:10: fatal error: lzma.h: No such file or directory
    7 | #include <lzma.h>
      |          ^~~~~~~~
compilation terminated.
make: *** [Makefile.housekeeping:1414: util/zbin] Error 1

NiKiZe, 2022-11-30:
You are missing dependencies as mentioned at https://ipxe.org/download#source_code

This is also explained in #189 (duplicate and invalid)
Did you search for this problem before opening a new issue?

closed 2022-11-30 by NiKiZe

DennisGlobal, 2022-12-01:
> You are missing dependencies as mentioned at https://ipxe.org/download#source_code
> 
> This is also explained in #189 (duplicate and invalid) Did you search for this problem before opening a new issue?

missing dependencies as mentioned at https://ipxe.org/download#source_code

liblzma or xz header files
- Both dependencies are unable to be located

mtools
- Was missing, is now installed > Suggested package: "floppyd" < Is the suggested one newer?

mkisofs
- Was missing, is now installed > Suggested package: "wodim cdrkit-doc" < Is the suggested one newer?

syslinux
- Was missing, is now installed