ipxe/ipxe #826: ipxe failed to fetch the tftp due to two dhcp interfaces

simoncheel, 2022-12-02:
Dear developer
I'm struggling on this issue for quite sometime

My computer has two interfaces , and both connect to the network that have dhcp server. during the pxe boot, the NIC can get the ip address from bootp server, but when try to load the boot configuration file failed from tftp server (connection timed out).

I searched around, and do a lot of test, I found the regular pxe boot loader does not have this issue

and finally find that the ipxe will try to active the interfaces from the begining, and when the first nic actived, it will not proceed the further checking, that will introduce a problem when I have two interfaces that both have dhcp will asign the ip address

can you pls guide me, how to resolve the issue in such situtation

expect your repsonse, thanks in advance.

stappersg, 2022-12-02:
> can you pls guide me, how to resolve the issue in such situtation

:smiley: 

> My computer has two interfaces , and both connect to the network that have dhcp server. during the pxe boot, the NIC can get the ip address from bootp server, but when try to load the boot configuration file failed from tftp server (connection timed out).

Start with using just one interface.

> expect your repsonse, thanks in advance.

Unaccepted, thank the iPXE project when you have a working iPXE setup.

simoncheel, 2022-12-02:
that's not a good idea... I need to have two dhcp server

NiKiZe, 2022-12-02:
iPXE tries each NIC, if it has no information about boot it will try the next NIC, make sure that only the one you want to boot from sends out next-server and filename.

This is not a bug or something that can be fixed in iPXE, however you might want to consider an embedded script with logic.

closed 2022-12-02 by NiKiZe