ipxe/ipxe #831: Accented characters are not displayed correctly (default/text mode) or at all (graphical mode)

whois-marvin-42, 2022-12-03:
Hi,

Using the current version of iPXE (previous ones not tested), accented characters are not displayed correctly (default text mode) ... or **at all** (graphical mode)!

When using the "basic" text mode, `echo` is -at least- able to display them using their `hex` code (huge pain in the *** trying to localize texts/menus/etc. with this _barbaric_ :wink: workaround) ... while some gibberish is rendered when using accented characters directly (ex.: writing french or other "_Latin based_" languages, since the required characters are part of the CP437 and the corresponding `keymap` are provided ... not talking about Arabic, Chinese and other people who will have to be even more _creative_ :tm: :zipper_mouth_face: :smirk: ).

On the other hand, only gibberish is displayed when switching to the graphical console :astonished: (which I was expecting to be able to use)!

By the -very quick- look I took in the source code, I don't see any regression from the -only?/very- old post found around this topic since this [commit](https://github.com/ipxe/ipxe/commit/a17ee36) should have fixed it?! But, still, something is actually wrong :sweat_smile:

The snippet provided here is a _debug_ script showing the error in both modes (tested on Linux with QEMU/KVM, using direct kernel boot ... aka _nothing is messing with the hardware/config prior to iPXE_ ... compiled with `CONSOLE_FRAMEBUFFER` and other required options to try bringing some localization to the user ... btw, thanks for the `dynamic` keymap feature :bow: ):
```shell
#!ipxe

set char_e_acute:hex 82
set char_e_acute ${char_e_acute:string}
set char_e_grave:hex 8a
set char_e_grave ${char_e_grave:string}

echo [DEBUG] Default text mode (console not set)
echo [DEBUG] ${char_e_acute}: é, ${char_e_grave}: è
prompt --key c Press 'c' to continue now ||

console --x 800 --y 600
echo [DEBUG] Using graphical console 800x600
echo [DEBUG] ${char_e_acute}: é, ${char_e_grave}: è
prompt --key c Press 'c' to continue now ||

console
echo [DEBUG] Console reset (back to default text mode)
echo [DEBUG] ${char_e_acute}: é, ${char_e_grave}: è
prompt --key c Press 'c' to continue now ||

exit
```

Thanks for your help.

mcb30, 2022-12-03:
iPXE currently has minimal support for non-ASCII characters.  Use UTF8 directly within iPXE scripts (e.g. `echo Héllo wôrld`: no need for `${char_e_acute}` etc.)

If the output console supports UTF8 then those characters will be displayed correctly.  This is currently limited to:
- serial console (assuming that the remote terminal supports UTF8)
- Linux console (when running as a Linux executable)
- UEFI text console (for characters that have glyphs defined within the UEFI firmware)
- UEFI framebuffer console (for characters that have glyphs defined within the UEFI firmware)

NiKiZe, 2022-12-03:
I was sure this was explained quite excellent by mcb30 in a commit but closest I can find is https://github.com/search?q=repo%3Aipxe%2Fipxe+unicode&type=commits
So maybe it was on IRC.

iPXE prints what you tell it to print, char-wise, however for PCBIOS mode what is actually shown is in the hands of the BIOS, and not all of them do CP437 (even if most assume that is what it does)
In FB mode maybe we could in theory maybe do more ... but that would require internal charmap/font. (so we probably won't)
Same goes for EFI, but here the firmware hopefully has more chars built in.

I'd say that for your usecase using UTF-8 will solve your perceived issue. (from which this issue can be closed.)
For other charsets such as CJK (Chinese, Japanese, Korean) it is up to EFI what is supported, and other chars is can't/won't fix.

whois-marvin-42, 2022-12-03:
Hi there,

Thanks for your replies.

@NiKiZe You're more than right, I understand now I assumed (as many it seems) wrongly what iPXE is rendering directly (expecting CP437) instead of what's being really supported by the PCBIOS underneath (some crappy stuff older than me :smirk:). My bad, apologies about that :bow: (I searched _a little_ before opening the issue :wink: explanation was probably provided _somewhere_ but thanks for pointing the source code ... btw, updating the docs would probably help avoiding that kind of topic being raised again :shrug:)

@mcb30 / @NiKiZe I read about the _minimal_ support (archives of the mailing list, pointing to the commit I linked previously) but was expecting consistency between text mode and graphical one ?! Wrongly, obviously, since it doesn't rely on iPXE (duly noted about the _internal charmap font_ and the "_so we won't_").

@mcb30
Even if I understand now iPXE is _delegating_ character representation _directly_ to the underlying PCBIOS/EFI capabilities (since there is no _mapping_ handled by iPXE and it's not intended to do so), my point is `echo Héllo wôrld` doesn't work for me with the code cloned from this repo (only with the headers updated as expected + without the trick of using `hex`) ... otherwise I wouldn't have posted the issue)

But OK, that one is probably on me too ... somehow :shrug:

Thanks for your enlightenment :bow: @NiKiZe already answered me on another topic + the lack of certain _features_ (so painful to manage the cursor position too ... trying to provide another kind of user experience) = I definitely understand now I was trying to push the boundaries way too far outside of what's expected to be possible within iPXE scripting = dropping the case (services like netboot.xyz are more than good after all, now understanding why he was not trying to _push the envelope_ even further :smirk:)

We'll keep managing _all the good stuff_ outside of iPXE :wink: I was just hoping to make some of it publically available but lesson learned = I'll stop trying to make iPXE do what is not its purpose (and keep using it _only_ for its _obvious killing features_ ... HTTPS, params, image verification, etc. :hearts: )

No matter what, you're providing a great tool. Immense respect

NiKiZe, 2022-12-03:
Maybe you can but this as a question on a higher level instead? Open a discussion about what you want to do, and ask for suggestions on how.

Some of this is on the "probably can't/won't" but notice the probably. But thinking about it, maybe we could have fontloading support, but that only works in efi FB.

stappersg, 2022-12-03:
> We'll keep managing all the good stuff outside of iPXE :wink: I was just hoping to make some of it publically available but lesson learned = I'll stop trying to make iPXE do what is not its purpose [...]

Think free, don't restrict your thinking.  Feel free to publish some where what is build upon / with _iPXE_, feel free to announce it here.