ipxe/ipxe #853: Unable to bring up a network device ("Down")

anderbubble, 2023-01-19:
We are unable to bring up an interface on an Intel Ethernet 10G X520 NIC when running in UEFI mode. (The same interface works in legacy/BIOS mode.) We have attempted using the latest release (1.21.1) as well as a build from master, but both produced the same result. We have tried with autoboot and (as a minimal test case) iflinkwait. (See attached screenshot.)

We will test again with a debug build, as recommended at https://ipxe.org/err/380861; but we would appreciate any insight on this issue that the iPXE developers might have. Is this device known / expected to work with iPXE today in UEFI mode? How can we get you more information to understand what's going on? (We'll get you whatever debug information we're able to extract from this latest build.)

Thanks!

[image]

mcb30, 2023-01-19:
Please try building with `DEBUG=intel,intelx`.  Since you say the card works on this system in BIOS mode but not in UEFI mode, it would be worth capturing the debug output for both BIOS and UEFI, for comparison.

You could also try hacking the code so that iPXE does not wait for a link status change notification before rechecking the link state:

```diff
--- a/src/drivers/net/intelx.c
+++ b/src/drivers/net/intelx.c
@@ -333,6 +333,9 @@ static void intelx_poll ( struct net_device *netdev ) {
        if ( eicr & INTELX_EIRQ_LSC )
                intelx_check_link ( netdev );
 
+       // always recheck link state
+       intelx_check_link ( netdev );
+
        /* Refill RX ring */
        intel_refill_rx ( intel );
 }
```

anderbubble, 2023-01-19:
Thanks: will do.

Can we compile with DEBUG=Intel,intelx,netdevice all together? Or is netdevice just meant to be a stand-in for the specific device anyway?

mcb30, 2023-01-19:
Yes, you can use all three together.  As described at https://ipxe.org/download#debug_builds the `DEBUG=` argument is essentially a comma-separated list of source files: using `DEBUG=intel,intelx,netdevice` will enable debugging (at level 1) for `intel.c`, `intelx.c`, and `netdevice.c`.

Please note that it **is** case-sensitive: your `DEBUG=Intel,intelx,netdevice` would give you an error because of the stray upper-case `I` at the start of `Intel`.

anderbubble, 2023-01-19:
Understood; that `I` was just because I was typing my last message on my phone. :/

We'll get that debug information generated and to you. Thanks!

anderbubble, 2023-01-20:
We have gathered the requested debug information. This is using iPXE builds with `DEBUG=intel,intelx,netdevice`. The interface we're trying to bring up has mac address 00:8C:FA:F2:12:3E, which is showing up as a different device in iPXE depending on which mode we're in.

First, bios mode works, with the intended interface coming up as net0:

[image]
[image]
[image]
[image]

In uefi mode this device appears to be net2, and is producing (with all its peers) the debug message "could not add SNP device: Error 0x7f45e082" with a link to https://ipxe.org/err/7f45e082. (Let me know if you need us to re-run these tests with `DEBUG=efi_utils`)

[image]
[image]
[image]
[image]

I am confused, though, that net2 appears to change its mac address later to 00:8c:fa:f2:12:40. Why would iPXE re-enumerate the device names in the same boot?

[image]

In any case, here we are at the iPXE shell identifying the desired interface and trying to bring it up (in uefi mode).

[image]

mcb30, 2023-01-20:
Thanks for the debug information.  First, a couple of clarifications:

In the BIOS case, you are using `undionly.kpxe` which (by design) will only ever see a single network interface since it includes no native iPXE drivers and can only use the UNDI API abstraction provided by the BIOS's PXE ROM for the selected boot NIC.

In the UEFI case, you seem to be using `ipxe.efi`, which includes all native iPXE drivers.  This is seeing the following NICs:
* `net0` is a Mellanox HCA connected to an Infiniband fabric
* `net1` is the first port of the Intel X520, and is the NIC that you are trying to boot from
* `net2` is the second port of the Intel X520

> I am confused, though, that net2 appears to change its mac address later to 00:8c:fa:f2:12:40. Why would iPXE re-enumerate the device names in the same boot?

I do not see this happening anywhere.  In your UEFI mode screenshots it is `net1` that consistently has the MAC address `00:8c:fa:f2:12:3e`.  The `net2` interface is the other port of your X520 and consistently has the MAC address `00:80:fa:f2:12:40`.

Could you please try building `make DEBUG=intel,intelx bin/intelx.pxe` and testing that in BIOS mode in place of `undionly.kpxe`?  That will test the native iPXE driver in isolation from the UEFI firmware, which will provide a good first cut as to whether the problem lies in the native iPXE driver or some more obscure interaction with whatever the firmware has previously done to the NIC.  If it works, it will also allow us to see what the link status register reports in the working case.

anderbubble, 2023-01-20:
> In the BIOS case, you are using undionly.kpxe which (by design) will only ever see a single network interface since it includes no native iPXE drivers and can only use the UNDI API abstraction provided by the BIOS's PXE ROM for the selected boot NIC.

This matches my understanding of downstream (Warewulf) that it explicitly always hands out undionly.kpxe in BIOS mode, though I was not involved in the decision making that led to this. I suppose I had assumed that we _had_ to use undionly.kpxe when running in BIOS mode.

> I do not see this happening anywhere.

I'm probably just misreading the output, then, as I trust your interpretation over mine. But here's what I was looking at:

[image]

That looks to me like 00:8c:fa:f2:12:3e being net2.

[image]

That looks to me like net2 reporting a different mac address.

> Could you please try building make DEBUG=intel,intelx bin/intelx.pxe and testing that in BIOS mode in place of undionly.kpxe

Will do. Thanks for your help!

anderbubble, 2023-01-20:
> I suppose I had assumed that we had to use undionly.kpxe when running in BIOS mode.

It appears we are / Warewulf is trying to follow the advice here: https://ipxe.org/howto/chainloading

> If you have machines which attempt to perform a UEFI network boot, then download http://boot.ipxe.org/ipxe.efi and save it to your TFTP server directory.
> You will need to configure your DHCP server to hand out undionly.kpxe as the boot file to BIOS machines and ipxe.efi as the boot file to UEFI machines.

I know this is a tangent, but why do the instructions here indicate that we should use undionly.kpxe, if that's not necessary (e.g., if we're going to be able to boot in BIOS mode with intelx.pxe)?

mcb30, 2023-01-21:
> > I do not see this happening anywhere.
> 
> I'm probably just misreading the output, then, as I trust your interpretation over mine. But here's what I was looking at:
> 
> [image]
> 
> That looks to me like 00:8c:fa:f2:12:3e being net2.
> 
> [image]
> 
> That looks to me like net2 reporting a different mac address.

Ah, I see the source of the confusion, sorry.  The various messages:
```
NETDEV net# registered (phys SNP-0x###### hwaddr ##:##:##:##:##:##)
NETDEV net# could not add SNP device: Error 0x7f45e082 (https://ipxe.org/7f45e082)
```
are due to something outside of iPXE reacting to iPXE's creation of an SNP device by creating two child SNP devices.  iPXE's SNP driver then attempts to attach to each child device and to register it (which will use the next available network device index, which at that point in time will be one higher than the corresponding underlying iPXE device).  This registration then (correctly) fails - using `DEBUG=efi_utils` would show precisely how this registration is being (correctly) inhibited, but there's no need to do that unless you're particularly interested.

For the devices that get successfully registered, the statement:
>     * `net0` is a Mellanox HCA connected to an Infiniband fabric
>     * `net1` is the first port of the Intel X520, and is the NIC that you are trying to boot from
>     * `net2` is the second port of the Intel X520
is correct.

anderbubble, 2023-01-21:
> Ah, I see the source of the confusion

Thank you so much for the explanation!

> Could you please try building make DEBUG=intel,intelx bin/intelx.pxe and testing that in BIOS mode in place of undionly.kpxe

This is in process, and we'll get you these new logs as soon as we can. Thanks again for the help!

anderbubble, 2023-01-24:
Here is a (failed) boot attempt in legacy/bios mode using the requested bin/intelx.pxe with DEBUG=intel,intelx (and an embedded script that drops to an iPXE shell after an autoboot attempt).

[image]

[image]

[image]

So, if I understand properly, this indicates that the problem isn't an issue with legacy/bios vs uefi, but an issue with undi vs intel driver; is that right?

What can we do to help track this down further? Or do you have everything you need at this point?

anderbubble, 2023-01-24:
> Something that will probably also just work is for you to use snponly.efi and avoid the native driver. But that would not help fixing any issues, just work around it.

I tried to build this, just to get this particular environment unstuck for now; but it didn't build as I expected.

```
$ make bin/snponly.efi EMBED=$(readlink -f shell.ipxe)
make: *** No rule to make target 'bin/snponly.efi'.  Stop.
```

How do I build this?

NiKiZe, 2023-01-24:
> I tried to build this, just to get this particular environment unstuck for now; but it didn't build as I expected.
>
> ```
> $ make bin/snponly.efi EMBED=$(readlink -f shell.ipxe)
> make: *** No rule to [...]

See https://ipxe.org/appnote/buildtargets

mcb30, 2023-01-24:
@anderbubble Thank you for the debug images!

> So, if I understand properly, this indicates that the problem isn't an issue with legacy/bios vs uefi, but an issue with undi vs intel driver; is that right?

Yes, that is correct.  This now seems to be a problem with the iPXE Intel driver.

> What can we do to help track this down further? Or do you have everything you need at this point?

I will probably need to be able to reproduce the problem for myself, because this is likely to involve quite a bit of experimentation to fix the bug.  I do have some X520 cards (since I wrote the iPXE Intel driver), but those don't seem exhibit this problem, unfortunately.

I took a look through unmerged pull requests that relate to this driver.  The only one I could find was https://github.com/ipxe/ipxe/pull/56 which apparently fixes an issue relating to autonegotiation of 1G links.  Is your switch running at 1G rather than 10G?  If so, then you could try the change in that PR to see if it helps.

Otherwise, is there any way you could arrange for remote IP-KVM access to the affected system, along with SSH root access to a Linux-based server on the same network?  (Preferably a dedicated network, so that my debugging wouldn't break anything at your end!)

mcb30, 2023-01-24:
> ```
> $ make bin/snponly.efi EMBED=$(readlink -f shell.ipxe)
> make: *** No rule to make target 'bin/snponly.efi'.  Stop.
> ```
> 
> How do I build this?

`make bin-x86_64-efi/snponly.efi`

gp397, 2023-01-25:
I have been reading this with interest.

I have been seeing almost identical symptoms but with a 25G intel E810 card in a Dell server.  We have tried a couple of different ipxe builds, including the latest commit to master to no avail.

I believe the cards that I am using use the "ice.c" driver rather than "intelx.c" as discussed above.

Before I raise a new issue with a very similar set of screenshots as above, I was wondering if there is any common code path between ice.c and intelx.c that might be worth looking at as that might solve both of our issues?

Thanks

gp397, 2023-01-25:
Also I note #498 which was closed but describes basically this set of symptoms with the hardware type I am using.

mcb30, 2023-01-25:
@gp397 Yes, the E810 will be using the `ice` driver.  There is no common driver code shared between that and the `intelx` driver.  (The 10G `intelx` driver reuses shared code from the 1G `intel` driver, and the 100G `ice` driver reuses shared code from the 40G `intelxl` driver, but there's no sharing between the `intel`/`intelx` pair and the `intelxl`/`ice` pair.)

Issue #498 is not likely to be relevant since that was using the `undionly` driver, and so not using any of the native iPXE Intel driver code.

Please do raise a separate issue for the E810, and include screenshots from a build with `DEBUG=intelxl,ice`.

mcb30, 2023-01-25:
> I'm also checking on whether this is a 1GB network connection, which seems not unlikely to me. (I've experienced a similar problem with link in a Dell iDRAC situation at a different site.)
> 
> I'm checking if we might be able to make a test node available; but if not I'll try to follow-up with Dell to see if _they_ could make such a node available. I think that's probably a better situation, anyway; but it could take a little bit.

@anderbubble Thank you.  I do have a card here that I think is nominally the same model as yours (X520-DA2, PCI IDs `8086:10fb`, iPXE driver `82599-sfp`).  It's the card that was used when developing the `intelx` driver, so it's definitely known-working.

If you can give me more details about your setup (e.g. what the port is plugged into, what cable type, what link speed, etc) then I might be able to try reproducing the problem here.

anderbubble, 2023-01-25:
@mcb30 I'm not 100% certain yet, but I think we've confirmed that this 10Gb port is plugged into a 1Gb switchport (possibly through [something like this](https://www.fs.com/products/20057.html)? I know I've had a similar problem with the Dell iDRAC before, where it wouldn't autoneg from 10GbE to 1GbE.

Any chance you've tested that scenario with your card? I'll keep trying to get explicit confirmation of that state on our side, but it seems a likely explanation. (Even if that works in your environment, it could still be a Dell-specific issue.)

mcb30, 2023-01-25:
> @mcb30 I'm not 100% certain yet, but I think we've confirmed that this 10Gb port is plugged into a 1Gb switchport (possibly through [something like this](https://www.fs.com/products/20057.html)? I [...]

I have not tested the scenario of using an SFP-to-RJ45 transceiver to connect to a 1000BASE-T switch.  I've order one of https://www.amazon.co.uk/gp/product/B07TXRYJGF to test with - will let you know what I find.

anderbubble, 2023-01-27:
The `snponly.efi` build is working in this environment now, and the end-user has further confirmed that this connected to a 1GbE network. I'm hopeful that when @mcb30 is able to test that scenario that you'll have the same issue, and the fix will be apparent; but, if not, let me know and I can try to get a lab environment with the specific Dell server.

mcb30, 2023-01-28:
> The `snponly.efi` build is working in this environment now, and the end-user has further confirmed that this is connected to a 1GbE network. I'm hopeful that when @mcb30 is able to test that scenario that you'll have the same issue, and the fix will be apparent; but, if not, let me know and I can try to get a lab environment with the specific Dell server.

Thanks for confirming.

I have the SFP-to-RJ45 transceiver and I have the card in the machine under my desk, but following some recent surgery it's likely to be about a week before I can physically get down there to plug it in and test it. :upside_down_face:

anderbubble, 2023-01-30:
Thanks for your help! I look forward to your results; but I hope you recover well, regardless!

anderbubble, 2023-02-06:
@mcb30 how is your recovery? Have you been able to test out this case of iPXE over an SFP-to-RJ45 transceiver? (Or, perhaps more specifically, over 1GbE?)

anderbubble, 2023-02-27:
@mcb30 I hope you are well.

I'm just checking in to see if you have been able to test out this case of iPXE over an SFP-to-RJ45 / tGbE transciever, and  if you have discovered anything about the behavior we saw in the process.

Thanks!