ipxe/ipxe #856: cross-compilation broken

hramrach, 2023-01-19:
When cross-compiling the util/elf2efi.c is compiled with host compiler but includes the target-specific header src/include/ipxe/efi/ProcessorBind.h

When binary for different supported architecture is built on another supported architecture the host settings are applied potentially silently building wrong binaries.

When building on unsupported architecture such as powerpc or s390 no settings are applied and the build fails.

mcb30, 2023-01-20:
> When binary for different supported architecture is built on another supported architecture the host settings are applied potentially silently building wrong binaries.

Not actually a problem, since the only parts of the EDK2 headers that we are relying upon are those that define the PE format headers, and the layout of those does not vary by CPU architecture.  (It does vary between 32-bit and 64-bit architectures but we already handle that, hence the existence of `elf2efi32` and `elf2efi64`.)

> When building on unsupported architecture such as powerpc or s390 no settings are applied and the build fails.

This is tricky.  For any host CPU not supported by EDK2 there will not be a `ProcessorBind.h` that we could use.  I've created #857 which takes the alternative approach of using the host's `<stdint.h>` and defining the EFI integer types `UINT16` etc in terms of the standard types `uint16_t` etc.

closed 2023-01-20 by mcb30

hramrach, 2023-01-20:
Thanks for the fix. 

I came up with a hack using the arm64 include as fallback when nothing else applied but not using the arch-specific header in the first place is much better.

hramrach, 2023-01-20:
It still does not build on bigendian platforms but whatever

```
[  140s] x86_64-suse-linux-objdump -ht bin-i386-efi/ipxe.efi.tmp | perl ./util/sortobjdump.pl >> bin-i386-efi/ipxe.efi.tmp.map
[  140s] ./util/elf2efi32 --subsystem=10 bin-i386-efi/ipxe.efi.tmp bin-i386-efi/ipxe.efi
[  140s] ELF section header outside file in bin-i386-efi/ipxe.efi.tmp
[  140s] make: *** [Makefile.efi:34: bin-i386-efi/ipxe.efi] Error 1
[  140s] rm bin-i386-efi/version.ipxe.efi.o
[  140s] error: Bad exit status from /var/tmp/rpm-tmp.nz0LrT (%build)

```

mcb30, 2023-01-21:
> It still does not build on bigendian platforms but whatever

A fair point.  Happy to accept a separate pull request to fix this.  I'd suggest using `#include <endian.h>` to get `htole32()` etc, since that header is provided by `glibc-headers`, which we already require for `<elf.h>`.  The challenge will be keeping the code neat and tidy (and within the 80 character width limit) while adding all the necessary byte-swapping calls.  Also, unlike iPXE's internal `htonl()` etc, we can't use `htole32()` etc in static initialisers, so some structure initialisation will have to be moved.

hramrach, 2023-01-21:
Yes, it's not an easy fix, and I don't really have an incentive to write and test it.

On modern ppc64 hardware it's possible to pick endian as needed, s390 is not typical architecture for building/serving boot files, riscv should be supported with newer EDK2 headers and littleendian anyway, and I will leave mips/loongarch to somebody who has some idea about that.

stappersg, 2023-01-21:
> and I will leave mips/loongarch to somebody who has some idea about that.

It is _loongarch_ that this issue now linked with #820.

PR #857 (2023-01-20): [efi] Do not rely on ProcessorBind.h when building host binaries
We cannot rely on the EDK2 ProcessorBind.h headers when compiling a binary for execution on the build host itself (e.g. elf2efi), since the host's CPU architecture may not even be supported by EDK2.

Fix by skipping ProcessorBind.h when building a host binary, and defining the bare minimum required to allow other EDK2 headers to compile cleanly.

Fixes: #856 

Reported-by: Michal Suchánek <msuchanek@suse.de>