ipxe/ipxe #860: make: *** No rule to make target 'bin/snponly.efi'.

stappersg, 2023-01-24:
> Something that will probably also just work is for you to use snponly.efi and avoid the native driver. But that would not help fixing any issues, just work around it.

I tried to build this, just to get this particular environment unstuck for now; but it didn't build as I expected.

```
$ make bin/snponly.efi EMBED=$(readlink -f shell.ipxe)
make: *** No rule to make target 'bin/snponly.efi'.  Stop.
```

How do I build this?

_Originally posted by @anderbubble in https://github.com/ipxe/ipxe/issues/853#issuecomment-1401346060_

stappersg, 2023-01-24:
I can reproduce that build error:
```text
   ...
  [DEPS] libgcc/__moddi3.c
  [DEPS] libgcc/implicit.c
  [DEPS] libgcc/icc.c
  [DEPS] libgcc/__divmoddi4.c
  [DEPS] libgcc/__divdi3.c
make: *** No rule to make target 'bin/snponly.efi'.  Stop.
stappers@trancilo:~/src/mailinglists/ipxe/src
$ export LANG=C
stappers@trancilo:~/src/mailinglists/ipxe/src
$ make bin/snponly.efi 
  [PARSEROM]
make: *** No rule to make target 'bin/snponly.efi'.  Stop.
stappers@trancilo:~/src/mailinglists/ipxe/src
$ 
```

stappersg, 2023-01-24:
Not even after a `git pull`:
```text
   ...
  [DEPS] libgcc/__umoddi3.c
  [DEPS] libgcc/__udivmoddi4.c
  [DEPS] libgcc/__udivdi3.c
  [DEPS] libgcc/__moddi3.c
  [DEPS] libgcc/__divmoddi4.c
  [DEPS] libgcc/__divdi3.c
make: *** No rule to make target 'bin/snponly.efi'.  Stop.
stappers@trancilo:~/src/mailinglists/ipxe/src
$ make bin/snponly.efi 
make: *** No rule to make target 'bin/snponly.efi'.  Stop.
stappers@trancilo:~/src/mailinglists/ipxe/src
$ 
```

stappersg, 2023-01-24:
For what it is worth (and because we are in different time zones): https://ipxe.org/howto/bisect

NiKiZe, 2023-01-24:
.efi is only valid for bin-x-efi, not PCBIOS, see https://ipxe.org/appnote/buildtargets

Please stop the spam.

closed 2023-01-24 by NiKiZe