ipxe/ipxe #862: Increased TCP_MAX_WINDOW_SIZE?

rowlap, 2023-01-24:
It's been 10 years since TCP_MAX_WINDOW_SIZE	was last increased in commit 07bc73e0872730d1e6784c7460f12b836c77838d.

I'm seeing artificially constrained transfer speeds to fetch the Linux initrd (HTTP) over a WAN link with 100ms RTT. The comment in the source doesn't reflect the expected bandwidth in 2023.
```
 *    c) WAN: expected bandwidth 2MB/s, typical RTT 100ms, minimum
 *       required window 200kB.
```

If changing the default isn't desirable, then an IPv4 config setting would be useful.

mcb30, 2023-01-24:
Yes, it probably is worth changing the default value.  Now that we have TCP SACK support, the cost of a dropped packet is orders of magnitude less than "retransmit the whole window", so there is much less of a potential downside to having a large window size.

Do you have a baseline expected bandwidth figure for your WAN link with 100ms RTT, from which the required window size could be calculated?

rowlap, 2023-01-24:
For WAN throughput, 2x would be good, 4x would be better, 8x would be ideal, beyond that it's diminishing returns (although I wouldn't say no to 50MB/s).

Can always patch locally for the specific network environment, but wanted to leverage an upstream bug report from the slow-boot-times we've observed, and good to hear about SACK support.

mcb30, 2023-01-25:
@rowlap I took some measurements between a handful of EC2 regions and came up with the figures in #863.  These fortuitously agree with the 8x multiplier that you suggested.

PR #863 (2023-01-25): [tcp] Update maximum window size to 2MB
The current maximum window size of 256kB was calculated based on rough link bandwidth and RTT measurements taken in 2012, and is too small to avoid filling the TCP window on some modern links.

Update the list of typical link bandwidth and RTT figures to reflect the modern world, and increase the maximum window size accordingly.

Fixes: #862
closed 2023-01-25 by mcb30