ipxe/ipxe #870: strange dhcp timeout on net2 followed by ipxe crash on net3 with no link.

db-at-work, 2023-01-27:
using a chain.ipxe EMBED that lists all the network interfaces:

    Try to boot (a)ll network adapters in turn                                  
                                                                                
    --- Detected network adapters ---                                           
    net(0): 0c:42:a1:7b:e4:3c - PCI 0000:19:00.0 15b3:1015 ConnectX-4Lx         
    net(1): 0c:42:a1:7b:e4:3d - PCI 0000:19:00.1 15b3:1015 ConnectX-4Lx         
    net(2): 1c:34:da:7f:9d:6a - PCI 0000:5e:00.0 15b3:1015 ConnectX-4Lx         
    net(3): 1c:34:da:7f:9d:6b - PCI 0000:5e:00.1 15b3:1015 ConnectX-4Lx         
                                                                                
    --- Alternatives ---                                                        
    Open (c)onfiguration                                                        
    (R)eboot computer                                                           
    Drop to iPXE (s)hell                                                        
    E(x)it and continue BIOS boot order                                         

by default does (a) all adapters in turn

first two, by PCI ID, that we don't want the node to PXE over timeout as expected:

    Configuring [dhcp] (net0 0c:42:a1:7b:e4:3c)...... ok
    http://bang/menu.ipxe.................. Connection timed out (http://ipxe.org/4c116092)
    Configuring [dhcp] (net1 0c:42:a1:7b:e4:3d)...... ok
    http://bang/menu.ipxe......... [bang.vlan6]

third that should work - but also times out for some reason

    Configuring [dhcp] (net2 1c:34:da:7f:9d:6a)...... ok
    http://bang/menu.ipxe..... [bang.lab]

eventually spews when going to 4th port (net3) after the third also times out

    http://bang/menu.ipxe.................. Connection timed out (http://ipxe.org/4c116092)
    nodnic_port_allocate_eq: eq allocation error (Status = -1)
    flexboot_nodnic_eth_open: nodnic_port_allocate_eq failed (Status = -1)
    Could not open net3: Error 0x00000001 (http://ipxe.org/00000001)
    "net4": no such network device
    "net5": no such network device
    "net6": no such network device
    "net7": no such network device
    "net8": no such network device
    "net9": no such network device
    DHCP for all NICs failed, exiting iPXE...
    Continuing BIOS boot order...

Whereas if the net3 is chosen from the menu manually instead of allowing the (a) try all in turn, it works:

    Configuring [dhcp] (net2 1c:34:da:7f:9d:6a)...... ok
    http://bang/menu.ipxe... ok
    boot.ipxe.cfg... ok
    ipxe/hostname-elmo-et8b1.ipxe... ok

NiKiZe, 2023-01-27:
What does ifstat show after the failed scenario?

db-at-work, 2023-01-27:
http://bang/menu.ipxe.................. Connection timed out (http://ipxe.org/4c116092)
    nodnic_port_create_cq: cq allocation error (Status = -1)
    flexboot_nodnic_create_cq: nodnic_port_create_cq failed (Status = -1)
    Could not open net3: Error 0x00000001 (http://ipxe.org/00000001)
    "net4": no such network device
    "net5": no such network device
    "net6": no such network device
    "net7": no such network device
    "net8": no such network device
    "net9": no such network device
    DHCP for all NICs failed, exiting iPXE...
    net0: 0c:42:a1:7b:e4:3c using ConnectX-4Lx on 0000:19:00.0 (open)
      [Link:up, TX:26 TXE:0 RX:417 RXE:342]
      [RXE: 1 x "Network unreachable (http://ipxe.org/28056090)"]
      [RXE: 331 x "Operation not supported (http://ipxe.org/3c086083)"]
      [RXE: 10 x "The socket is not connected (http://ipxe.org/380f6093)"]
    net1: 0c:42:a1:7b:e4:3d using ConnectX-4Lx on 0000:19:00.1 (open)
      [Link:up, TX:5 TXE:0 RX:253 RXE:214]
      [RXE: 210 x "Operation not supported (http://ipxe.org/3c086083)"]
      [RXE: 4 x "The socket is not connected (http://ipxe.org/380f6093)"]
    net2: 1c:34:da:7f:9d:6a using ConnectX-4Lx on 0000:5e:00.0 (open)
      [Link:up, TX:4 TXE:0 RX:130 RXE:109]
      [RXE: 53 x "The socket is not connected (http://ipxe.org/380f6093)"]
      [RXE: 56 x "Operation not supported (http://ipxe.org/3c086083)"]
    net3: 1c:34:da:7f:9d:6b using ConnectX-4Lx on 0000:5e:00.1 (closed)
      [Link:down, TX:0 TXE:0 RX:0 RXE:0]
      [Link status: Unknown (http://ipxe.org/1a086194)]
    Continuing BIOS boot order...

NiKiZe, 2023-01-27:
Do you get IP on all nets and that maybe interferes?
Walk thru config menu and check state?

db-at-work, 2023-01-27:
Yes, the first two nets get IPs via DHCP they have forced bad next-server info in the DHCP options.
Third net should IP via DHCP with a good next-server and next-file.

The only interference I can imagine would be in the next-server next-file info from DHPC is updated with latest net DHCP results.

None of this should make it so the net3 link-down causes a ipxe traceback exception.

NiKiZe, 2023-01-27:
net2 not booting might be due to subnet conflicts or routing issues ?
net3 is only tried after net2 fails, you have not tried net3 on its own (from what I can see)

db-at-work, 2023-01-27:
net0 and net1 are on a different vlan - no IP or routing (server and nodes are on same broadcast domain) conflicts with net2.

net2 times out if net0 and net1 are tried and attempt to chain - routing to get to http would only occur if DHCP/DNS options from earlier DHCP results are not purged / ignored when new net is attempted.

net3 being tried should not have a traceback - manually selected or not.

NiKiZe, 2023-01-27:
But you are "only trying" net3 after net2 failure.
Does net3 fail if it is the first that is tried?

I still think you should keep net2 boot failure here. And treat net3 failing after net2 as a separate issue.

mcb30, 2023-01-28:
@db-at-work Your embedded script is failing to close the previous network interface after a failed boot attempt.  The iPXE `autoboot` logic does this automatically, because memory in iPXE is extremely limited and you are unlikely to be able to open several network devices at once.  Your embedded script seems to be attempting to partially replicate the logic of `autoboot`, but is missing the `ifclose` step.

db-at-work, 2023-01-28:
Excellent note for ifclose. I will add, and report back.

stappersg, 2023-02-18:
> Excellent note for ifclose. I will add, and report back.

How did it go?

db-at-work, 2023-02-24:
sorry it took so long to verify

adding the ifclose to this loop 

```:defloop
ifconf -c dhcp net${a} && chain --replace --autofree http://bang/menu.ipxe && goto exit ||
ifclose net${a}
inc a
iseq ${a} 10 || goto defloop
```

seems to have done the trick

```   Try to boot (a)ll network adapters in turn

   --- Detected network adapters ---
   net(0): 0c:42:a1:51:3d:c4 - PCI 0000:01:00.0 15b3:1015 ConnectX-4Lx
   net(1): 0c:42:a1:51:3d:c5 - PCI 0000:01:00.1 15b3:1015 ConnectX-4Lx
   net(2): 0c:42:a1:1a:03:f2 - PCI 0000:07:00.0 15b3:1015 ConnectX-4Lx
   net(3): 0c:42:a1:1a:03:f3 - PCI 0000:07:00.1 15b3:1015 ConnectX-4Lx

   --- Alternatives ---
   Open (c)onfiguration
   (R)eboot computer
   Drop to iPXE (s)hell
   E(x)it and continue BIOS boot order
Waiting for link-up on net0................. Down (https://ipxe.org/38086193)
Waiting for link-up on net1................. Down (https://ipxe.org/38086193)
Configuring [dhcp] (net2 0c:42:a1:1a:03:f2)...... ok
http://bang/menu.ipxe... ok
boot.ipxe.cfg... ok
ipxe/mac-0c42a11a03f2.ipxe... No such file or directory (https://ipxe.org/2d0c618e)
ipxe/product-Leopard%20ORv2-DDR4.ipxe... No such file or directory (https://ipxe.org/2d0c618e)
http://bang/boot.ipxe... ok
```

thank you

closed 2023-02-24 by mcb30