ipxe/ipxe #876: PCI Bridge Driver causes freeze

mj-p, 2023-02-02:
Hey,
I have a notebook here promting the following (OCR'ed):
```
EFIPCI 0000:00:14.2 (8086:9d31 class 118000) has no driver
EFIPCI 0000:00:16.0 (8086:9d3a class 078000) has no driver 
EFIPCI 0000:00:1Y.0 (8086:9d03 class 010601) has no driver
EFIPCI 0000:00:1c.0 (8086:9dI4 class 060400) has driver -bridge-
EFIDRV PciRoot(0x0)/Pci(OxiC,Ox0) has driver "PCI-
EFIDRV PciRoot(0x0)/Pci(OxiC,Ox0) before disconnecting: 
EFIDRV PciRoot(0x0)/Pci(OxiC‘Ox0) Pcilo supported
HANDLE PciRoot(0x0)/Pci(OxiC,Ox0) Pcilo opened 26x (H) by DxeCore(?) for <NULL>
HANDLE PciRoot(0x0)/Pci(Ox1C, Ox0) Pc i In opened 4x (G) by Storage Security Command Dr iver 
HANDLE PciRoot(0x0)/Pci(0x1C, Ox0) Pcilo opened 4x (T) by ISA Bus Dr iver
HANDLE PciRoot(0x0)/Pci(OxiC,Ox0) Pcilo opened 4x (G) by AHCI Bus Driver
HANDLE PciRoot(0x0)/Pci(0x1C,Ox0) Pcilo opened lx (D) by Integrated Touch Driver 
HANDLE PciRoot(0x0)/Pci(OxiC,0x0) DevicePath supported
HANDLE PciRoot(0x0)/Pci(OxiC,0x0) DevicePath opened 82x (H) by DxeCore(?) for <NULL> 
HANDLE PciRoot(0x0)/Pci(0x1C,Ox0) DevicePath opened 4x (T) by Realtek UEF I UND I Driver
HANDLE PciRoot(0x0)/Pci(OxiC,Ox0) DevicePath opened 4x (T) by Platform Console Management Driver 
HANDLE PciRoot(0x0)/Pci(0x1C,Ox0) DevicePath opened 4x (T) by Simple Network Protocol Driver 
HANDLE PciRoot(0x0)/Pci(OxiC,0x0) DevicePath opened 4x (T) by Platform Console Management Driver 
HANDLE PciRoot(13x0)/Pci(OxiC,Ox0) DevicePath opened 4x (G) by SD Host Controller Driver
EFIDRV PciRoot(0x0)/Pci(0x1C,Ox0) disconnecting existing drivers 
EFIDRV PciRoot(Ox0)/Pci(Ox1C,Ox0) after disconnecting:
HANDLE PciRoot(Ox0)/Pci(0x1C,Ox0) Pcilo supported
HANDLE PciRoot(0x0)/Pci(0x1C,Ox0) Pcilo opened 26x (H) by DxeCore(?) for <NULL>
HANDLE PciRoot(0x0)/Pci(OxiC Ox0) Pcilo opened 4x (G) by Storage Security Command Driver 
HANDLE PciRoot(0x0)/Pci(OxiC Ox0) Pcilo opened 4x ( T) by I SA Bus Driver
HANDLE PciRoot(OxiD/Pci(OxiC Ox0) Pcilo opened 4x (G) by AHC I Bus Driver
HANDLE PciRoot(0x0)/Pci(0x1C,Ox0) Pcilo opened 1 x (D) by Integrated Touch Driver 
HANDLE PciRoot(0x0)/Pci(0x1C,Ox0) DevicePath supported,
HANDLE PciRoot(0x0)/Pci(0x1C,Ox0) DevicePath opened 82x (H) by DxeCore(?) for <NULL> 
HANDLE PciRoot(0x0)/Pci(Ox1C,Ox0) DevicePath opened 4x (T) by Rea 1 tek UEFI UNDI Driver
HANDLE PciRoot(0x0)/Pci(Ox1C,Ox0) DevicePath opened 4x (T) by Platform Console Management Driver 
HANDLE PciRoot(0x0)/Pci(OxIC,Ox0) DevicePath opened 4x (T) by Simple Network Protocol Driver 
HANDLE PciRoot(0x0)/Pci(OxiC,Ox0) DevicePath opened 4x c. T) by Platform Console Management Driver 
HANDLE PciRoot(0x0)/Pci(0x1C,Ox0) DevicePath opened 4x (G) by SD Host Controller Driver
EFIDRV PciRoot(0x0)/Pci(OxIC,Ox0) connecting new drivers
EFIPCI 0000:00:1c.0 (8086:9d14 class 060400) has driver -bridge-
EFIDRV PciRoot(0x0)/Pci(0x1C,Ox0) has driver -PCI-
EF1DRV PciRoot(0x0)/Pci(Ox1C,Ox0) DRIVER_START
EFIPcI 0000:00:1c.0 (8086:9d14 class 060400) has driver -bridge-
```

If I compile without pcibridge, it works as in later down the list the realtek driver is found and used.
Is this intended behaviour to actually start the pci bridge driver?

mcb30, 2023-02-02:
Good catch.  No, the `bridge` driver is required only for BIOS platforms (and even then, only for builds that include the `ena` network driver).  It should never be used in UEFI platforms.

Testing so far has missed this since VMs tend to place virtual NICs directly onto the host bridge.

I need to think of the best way to fix this.  In the meantime: what is your name and email (for the eventual commit log message)?

mj-p, 2023-02-02:
Thanks for your hard work!

Marcel Petersen <mp@sbe.de>

mcb30, 2023-02-03:
@mj-p Should now be fixed - please test and let us know if not!

PR #878 (2023-02-03): [efi] Do not attempt to drive PCI bridge devices
The "bridge" driver introduced in 3aa6b79 ("[pci] Add minimal PCI bridge driver") is required only for BIOS builds using the ENA driver, where experimentation shows that we cannot rely on the BIOS to fully assign MMIO addresses.

Since the driver is a valid PCI driver, it will end up binding to all PCI bridge devices even on a UEFI platform, where the firmware is likely to have completed MMIO address assignment correctly.  This has no impact on most systems since there is generally no UEFI driver for PCI bridges: the enumeration of the whole PCI [... cut]
closed 2023-02-03 by mcb30