ipxe/ipxe #877: Booting ipxe from usb causes megaraid_sas driver to fail

zito, 2023-02-03:
Hi,
the title was already used in the past here: https://forum.ipxe.org/showthread.php?tid=32084
I am confirming the problem with ipxe via PXE protocol over network (not from usb).
The problem takes me two days of experimenting with an old Dell R710 (or precisely two servers) and PERC H700.
We have developed rich ipxe menu-system, ipxe is great tool.
 I tried successively Debian installer stable (11),  oldstable(10), oldoldstable(9), unstable with kernel 6.1.8. Through chain-loading netboot.xyz also some versions of CentOS and Fedora. Always megaraid_sas failed to communicate with Raid controller. I tried BIOS mode and UEFI mode too. I am not saying the problem is in ipxe itself. Probably the problem is somewhere in some resource collision. But who knows? This issue can be help-full for others.
We needed server installed ASAP, so a colleague took fash drive and wrote Debian installer on it. Insert the drive into R710 a voila, megaraid_sas worked!
Output of megaraid_sas from a Debian stable installer loaded from iPXE:
```
~ # uname -a
Linux vrabec-1 5.10.0-20-amd64 #1 SMP Debian 5.10.158-2 (2022-12-13) x86_64 GNU/Linux
~ # lspci -s 03:00.0 -v -nn
03:00.0 RAID bus controller [0104]: Broadcom / LSI MegaRAID SAS 2108 [Liberator] [1000:0079] (rev 05)
        DeviceName: Integrated RAID    
        Subsystem: Dell PERC H700 Integrated [1028:1f17]
        Flags: fast devsel, IRQ 45, IOMMU group 24
        I/O ports at fc00 [size=256]
        Memory at df1bc000 (64-bit, non-prefetchable) [size=16K]
        Memory at df1c0000 (64-bit, non-prefetchable) [size=256K]
        Expansion ROM at df100000 [disabled] [size=256K]
        Capabilities: [50] Power Management version 3
        Capabilities: [68] Express Endpoint, MSI 00
        Capabilities: [d0] Vital Product Data
        Capabilities: [a8] MSI: Enable- Count=1/1 Maskable- 64bit+
        Capabilities: [c0] MSI-X: Enable- Count=15 Masked-
        Capabilities: [100] Advanced Error Reporting
        Capabilities: [138] Power Budgeting <?>
        Kernel modules: megaraid_sas
~ # dmesg | grep mega
[  120.687962] megasas: 07.714.04.00-rc1
[  120.688596] megaraid_sas 0000:03:00.0: BAR:0x1  BAR's base_addr(phys):0x00000000df1bc000  mapped virt_addr:0x00000000d73e5496
[  120.688601] megaraid_sas 0000:03:00.0: FW now in Ready state
[  120.688605] megaraid_sas 0000:03:00.0: 63 bit DMA mask and 32 bit consistent mask
[  120.689224] megaraid_sas 0000:03:00.0: requested/available msix 1/1
[  120.689227] megaraid_sas 0000:03:00.0: current msix/online cpus      : (1/24)
[  120.689230] megaraid_sas 0000:03:00.0: RDPQ mode     : (disabled)
[  127.720734] megaraid_sas 0000:03:00.0: Failed to init firmware
[  127.721880] megaraid_sas 0000:03:00.0: Failed from megasas_init_fw 6467
```
Worth noting the above lspci output has missing flag `bus master` and capability MSI-X is `enable-`. While on a system running initialized MegaRAID card:
```
hermes:~# lspci -s 03:00.0 -v -nn
03:00.0 RAID bus controller [0104]: Broadcom / LSI MegaRAID SAS 2108 [Liberator] [1000:0079] (rev 05)
        DeviceName: Integrated RAID                         
        Subsystem: Dell PERC H700 Integrated [1028:1f17]
        Flags: bus master, fast devsel, latency 0, IRQ 37, IOMMU group 17
        I/O ports at fc00 [size=256]
        Memory at df1bc000 (64-bit, non-prefetchable) [size=16K]
        Memory at df1c0000 (64-bit, non-prefetchable) [size=256K]
        Expansion ROM at df100000 [disabled] [size=256K]
        Capabilities: [50] Power Management version 3
        Capabilities: [68] Express Endpoint, MSI 00
        Capabilities: [d0] Vital Product Data
        Capabilities: [a8] MSI: Enable- Count=1/1 Maskable- 64bit+
        Capabilities: [c0] MSI-X: Enable+ Count=15 Masked-
        Capabilities: [100] Advanced Error Reporting
        Capabilities: [138] Power Budgeting <?>
        Kernel driver in use: megaraid_sas
        Kernel modules: megaraid_sas
hermes:~# dmesg | grep mega
[    1.753145] megasas: 07.714.04.00-rc1
[    1.753355] megaraid_sas 0000:03:00.0: BAR:0x1  BAR's base_addr(phys):0x00000000df1bc000  mapped virt_addr:0x(____ptrval____)
[    1.753357] megaraid_sas 0000:03:00.0: FW now in Ready state
[    1.753359] megaraid_sas 0000:03:00.0: 63 bit DMA mask and 32 bit consistent mask
[    1.753704] megaraid_sas 0000:03:00.0: requested/available msix 1/1
[    1.753706] megaraid_sas 0000:03:00.0: current msix/online cpus      : (1/24)
[    1.753707] megaraid_sas 0000:03:00.0: RDPQ mode     : (disabled)
[    1.863822] megaraid_sas 0000:03:00.0: controller type       : MR(512MB)
[    1.863824] megaraid_sas 0000:03:00.0: Online Controller Reset(OCR)  : Enabled
[    1.863827] megaraid_sas 0000:03:00.0: Secure JBOD support   : No
[    1.863828] megaraid_sas 0000:03:00.0: NVMe passthru support : No
[    1.863830] megaraid_sas 0000:03:00.0: FW provided TM TaskAbort/Reset timeout        : 0 secs/0 secs
[    1.863832] megaraid_sas 0000:03:00.0: JBOD sequence map support     : No
[    1.863834] megaraid_sas 0000:03:00.0: PCI Lane Margining support    : No
[    1.863836] megaraid_sas 0000:03:00.0: megasas_init_mfi: fw_support_ieee=67108864
[    1.863866] megaraid_sas 0000:03:00.0: INIT adapter done
[    1.863869] megaraid_sas 0000:03:00.0: JBOD sequence map is disabled megasas_setup_jbod_map 5749
[    1.919770] megaraid_sas 0000:03:00.0: pci id                : (0x1000)/(0x0079)/(0x1028)/(0x1f17)
[    1.919773] megaraid_sas 0000:03:00.0: unevenspan support    : no
[    1.919775] megaraid_sas 0000:03:00.0: firmware crash dump   : no
[    1.919776] megaraid_sas 0000:03:00.0: JBOD sequence map     : disabled
```
Regards

mcb30, 2023-02-03:
> Worth noting the above lspci output has missing flag `bus master` and capability MSI-X is `enable-`. While on a system running initialized MegaRAID card:

It also seems to have a different IRQ and IOMMU group.  This suggests that something has changed in the PCI topology.

It's probably worth comparing the kernel `dmesg` output from a known-good and known-bad boot to find the **first** difference that the kernel reports, which may be nothing to do with the RAID card itself.

NiKiZe, 2023-02-03:
Which iPXEs binary did you use? Efi? If so ipxe.efi or snponly.efi? Does the behaviour change if you switch?

We had one report of ISA card probing was causing an issue, maybe something similar here.

zito, 2023-02-03:
The lspci outputs from the original post was from different servers, sorry. I have installed Debian stable now (debootstrap from CentOS already installed on the raid device) and now the differences are exactly comparable. I did lspci -nnvv this time and the outputs are attached
[lspci-nnvv.initialized.txt](https://github.com/ipxe/ipxe/files/10583011/lspci-nnvv.initialized.txt)
[lspci-nnvv.ipxe.txt](https://github.com/ipxe/ipxe/files/10583012/lspci-nnvv.ipxe.txt)

The version of ipxe is v1.21.1-282-gbc19 (build from src with DOWNLOAD_PROTO_HTTPS). The problem is in UEFI mode and BIOS mode too. Worth noting that in BIOS mode kernel loading stops at "Probing edd" and `edd=off` is neccessary. So the BIOS is not able to fulfill disk requests already?

NiKiZe, 2023-02-03:
So for bios you used undionly.kpxe and the same issue?

The request was for dmesg output on the same machine preferable same kernel, and working vs non working boot.

zito, 2023-02-03:
I build following targets
```
make bin/ipxe.pxe bin/ipxe.efi EMBED=menuchain.ipxe
```
For BIOS mode `ipxe.pxe` is used for UEFI `ipxe.efi`...

Complete dmesg for ipxe boot and disk boot are attached.
[dmesg.initialized.txt](https://github.com/ipxe/ipxe/files/10583375/dmesg.initialized.txt)
[dmesg.ipxe.txt](https://github.com/ipxe/ipxe/files/10583386/dmesg.ipxe.txt)

NiKiZe, 2023-02-04:
Could you please try undionly.kpxe and/or snponly.efi ?

bin/ipxe.efi is not valid build target, you need bin-x86_64-efi/ipxe.efi

zito, 2023-02-04:
Sorry, my fault. I looked at shell history and this item was probably some bad try...
I did `git pull` and rebuild the targets
```
zito@builder:~/ipxe2/src$ make bin-x86_64-pcbios/undionly.kpxe bin-x86_64-pcbios/ipxe.pxe  bin-x86_64-efi/ipxe.efi EMBED=menuchain.ipxe
make[1]: 'bin-x86_64-pcbios/undionly.kpxe' is up to date.
make[1]: 'bin-x86_64-pcbios/ipxe.pxe' is up to date.
make[1]: 'bin-x86_64-efi/ipxe.efi' is up to date.
zito@builder:~/ipxe2/src$ git describe --tags --always --long --abbrev=1 --match "v*"
v1.21.1-381-g6c033
```
The problem with `ipxe.pxe` is still there but with `undionly.kpxe` is gone! Debian Live loaded through `undionly.kpxe` works. I will be back in the evening and I will do more experiments in this direction.
Thanks for your time Christian!

NiKiZe, 2023-02-04:
Thank you for testing, this gives us at least some data points to work with.

@mcb30 what would be the best way to build ipxe.pxe or .efi without ISA drivers, or do you have better ideas on what is going on here?

mcb30, 2023-02-04:
I haven't had time to take more than a trivial look at this yet, but I notice the following in both `dmesg` outputs:
```
[    0.234991] DMAR-IR: This system BIOS has enabled interrupt remapping
               on a chipset that contains an erratum making that
               feature unstable.  To maintain system stability
               interrupt remapping is being disabled.  Please
               contact your BIOS vendor for an update
```
Might be worth disabling that in the BIOS to see if the problem goes away.

mcb30, 2023-02-04:
Also might be worth trying to disable DMAR completely, to see if that also works around the problem.

mcb30, 2023-02-04:
@zito I am puzzled how you managed to get `ipxe.pxe` to work at all.  According to your `lspci` output, all of the NICs have PCI IDs 14e4:1639.  This vendor:device ID is not supported natively by iPXE.  (It may just be missing a `PCI_ROM()` line, but there should be no native driver that claims to work with it.)

Could you please share the output from `ifstat` so we can see how iPXE thinks it is driving the card?

zito, 2023-02-04:
> I haven't had time to take more than a trivial look at this yet, but I notice the following in both `dmesg` outputs:
>
> ```
> [    0.234991] DMAR-IR: This system BIOS has enabled interrupt [...]

To be honest this R710 doesn't have latest possible BIOS version. The latest BIOS from Dell is 6.6.0, but this one has version 6.4.0. Unfortunately this old server (from 2011) is no longer online upgradable using LifeCycle controller. I will try upgrade later maybe. But another one R710 running with BIOS 6.6.0 has the exactly same log line in dmesg.

> Might be worth disabling that in the BIOS to see if the problem goes away.

Hmm, I don't know where. I see the following options where I'm not certain I know what they do:
under `Integrated Devices` there is `I/OAT DMA Engine .................. Enabled`
Help about this option says:
```
This field enables/disables the I/O Acceleration Technology
(I/OAT) option.                                            
                                                           
This feature should be enabled only if the hardware and    
software support I/OAT.                                    
```
under `PCI IRQ Assignment` there is
```
Embedded NIC 1 .......................................... IRQ 15
Embedded NIC 2 .......................................... IRQ 14
Embedded NIC 3 .......................................... IRQ 15
Embedded NIC 4 .......................................... IRQ 14
Integrated RAID ......................................... IRQ 15
Embedded I/OAT DMA Engine 1 ............................. IRQ 15
Embedded I/OAT DMA Engine 2 ............................. IRQ 14
Embedded I/OAT DMA Engine 3 ............................. IRQ 11
Embedded I/OAT DMA Engine 4 ............................. IRQ 10
Embedded I/OAT DMA Engine 5 ............................. IRQ 15
Embedded I/OAT DMA Engine 6 ............................. IRQ 14
Embedded I/OAT DMA Engine 7 ............................. IRQ 11
Embedded I/OAT DMA Engine 8 ............................. IRQ 10
Embedded I/OAT DMA Engine 8 ............................. IRQ 10
Embedded USB UHCI Controller 4 .......................... IRQ 14
Embedded USB UHCI Controller 5 .......................... IRQ 11
Embedded USB EHCI Controller 2 .......................... IRQ 10
Embedded USB UHCI Controller 1 .......................... IRQ 6 
Embedded USB UHCI Controller 2 .......................... IRQ 5 
Embedded USB EHCI Controller 1 .......................... IRQ 6 
Embedded Video .......................................... IRQ 10
Embedded SATA 1 ......................................... IRQ 14
```
I tried to disable `I/OAT DMA Engine` without success.

zito, 2023-02-04:
> Also might be worth trying to disable DMAR completely, to see if that also works around the problem.

I don't know how unfortunately.

I forgot to mention, that `edd=off` is needed on kernel command line only if there is the bad state. So the problem can be detected immediately when `edd=off` is not set and the loading of kernel hangs...

zito, 2023-02-04:
> @zito I am puzzled how you managed to get `ipxe.pxe` to work at all. According to your `lspci` output, all of the NICs have PCI IDs 14e4:1639. This vendor:device ID is not supported natively by [...]

:-) really?

> Could you please share the output from `ifstat` so we can see how iPXE thinks it is driving the card?

```
root@debian:~# ifstat 
       eno1                eno2                eno3                eno4       
 KB/s in  KB/s out   KB/s in  KB/s out   KB/s in  KB/s out   KB/s in  KB/s out
    0.00      0.00      0.00      0.00      0.00      0.00      0.00      0.00
    0.00      0.00      0.00      0.00      0.00      0.00      0.00      0.00
    0.00      0.00      0.00      0.00      0.00      0.00      0.00      0.00
    0.00      0.00      0.00      0.00      0.00      0.00      0.00      0.00
    0.00      0.00      0.00      0.00      0.00      0.00      0.00      0.00
^C
```
It seems to me, you mean some different utility. This is the output of `ifstat` on Debian...

Maybe the ipxe settings screen be better?
```

                          iPXE configuration settings
     
  pci/                                                                         
  builtin/                                                                     
  smbios/                                                                      
  netX/                                                                        
  net0/                                                                        
  busid.......... 01:14:e4:16:39
  busloc......... 0x100
  bustype........ PCI
  chip........... undipci
  hwaddr......... bc:30:5b:d7:3a:1c
  ifname......... net0
  mac............ bc:30:5b:d7:3a:1c
  mtu............ <not specified>
  active-scan.... <not specified>
  key............ <not specified>
  ssid........... <not specified>
  ...

                      Ctrl-X - exit configuration utility
```
So even `ipxe.pxe` can work in "undi" mode? I am going to read more about different flavours of ipxe executables.
I red about them in the past and now I have to refresh knowledge...
Thanks for interesting points.

NiKiZe, 2023-02-04:
ifstat in iPXE

zito, 2023-02-04:
> ifstat in iPXE

oh, sorry
```
net0: bc:30:5b:d7:3a:1c using undipci on 0000:01:00.0 (Ethernet) [open]
  [Link:up, TX:130 TXE:1 RX:551 RXE:331]
  [TXE: 1 x "Network unreachable (https://ipxe.org/28086011)"]
  [RXE: 240 x "Operation not supported (https://ipxe.org/3c086003)"]
  [RXE: 85 x "The socket is not connected (https://ipxe.org/380a6001)"]
  [RXE: 6 x "Invalid argumentt(https://ipxe.org/1c056002)"]
```

zito, 2023-02-04:
> Could you please try undionly.kpxe and/or snponly.efi ?

@NiKiZe as I read https://ipxe.org/appnote/buildtargets now it seems to me I misused ipxe.pxe? For the PXE environment is the right combination undionly.kpxe (pcbios) & snponly.efi (UEFI), right?
Chain-loading ipxe.pxe and ipxe.efi through PXE has no sense?

mcb30, 2023-02-05:
> @NiKiZe as I read https://ipxe.org/appnote/buildtargets now it seems to me I misused ipxe.pxe? For the PXE environment is the right combination undionly.kpxe (pcbios) & snponly.efi (UEFI), right? Chain-loading ipxe.pxe and ipxe.efi through PXE has no sense?

No, you haven't misused anything!

For UEFI, chaining `ipxe.efi` is usually the right thing to do: under UEFI iPXE has the option to use a native driver if one is available, falling back to using the SNP or NII/UNDI driver if it doesn't have a native driver.

For BIOS, it's slightly more complicated: the fallback mechanism doesn't really exist, so you have a choice of `undionly.kpxe` to use the UNDI driver, or `ipxe.pxe` if there is an iPXE native driver for the NIC.  Using `undionly.kpxe` will generally work with all NICs, but may be slower or missing some features such as VLAN support.

> ```
> net0: bc:30:5b:d7:3a:1c using undipci on 0000:01:00.0 (Ethernet) [open]
>   [Link:up, TX:130 TXE:1 RX:551 RXE:331]
>   [TXE: 1 x "Network unreachable (https://ipxe.org/28086011)"]
>   [RXE: 240 x [...]

Yours is an exciting special case.  There is no native driver in iPXE for this card.  However, the not-really-existent fallback mechanism to use the UNDI driver has actually succeeded in reloading the UNDI driver from the NIC's PXE ROM.  This doesn't usually work, and is a very rarely used code path (in both iPXE and in the NIC's PXE ROM) full of quirky behaviour.

It's not hugely surprising that reloading the UNDI driver from the NIC's PXE ROM could have ended up stomping on some state used by the RAID card's expansion ROM, which subsequently prevents the RAID driver from initialising properly.

Let's restrict further debugging attempts to using only `ipxe.efi` in UEFI mode.  Could you please post:

* `ifstat` output when using `ipxe.efi`
* `dmesg` and `lspci` output from a UEFI boot via `ipxe.efi`
* `dmesg` and `lspci` output from a UEFI boot from local disk, using the same kernel and with the same kernel command-line options as for the boot via `ipxe.efi`

zito, 2023-02-05:
> Yours is an exciting special case. There is no native driver in iPXE for this card. However, the not-really-existent fallback mechanism to use the UNDI driver has actually succeeded in reloading the [...]

Yes these NICs need a firmware from Broadcom to work in Linux. I have modified Debian installer initrd.img - added firmware files from the package `firmware-bnx2`...

> It's not hugely surprising that reloading the UNDI driver from the NIC's PXE ROM could have ended up stomping on some state used by the RAID card's expansion ROM, which subsequently prevents the RAID driver from initialising properly.

aha

> Let's restrict further debugging attempts to using only `ipxe.efi` in UEFI mode. Could you please post:
>     * `ifstat` output when using `ipxe.efi`
```
iPXE> ifstat
net0: bc:30:5b:d7:3a:1c using NII on NII-0000:01:00.0 (Ethernet) [open]
  [Link:up, TX:139 TXE:2 RX:741 RXE:422]
  [TXE: 1 x "Network unreachable (https://ipxe.org/28086090)"]
  [TXE: 1 x "Error 0x2a086089 (https://ipxe.org/2a086089)"]
  [RXE: 237 x "Operation not supported (https://ipxe.org/3c086083)"]
  [RXE: 75 x "The socket is not connected (https://ipxe.org/380a6093)"]
  [RXE: 95 x "Invalid argument (https://ipxe.org/1c056082)"]
  [RXE: 1 x "The socket is not connected (https://ipxe.org/380f6093)"]
net1: bc:30:5b:d7:3a:1e using NII on NII-0000:01:00.1 (Ethernet) [closed]
  [Link:up, TX:0 TXE:0 RX:0 RXE:0]
net2: bc:30:5b:d7:3a:20 using NII on NII-0000:02:00.0 (Ethernet) [closed]
  [Link:up, TX:0 TXE:0 RX:0 RXE:0]
net3: bc:30:5b:d7:3a:22 using NII on NII-0000:02:00.1 (Ethernet) [closed]
  [Link:up, TX:0 TXE:0 RX:0 RXE:0]
iPXE> show version
builtin/version:string = 1.21.1+ (g6c033)
```

>     * `dmesg` and `lspci` output from a UEFI boot via `ipxe.efi`
[dmesg.ipxe.pxe.txt](https://github.com/ipxe/ipxe/files/10611067/dmesg.ipxe.pxe.txt)
[lspci-vvnn.ipxe.pxe.txt](https://github.com/ipxe/ipxe/files/10611069/lspci-vvnn.ipxe.pxe.txt)

>     * `dmesg` and `lspci` output from a UEFI boot from local disk, using the same kernel and with the same kernel command-line options as for the boot via `ipxe.efi`
[dmesg.local.txt](https://github.com/ipxe/ipxe/files/10611088/dmesg.local.txt)
[lspci-vvnn.local.txt](https://github.com/ipxe/ipxe/files/10611089/lspci-vvnn.local.txt)

mcb30, 2023-02-05:
> > ```
> > * `dmesg` and `lspci` output from a UEFI boot via `ipxe.efi`
> > ```
> 
> [dmesg.ipxe.pxe.txt](https://github.com/ipxe/ipxe/files/10611067/dmesg.ipxe.pxe.txt) [lspci-vvnn.ipxe.pxe.txt](https://github.com/ipxe/ipxe/files/10611069/lspci-vvnn.ipxe.pxe.txt)

This output appears to be from a BIOS boot, not a UEFI boot as I asked for.

The idea is to make the `dmesg` outputs as close to possible as identical, so that it becomes possible to find any significant differences.  This gets much harder to do when there are unnecessary differences included.

zito, 2023-02-05:
> This output appears to be from a BIOS boot, not a UEFI boot as I asked for.
>
> The idea is to make the `dmesg` outputs as close to possible as identical, so that it becomes possible to find any [...]

Oh, sorry for the filenamei. It's my mistake  but outputs are really from UEFI.
I should name the files with ipxe.efi substring...
I did not touch server boot mode today. It is UEFI.

```
zito@bobek:/tmp$ grep -i efi dmesg.ipxe.pxe.txt 
[    0.000000] efi: EFI v2.10 by Dell Inc.
[    0.000000] efi: ACPI 2.0=0xcea7b000 ACPI=0xcea7a000 SMBIOS=0xcea78000 
[    0.064131] clocksource: refined-jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 7645519600211568 ns
[    3.649184] pci 0000:07:03.0: BAR 0: assigned to efifb
[    3.741267] Registered efivars operations
[    6.924962] efifb: probing for efifb
[    6.928557] efifb: framebuffer at 0xd5800000, using 1200k, total 1200k
[    6.935072] efifb: mode is 640x480x32, linelength=2560, pages=1
[    6.940982] efifb: scrolling: redraw
[    6.944551] efifb: Truecolor: size=8:8:8:8, shift=24:16:8:0
[    6.981234] fb0: EFI VGA frame buffer device
[    7.797186] tsc: Refined TSC clocksource calibration: 2526.999 MHz
[    8.002495] fb0: switching to mgag200drmfb from EFI VGA
[   24.967749] systemd-gpt-auto-generator[670]: EFI loader partition unknown, exiting.
[   24.975429] systemd-gpt-auto-generator[670]: (The boot loader did not set EFI variable LoaderDevicePartUUID.)
[   26.613886] pstore: ignoring unexpected backend 'efi'
```

I will try to re-upload files with the right filename now... (edit the post)

mcb30, 2023-02-05:
Thanks.  Looking at the diff, there are some curious quirks:
```
pci 0000:00:1f.0: quirk_ich7_lpc+0x0/0xc0 took 35156 usecs
```
```
pci 0000:03:00.0: quirk_blacklist_vpd+0x0/0x20 took 10270 usecs
```
indicating (I think) that something is causing unexpected delays.

I note also that the "failed to init firmware" error message comes from the following code:
```
	/*
	 * Issue the init frame in polled mode
	 */

	if (megasas_issue_polled(instance, cmd)) {
		dev_err(&instance->pdev->dev, "Failed to init firmware\n");
		megasas_return_cmd(instance, cmd);
		goto fail_fw_init;
	}
```
which shows that the failure is most likely some kind of timeout.

Lastly, I note that the `clocksource: Switched to clocksource tsc` message happens before the megaraid firmware is started in `dmesg.local.txt` but after the (failed) attempt to start the megaraid firmware in `dmesg.ipxe.efi.txt`.

Taken together, these suggest that something odd is happening with the system's timekeeping and that this may be causing the problem.  There is no hint I could see as to why the timekeeping might be failing this way.

You also still have some unnecessarily different kernel command-line parameters:
```diff
-Command line: BOOT_IMAGE=/boot/vmlinuz-5.10.0-21-amd64 root=UUID=9df4f6e2-2a51-4d6c-9bf6-f8dea527b9a1 ro console=ttyS1,115200 quiet
+Command line: vmlinuz boot=live config splash netboot=nfs nfsroot=10.0.156.101:/srv/boot/debian-live-11-amd64 initrd=initrd.img console=ttyS1,115200
```
It would be helpful to eliminate as many as possible of the differences (e.g. `config`, `splash`, `quiet`), since those will cause some changes in behaviour and so introduce some noise that makes debugging harder.

zito, 2023-02-05:
...
> It would be helpful to eliminate as many as possible of the differences (e.g. `config`, `splash`, `quiet`), since those will cause some changes in behaviour and so introduce some noise that makes debugging harder.

Parameters as you suggested:
[desg.local.2.txt](https://github.com/ipxe/ipxe/files/10613902/desg.local.2.txt)
[dmesg.ipxe.efi.2.txt](https://github.com/ipxe/ipxe/files/10613903/dmesg.ipxe.efi.2.txt)

mcb30, 2023-02-08:
@zito Thanks for your patience in narrowing down the problem.  Those logs are much easier to compare now.  Unfortunately, there is still nothing obvious that suggests a root cause.

I think we need to try disabling DMAR to rule that out.  You should have a setting for this somewhere in the BIOS setup screen.  It could be labelled "IOMMU", "DMAR", "DMA Remapping", "AMD-Vi", "VT-d", "Directed I/O", or something else: there isn't a standard name that vendors seem to use.

If you've successfully disabled DMAR in the BIOS then you should no longer see the various `Adding to iommu group` messages in the `dmesg` output.

zito, 2023-02-09:
@mcb30 Thanks for your support! I have already succeeded with the `undionly.kpxe` & `snponly.efi` iPXE variants. This works for other my systems too (as I tested so far).

For `ipxe.pxe` I am not able to change anything in BIOS to achieve disabling DMAR you have written. 
The System Setup Options description is from page 57 of [Hardware Owner's Manual](https://downloads.dell.com/manuals/all-products/esuprt_ser_stor_net/esuprt_poweredge/poweredge-r710_owner%27s%20manual_en-us.pdf).
I have tried to disable options `I/OAT DMA Engine` and `SR-IOV Global Enable` without success. I did not find any other option related to DMA or IOMMU. Probably it is not possible to disable DMAR on R710.

The solution for Dell PE R710 server is to use `undionly.kpxe` and `snponly.efi` currently. Maybe this can be closed with this solution?

Thank you for your effort!

stappersg, 2023-02-10:
> The solution for Dell PE R710 server is to use `undionly.kpxe` and `snponly.efi`

That is some kind of the previously suggested [_without ISA drivers_](https://github.com/ipxe/ipxe/issues/877#issuecomment-1416721673).

zito, 2023-02-10:
> @zito One potentially useful data point (if you don't mind testing) would be to try `snp.efi` (rather than `snponly.efi`). This would still not include any iPXE native drivers but would bind to all network interfaces (instead of just one), and so would give us a bisection point to determine whether or not it's an iPXE native driver that's causing problems.

`snp.efi` works like `snpolnly.efi`. 
```
iPXE> ifstat
net0: bc:30:5b:d7:3a:1c using NII on NII-0000:01:00.0 (Ethernet) [open]
  [Link:up, TX:15 TXE:2 RX:47 RXE:20]
  [TXE: 1 x "Network unreachable (https://ipxe.org/28086090)"]
  [TXE: 1 x "Error 0x2a086089 (https://ipxe.org/2a086089)"]
  [RXE: 15 x "Operation not supported (https://ipxe.org/3c086083)"]
  [RXE: 1 x "The socket is not connected (https://ipxe.org/380f6093)"]
  [RXE: 2 x "The socket is not connected (https://ipxe.org/380a6093)"]
  [RXE: 2 x "Operation not supported (https://ipxe.org/3c046083)"]
net1: bc:30:5b:d7:3a:1e using NII on NII-0000:01:00.1 (Ethernet) [closed]
  [Link:up, TX:0 TXE:0 RX:0 RXE:0]
net2: bc:30:5b:d7:3a:20 using NII on NII-0000:02:00.0 (Ethernet) [closed]
  [Link:up, TX:0 TXE:0 RX:0 RXE:0]
net3: bc:30:5b:d7:3a:22 using NII on NII-0000:02:00.1 (Ethernet) [closed]
  [Link:up, TX:0 TXE:0 RX:0 RXE:0]
```

> As a separate test, it could also be worth trying `ipxe.efi` (not `snponly.efi` or `snp.efi`) with non-PCI device support in iPXE temporarily hacked out:
> 
> ```diff
> --- a/src/core/device.c
> +++ b/src/core/device.c
> @@ -83,6 +83,9 @@ static void probe_devices ( void ) {
>         struct root_device *rootdev;
>         int rc;
>  
> +       // skip device probing
> +       return;
> +
>         for_each_table_entry ( rootdev, ROOT_DEVICES ) {
>                 list_add ( &rootdev->dev.siblings, &devices );
>                 INIT_LIST_HEAD ( &rootdev->dev.children );
> ```

I'm afraid this disables all the network devices...
```
iPXE 1.21.1+ (gef0a) -- Open Source Network Boot Firmware -- https://ipxe.org
Features: DNS HTTP HTTPS iSCSI TFTP VLAN SRP AoE EFI Menu

No more network devices                  

                                         
Boot failed. 
```

mcb30, 2023-02-10:
> I'm afraid this disables all the network devices...

Sorry, my mistake.  Please try this (still untested) patch instead:

```patch
--- a/src/core/device.c
+++ b/src/core/device.c
@@ -52,6 +52,10 @@ int device_keep_count = 0;
 static int rootdev_probe ( struct root_device *rootdev ) {
        int rc;
 
+       //
+       if ( strcmp ( rootdev->dev.name, "EFI" ) != 0 )
+               return 0;
+
        DBG ( "Adding %s root bus\n", rootdev->dev.name );
        if ( ( rc = rootdev->driver->probe ( rootdev ) ) != 0 ) {
                DBG ( "Failed to add %s root bus: %s\n",
@@ -68,6 +72,11 @@ static int rootdev_probe ( struct root_device *rootdev ) {
  * @v rootdev          Root device
  */
 static void rootdev_remove ( struct root_device *rootdev ) {
+
+       //
+       if ( strcmp ( rootdev->dev.name, "EFI" ) != 0 )
+               return;
+
        rootdev->driver->remove ( rootdev );
        DBG ( "Removed %s root bus\n", rootdev->dev.name );
 }
```

zito, 2023-02-10:
> > I'm afraid this disables all the network devices...
> 
> Sorry, my mistake. Please try this (still untested) patch instead:
> 
> ```diff
> --- a/src/core/device.c
> +++ b/src/core/device.c
> @@ -52,6 +52,10 @@ int device_keep_count = 0;
>  static int rootdev_probe ( struct root_device *rootdev ) {
>         int rc;
>  
> +       //
> +       if ( strcmp ( rootdev->dev.name, "EFI" ) != 0 )
> +               return 0;
> +
>         DBG ( "Adding %s root bus\n", rootdev->dev.name );
>         if ( ( rc = rootdev->driver->probe ( rootdev ) ) != 0 ) {
>                 DBG ( "Failed to add %s root bus: %s\n",
> @@ -68,6 +72,11 @@ static int rootdev_probe ( struct root_device *rootdev ) {
>   * @v rootdev          Root device
>   */
>  static void rootdev_remove ( struct root_device *rootdev ) {
> +
> +       //
> +       if ( strcmp ( rootdev->dev.name, "EFI" ) != 0 )
> +               return;
> +
>         rootdev->driver->remove ( rootdev );
>         DBG ( "Removed %s root bus\n", rootdev->dev.name );
>  }
> ```

It works

mcb30, 2023-02-10:
> It works

Great!  This is promising!  Let's see if we can narrow it down a bit further.  Could you try bypassing only ISA device probing? :

```patch
--- a/src/core/device.c
+++ b/src/core/device.c
@@ -52,6 +52,10 @@ int device_keep_count = 0;
 static int rootdev_probe ( struct root_device *rootdev ) {
        int rc;
 
+       //
+       if ( strcmp ( rootdev->dev.name, "ISA" ) == 0 )
+               return 0;
+
        DBG ( "Adding %s root bus\n", rootdev->dev.name );
        if ( ( rc = rootdev->driver->probe ( rootdev ) ) != 0 ) {
                DBG ( "Failed to add %s root bus: %s\n",
@@ -68,6 +72,11 @@ static int rootdev_probe ( struct root_device *rootdev ) {
  * @v rootdev          Root device
  */
 static void rootdev_remove ( struct root_device *rootdev ) {
+
+       //
+       if ( strcmp ( rootdev->dev.name, "ISA" ) == 0 )
+               return;
+
        rootdev->driver->remove ( rootdev );
        DBG ( "Removed %s root bus\n", rootdev->dev.name );
 }
```

zito, 2023-02-10:
Failed. ISA drivers are not the cause of problem.

mcb30, 2023-02-10:
Thanks.  There are only a handful of root drivers to try: `"EFI"` (which you will need, otherwise you'll get the `No network devices` error), `"PCI"` (shouldn't have any effect under UEFI anyway), `"ISA"`, `"EISA"`, `"ISAPnP"`, `"MCA"`, `"3c509"`, and `"Hyper-V"`.

You can build with `DEBUG=device` to see `Adding XXX root bus` messages as iPXE tries each one, to confirm that you've managed to successfully bypass the one you're currently trying to test without.

zito, 2023-02-10:
Right the next `EISA` is the problem. The debug output was

```
Adding EFI root bus
Adding Hyper-V root bus
Failed to add Hyper-V root bus: No such device (https://ipxe.org/2c834087)
Adding ISA root bus
Adding ISAPnP root bus
Adding MCA root bus
Adding PCI root bus
ok
```

mcb30, 2023-02-10:
@zito This is starting to make sense!

Your `lspci` output shows:
```
03:00.0 RAID bus controller [0104]: Broadcom / LSI MegaRAID SAS 2108 [Liberator] [1000:0079] (rev 05)
        ...
	Region 0: I/O ports at fc00 [size=256]
```
and the EISA probe sequence will currently write a byte to the I/O port at `fc80`.

I've created #888 which could potentially fix this.  Could you please test it and report back?

zito, 2023-02-11:
```
PXE boot - Embedded NIC 1: Broadcom NetXtreme II Gigabit Ethernet (BCM5709)
.iPXE initialising devices...Adding 3c509 root bus
Adding EFI root bus
Adding EISA root bus
Failed to add EISA root bus: No such device (https://ipxe.org/2c004087)
Adding Hyper-V root bus
Failed to add Hyper-V root bus: No such device (https://ipxe.org/2c834087)
Adding ISA root bus
Adding ISAPnP root bus
Adding MCA root bus
Adding PCI root bus

iPXE 1.21.1+ (gbe8e) -- Open Source Network Boot Firmware -- https://ipxe.org
Features: DNS HTTP HTTPS iSCSI TFTP VLAN SRP AoE EFI Menu
Configuring (net0 bc:30:5b:d7:3a:1c)...

...

http://10.0.156.101/boot/debian-live-11-amd64/boot/vmlinuz... ok                                                                                              
http://10.0.156.101/boot/debian-live-11-amd64/boot/initrd.img... ok                                                                                           
vmlinuz : 7019136 bytes [EFI] [SELECTED] "boot=live netboot=nfs nfsroot=10.0.156.101:/srv/boot/debian-live-11-amd64 initrd=initrd.img console=ttyS1,115200"   
initrd.img : 85177521 bytes                                                                                                                                   
EFI stub: Loaded initrd from command line option                                                                                                              
Removed PCI root bus                                                                                                                                          
Removed MCA root bus                                                                                                                                          
Removed ISAPnP root bus                                                                                                                                       
Removed ISA root bus                                                                                                                                          
Removed EFI root bus                                                                                                                                          
Removed 3c509 root bus                                                                                                                                        
[    0.000000] microcode: microcode updated early to revision 0x1f, date = 2018-05-08                                                                         
[    0.000000] Linux version 5.10.0-21-amd64 (debian-kernel@lists.debian.org) (gcc-10 (Debian 10.2.1-6) 10.2.1 20210110, GNU ld (GNU Binutils for Debian) 2.3)

...

root@debian:~# dmesg | grep mega
[    7.985551] megasas: 07.714.04.00-rc1
[    7.989605] megaraid_sas 0000:03:00.0: BAR:0x1  BAR's base_addr(phys):0x00000000df1bc000  mapped virt_addr:0x(____ptrval____)
[    7.989610] megaraid_sas 0000:03:00.0: FW now in Ready state
[    7.989614] megaraid_sas 0000:03:00.0: 63 bit DMA mask and 32 bit consistent mask
[    7.990157] megaraid_sas 0000:03:00.0: requested/available msix 1/1
[    7.990160] megaraid_sas 0000:03:00.0: current msix/online cpus      : (1/24)
[    7.990162] megaraid_sas 0000:03:00.0: RDPQ mode     : (disabled)
[    8.084104] megaraid_sas 0000:03:00.0: controller type       : MR(512MB)
[    8.090296] megaraid_sas 0000:03:00.0: Online Controller Reset(OCR)  : Enabled
[    8.090298] megaraid_sas 0000:03:00.0: Secure JBOD support   : No
[    8.090300] megaraid_sas 0000:03:00.0: NVMe passthru support : No
[    8.090302] megaraid_sas 0000:03:00.0: FW provided TM TaskAbort/Reset timeout        : 0 secs/0 secs
[    8.090304] megaraid_sas 0000:03:00.0: JBOD sequence map support     : No
[    8.090306] megaraid_sas 0000:03:00.0: PCI Lane Margining support    : No
[    8.090310] megaraid_sas 0000:03:00.0: megasas_init_mfi: fw_support_ieee=67108864
[    8.092055] megaraid_sas 0000:03:00.0: INIT adapter done
[    8.102517] megaraid_sas 0000:03:00.0: JBOD sequence map is disabled megasas_setup_jbod_map 5749
[    8.176118] megaraid_sas 0000:03:00.0: pci id                : (0x1000)/(0x0079)/(0x1028)/(0x1f17)
[    8.198613] megaraid_sas 0000:03:00.0: unevenspan support    : no
[    8.210963] megaraid_sas 0000:03:00.0: firmware crash dump   : no
[    8.210967] megaraid_sas 0000:03:00.0: JBOD sequence map     : disabled
root@debian:~# lsscsi 
[0:2:0:0]    disk    DELL     PERC H700        2.10  /dev/sda 
[1:0:0:0]    cd/dvd  TEAC     DVD-ROM DV-28SW  R.2A  /dev/sr0 
root@debian:~# lsblk
NAME   MAJ:MIN RM   SIZE RO TYPE MOUNTPOINT
loop0    7:0    0   1.5G  1 loop /usr/lib/live/mount/rootfs/filesystem.squashfs
sda      8:0    0 558.4G  0 disk 
├─sda1   8:1    0   256M  0 part 
├─sda2   8:2    0    16G  0 part 
└─sda3   8:3    0    16G  0 part 
sr0     11:0    1  1024M  0 rom  
root@debian:~# 
```
@mcb30 brilliant work! :+1: :1st_place_medal:

mcb30, 2023-02-11:
@zito Thanks for testing, and for all your patience helping to track this down to a root cause.

closed 2023-02-11 by mcb30

zito, 2023-02-11:
Michael thank you for your great work! Have a nice day

PR #888 (2023-02-11): [eisa] Check for system board presence before probing for slots
EISA expansion slot I/O port addresses overlap space that may be assigned to PCI devices, which can lead to register reads and writes with unwanted side effects during EISA probing.

Reduce the chances of performing EISA probing on PCI devices by probing EISA slot vendor and product ID registers only if the EISA system board vendor ID register indicates that the motherboard supports EISA.

Fixes: #877 

Debugged-by: Václav Ovsík <vaclav.ovsik@gmail.com>
Tested-by: Václav Ovsík <vaclav.ovsik@gmail.com>