ipxe/ipxe #883: vmware crash after using exit

steverweber, 2023-02-07:
vmware crash after ipxe exit 1 or exit 0, however chainload works.

I'm using EFI enabled virtual machine in vmware. The virtual LOM terminal crashes after using `exit 0` from a custom ipxe menu.

VMware gives useless error and the VM stops running:
`An unrecoverable IOException occurred so the connection was closed`

I'm not sure what the ESX version is but I was able to 

git bisect the issue to commit 
[[efi] Bind to only the topmost instance of the SNP or NII protocols](https://github.com/ipxe/ipxe/commit/68734b9a4dafa540e5333d7af3849b59a10f7a93)

mcb30, 2023-02-07:
Thanks. The commit you've identified is a bugfix commit and so cannot be reverted.

There is unfortunately not enough information in your description for anyone to be able to reproduce the error. Could you please simplify it down to a minimal reproduction (eg take out your custom menu, specify versions, etc)?

steverweber, 2023-02-07:
This bug report was more of a heads up of what commit might have caused vmware issues when using the `exit` command.

Here is my test case. This one passes. when building ipxe with commit `68734b9a4da` and latter I get the fail.

```
cd ~
# this branch works
branch=2fef0c541e4e2417fc285c4d9ddfcb6f23f394ad
# this branch fails - https://github.com/ipxe/ipxe/issues/883
#branch=68734b9a4dafa540e5333d7af3849b59a10f7a93
if test -e ~/ipxe/.ipxe_done_${branch} ; then
    echo 'already built'
    exit 0
fi
rm -rf ~/ipxe
git clone https://github.com/ipxe/ipxe.git ~/ipxe
cd ~/ipxe
git fetch
git checkout ${branch}
git pull
cd ~/ipxe/src
#sed -i 's|http://ca.ipxe.org/auto|http://bringup.fast.uwaterloo.ca/ca/auto|g' config/crypto.h
## disable the https verify !!
## TODO: would be nice to fix this!
## https://github.com/ipxe/ipxe/issues/623
sed -i 's|OCSP_CHECK|OCSP_CHECK_DISABLED|g' config/crypto.h
<<"EOF" cat > init.ipxe
#!ipxe
exit 1
EOF
<<"EOF" cat > config/local/general.h
#ifndef CONFIG_GENERAL_LOCAL_H
#define CONFIG_GENERAL_LOCAL_H
#define CONSOLE_FRAMEBUFFER
#define CONSOLE_CMD
// note: NET_PROTO_LLDP is enabled by default for latest efi builds
// #define NET_PROTO_LLDP
#define DOWNLOAD_PROTO_HTTPS
#define NSLOOKUP_CMD
#define VLAN_CMD
#define REBOOT_CMD
#define POWEROFF_CMD
//#define IMAGE_TRUST_CMD
#define PCI_CMD
#define PARAM_CMD
#define NEIGHBOUR_CMD
#define PING_CMD
#define IPSTAT_CMD
#define CPUID_SETTINGS
//#define MEMMAP_SETTINGS https://github.com/ipxe/ipxe/issues/429
#endif /* CONFIG_GENERAL_LOCAL_H */
EOF
make clean
make -j4 EMBED=init.ipxe
#make -j4 bin-x86_64-efi/ipxe.efi EMBED=init.ipxe DEBUG=dns:7
make -j4 bin-x86_64-efi/ipxe.efi EMBED=init.ipxe
touch ~/ipxe/.ipxe_done_${branch}
```

mcb30, 2023-02-07:
Thanks. Please try to reduce the test case to something simpler, ideally without any customisations such as config options or an embedded script.

steverweber, 2023-02-08:
yes sorry about that. here is a reduced example.

```
cd ~
# this branch works; ipxe exits and the system continues with the next boot device
# branch=2fef0c541e4e2417fc285c4d9ddfcb6f23f394ad
# this branch fails; VM crashes.
branch=68734b9a4dafa540e5333d7af3849b59a10f7a93
rm -rf ~/ipxe
git clone https://github.com/ipxe/ipxe.git ~/ipxe
cd ~/ipxe/src
<<"EOF" cat > init.ipxe
#!ipxe
exit 1
EOF
make clean
make -j4 EMBED=init.ipxe
make -j4 bin-x86_64-efi/ipxe.efi EMBED=init.ipxe
```

steverweber, 2023-02-08:
yes sorry here is a reduced example. this has the same crash.

```
cd ~
rm -rf ~/ipxe
git clone https://github.com/ipxe/ipxe.git ~/ipxe
cd ~/ipxe/src
# this commit works 2fef0c541e4e2417fc285c4d9ddfcb6f23f394ad
# this commit fails; VM crashes.
git checkout 68734b9a4dafa540e5333d7af3849b59a10f7a93
make clean
make -j4
make -j4 bin-x86_64-efi/ipxe.efi
```

kbucheli, 2023-03-10:
I have the same issue, using commit 8b645eea on  VMware ESXi version 7.0.3.

steverweber, 2023-04-09:
It seems that the latest vSphere/EXIv8 is not setting VM's default to EFI so this issues is going to get more common. 

a current workaround is to checkout master and revert this bug fix commit.

```
git revert --no-commit 68734b9a4dafa540e5333d7af3849b59a10f7a93
```

NiKiZe, 2023-04-09:
Don't know if it will help narrow down the issue and help finding a solution but building with DEBUG=snp and then doing the exact same exit, maybe in embedded script.
And build both with and without the commit reverted.
After this comparing the output from the 2 runs to se if there is any special case that can be handled.

I'm also curious if you can use snponly.efi and have better luck. But maybe also compare if there is any difference between ipxe.efi, snp.efi, snponly.efi.
If you know it is booted with VMware maybe send snponly.efi instead (if that works)

It would probably also be good to bring this issue to the attention of VMware. They might have a bug in their firmware implementation/driver, at least they might be interested in trying to mitigate the issue.

steverweber, 2023-04-09:
Sorry; dont have a serial connection setup for catching more verbose logs.

with `git revert --no-commit 68734b9a4dafa540e5333d7af3849b59a10f7a93`
- pass: boot snp.efi ; pass: exit 0  
    - logs: RED PciRoot0x0/pci0x16,0x0,00x0,0x0,mac...,0x1 is SNP device  ... 3xlines
    - logs: GREEN ... IPV4(0,0,0,0)  is SNP device  ...  2xlines
    - logs: YELLO ... IPV6(0,....)  is SNP device  ...  2xlines
    - logs: BLUE ... IPV6(0,....)  is SNP device  ...  2xlines
- pass: boot snponly.efi ; FAIL: exit 0
    - logs: none

with branch master 
- pass: boot snp.efi ; FAIL: exit 0
    - logs: RED PciRoot0x0/pci0x16,0x0,00x0,0x0,mac...,0x1 is SimpleNetwork device) ... 3xlines
- pass: boot snponly.efi ; FAIL: exit 0
    - logs: none

sadly I cant give this issue more investigation. Hope the little details I provided give some hints.

mcb30, 2023-06-07:
With a serial console attached, there are more details on the crash:

```
!!!! X64 Exception Type - 0D(#GP - General Protection)  CPU Apic ID - 00000000 !!!!
ExceptionData - 0000000000000000
RIP  - 00000000060D298E, CS  - 0000000000000018, RFLAGS - 0000000000010246
RAX  - 0000000007282030, RCX - 0000000007282120, RDX - 0000000007282030
RBX  - 0000000000000006, RSP - 0000000007FC97D0, RBP - 0000000007FC9860
RSI  - 00000000079F1618, RDI - 0000000007297830
R8   - 0000000007282120, R9  - 0000000007297798, R10 - 0000000000000000
R11  - 0000000000000000, R12 - 0000000007282018, R13 - 00000000060DF380
R14  - 0000000007A23998, R15 - 00000000060DF301
DS   - 0000000000000008, ES  - 0000000000000008, FS  - 0000000000000008
GS   - 0000000000000008, SS  - 0000000000000008
CR0  - 0000000080010033, CR2 - 0000000000000000, CR3 - 0000000007C01000
CR4  - 0000000000000668, CR8 - 0000000000000000
DR0  - 0000000000000000, DR1 - 0000000000000000, DR2 - 0000000000000000
DR3  - 0000000000000000, DR6 - 00000000FFFF0FF0, DR7 - 0000000000000400
GDTR - 00000000FFFFFC10 000000000000002F, LDTR - 0000000000000000
IDTR - 0000000007B53460 0000000000000FFF,   TR - 0000000000000000
FXSAVE_STATE - 0000000007FC9430
!!!! Can't find image information. !!!!
```

The instruction pointer here (0x60d298e) is within UefiPxeBcDxe (according to the UEFI shell utilities), and the crash occurs during our call to `UninstallMultipleProtocolInterfaces()` within `efi_snp_remove()`.

mcb30, 2023-06-07:
Some experimentation shows that it is the uninstallation of `EFI_SIMPLE_NETWORK_PROTOCOL` that triggers the crash: all other protocols can be uninstalled successfully.

mcb30, 2023-06-07:
A **lot** of further investigation has got closer to the root cause.

Calling `UninstallMultipleProtocolInterfaces()` leads to a call to `PxeBcStop()` in UefiPxeBcDxe.  This attempts to jump through an invalid pointer at the line
```c
 //
  // Stop functionality of PXE Base Code protocol
  //
  Status = Private->PxeBc.Stop (&Private->PxeBc);
```

A memory dump shows that the contents of `Private->PxeBc` are garbage: they appear to have been overwritten by UCS-2 character data, which in turn appears to be a constructed HII query string referencing `"HTTP_BOOT_CONFIG_IFR_NVDATA"`.  This string starts at exactly the same address as the `PXEBC_PRIVATE_DATA` structure, which is allocated by `AllocatePool()`.

Adding wrappers to trace calls to `AllocatePool()` and `FreePool()` shows that the `PXEBC_PRIVATE_DATA` structure is freed before the return from `ConnectController()` in `efi_driver_connect()`.  However, the freed structure is still installed as the "PxeBcPrivate" protocol instance (using UefiPxeBcDxe's module GUID as the protocol GUID, via `gEfiCallerIdGuid`).

Further debugging shows that an as yet unspecified `EFI_INVALID_PARAMETER` error occurs during installation of the IPv4 child handles.  This results in `PxeBcStart()` exiting via its `ON_ERROR` code path.  This code path calls `UninstallProtocolInterface()` to uninstall the "PxeBcPrivate" protocol instance.  The uninstallation fails with `EFI_ACCESS_DENIED`, but UefiPxeBcDxe ignores this error and frees the containing data structure anyway.

This therefore leaves the `PXEBC_PRIVATE_DATA` structure installed but using freed memory.  Some time later, the HttpBootDxe driver ends up allocating memory to construct its HII query string, which overwrites the `PXEBC_PRIVATE_DATA` structure.  The problem remains hidden until we call `UninstallMultipleProtocolInterfaces()` to uninstall our SNP protocol instance, at which point UefiPxeBcDxe calls through the now garbage pointer, resulting in a general protection fault that terminates the VM.

mcb30, 2023-06-08:
Some further digging through the disassembly and execution paths of VMware's UefiPxeBcDxe show that the call to `PxeBcCreateIp4Children()` succeeds as expected.  The `EFI_INVALID_PARAMETER` error arises from a call to `InstallMultipleProtocolInterfaces()` that attempts to install copies of the IPv4 `EFI_LOAD_FILE_PROTOCOL` and `EFI_PXE_BASE_CODE_PROTOCOL` onto the NIC handle (along with some VMware private protocol with GUID `5190120d-453b-4d48-958d-f0bab3bc2161` and a NULL interface pointer).  This call fails since the NIC handle already has these protocols installed.  (This failure then triggers the broken error handling described in the previous message, which results in a use-after-free bug and a consequent system crash.)

This code that calls `InstallMultipleProtocolInterfaces()` from within `PxeBcStart()` appears to be specific to VMware: it does not appear in any revision of `PxeBcDriver.c` within the whole EDK2 history.

There is some conditional logic visible in the disassembly that could cause the erroneous call to `InstallMultipleProtocolInterfaces()` to be skipped.  However, it is very unclear what conditions are actually being tested.

The fix will almost certainly be to add VMware's UefiPxeBcDxe driver to the veto list of drivers that iPXE will forcibly unload or destroy on entry.

closed 2023-06-08 by mcb30