ipxe/ipxe #885: Connection timed out 4c1b2092

mc3001, 2023-02-08:
I am able to get a DHCP address but can't reach anything after.
[image]

Can't ping anything either

mcb30, 2023-02-08:
Did you try looking at the URL that the error message helpfully provides to you?

https://ipxe.org/4c116092

mc3001, 2023-02-08:
my dns is ok. 
I just can't reach anything

mc3001, 2023-02-08:
[image]

NiKiZe, 2023-02-08:
Can you tcpdump the connection and check if DNS requests are made, and where to.

iPXE is clear on that DNS requests is timing out. Now you need to find why. Incorrect server address, routing or firewall issues, hardware issues, or iPXE software issues.

Do you have any other machine where it works? Have you tried a virtual machine so that you can dump the traffic?

mcb30, 2023-02-08:
> my dns is ok. I just can't reach anything

Your DNS is not OK.  The only possible way to see the error https://ipxe.org/4c116092 is if iPXE fails to get a response from your DNS server.  Please don't assume that you know more than the people trying to help you.

mc3001, 2023-02-10:
I'm just saying that I can't even ping anything around me or even its gateway via ip address, so DNS is not the issue here. As I mentioned I am able to get DHCP and the DNS address is set correctly:
[image]

This issue is specific to these HP Zcentral 4R WS machines. We have other computers that can see the network just fine through ipxe on the same network. I know that this may be a driver issue. We added the drivers in before compiling the snponly.efi

NiKiZe, 2023-02-10:
You might have to really check the traffic here with tcpdump or similar to see what is going on.

Is the requests going out correctly? Are they coming back? What about http requests on IP instead?

mcb30, 2023-02-10:
> I'm just saying that I can't even ping anything around me or even its gateway via ip address, so DNS is not the issue here.

So far you have shown us only a single error message relating to a connection attempt, and that error was a DNS timeout (https://ipxe.org/4c116092).  It's entirely possible that your problem may turn out to be lower in the stack.  But please try actually showing us the problem, instead of just telling us your interpretation of what you think the problem might be.

Try, for example, showing us the errors that you get when trying to ping something by IP address.

The output from the `route` command would also be useful to see.

mc3001, 2023-02-10:
[image]

mc3001, 2023-02-10:
[image]

NiKiZe, 2023-02-10:
What is the IP of the DHCP server? And are you using and DHCP relay agent on switch or similar?

mc3001, 2023-02-10:
the dncp/dns is on the image on the post before where I tried to ping it and it timed out

NiKiZe, 2023-02-10:
That is your gateway, which might be the same, but that is not clear. And you did not answer the questions.

We want to help you, but you need to answer the questions. And stop ignoring half of it.

mc3001, 2023-02-11:
the first part is the dns/dhcp 
[image]

the second one is the gateway 
[image]

there is a dhcp helper address on the router but no dhcp relay agent on a switch

NiKiZe, 2023-02-11:
In that case you are back to tcpdump/wireshark

NiKiZe, 2023-05-28:
I'm closing this as no follow up was provided, feel free to share any tcpdump showing that the DNS requests are coming thru, but not used by iPXE and we are happy to reopen.

closed 2023-05-28 by NiKiZe