ipxe/ipxe #889: Kernel command line parsing in bzimage matches substrings too

filippog, 2023-02-13:
Hello,
I am booting debian-installer via ipxe with `kernel`/`initrd`/`boot` commands in a ipxe script.

While debugging a low memory system I have passed `lowmem=2` to the kernel command line ([list of d-i arguments](https://www.debian.org/releases/bullseye/amd64/ch05s03.en.html#installer-args)), after doing this ipxe fails to load the initrd altogether.

I believe this is due to the fact that [`bzimage_parse_cmdline`](https://github.com/ipxe/ipxe/blob/3c83843e111ece30d3dfb5143d5e6aed6164d587/src/arch/x86/image/bzimage.c#L253) will parse `lowmem=2` as `mem=2` and thus reserve two bytes for initrd memory (!). What do you think ?

stappersg, 2023-02-13:
Text from http://ipxe.org/2a818006 _No buffer space available_

> What do you think?

I think that _lowmem_ is _lowmem_.

Advice: Assign more memory to the test VM or test with machine that has more memory.

Request: Report memory size  for the working and non-working configuration.

mcb30, 2023-02-13:
> I believe this is due to the fact that [`bzimage_parse_cmdline`](https://github.com/ipxe/ipxe/blob/3c83843e111ece30d3dfb5143d5e6aed6164d587/src/arch/x86/image/bzimage.c#L253) will parse `lowmem=2` as `mem=2` and thus reserve two bytes for initrd memory (!). What do you think ?

Yes, this looks very plausible.  `bzimage_parse_cmdline()` does not currently allow for parameters that are substrings of other parameters.  I'll put together a fix.

mcb30, 2023-02-13:
@filippog This is fixed in #892, thanks for reporting.

If you could let me know your email address in the next few hours then I'll add in a credit in the commit log, otherwise I'll merge it tomorrow morning.

filippog, 2023-02-14:
> @filippog This is fixed in #892, thanks for reporting.
> 
> If you could let me know your email address in the next few hours then I'll add in a credit in the commit log, otherwise I'll merge it tomorrow morning.

Thank you for the quick action on this @mcb30 ! Please use `filippo@esaurito.net` for the credit, thank you again!

PR #892 (2023-02-14): [image] Check delimiters when parsing command-line key-value arguments
The Linux kernel bzImage image format and the CPIO archive constructor will parse the image command line for certain arguments of the form "key=value".  This parsing is currently implemented using strstr() in a way that can cause a false positive suffix match.  For example, a command line containing "highmem=<n>" would erroneously be treated as containing a value for "mem=<n>".

Fix by centralising the logic used for parsing such arguments, and including a check that the argument immediately follows a whitespace delimiter ( [... cut]
closed 2023-02-14 by mcb30