ipxe/ipxe #893: implement LoadFile2 initrd loading

yetist, 2023-02-14:
ipxe needs to implement the LoadFile2 protocol in order to load the initrd.

The following software already provides support:
linux: https://lore.kernel.org/linux-efi/20200207202637.GA3464906@rani.riverdale.lan/T/#m4a25eb33112fab7a22faa0fd65d4d663209af32f
grub: https://lists.gnu.org/archive/html/grub-devel/2022-10/msg00095.html
u-boot: https://patchwork.ozlabs.org/project/uboot/patch/20200221075546.864563-1-ilias.apalodimas@linaro.org/

mcb30, 2023-02-14:
iPXE is perfectly capable of loading the initrd via the kernel's `initrd=...` command-line argument, which has been supported for many years.  As per the documentation at https://ipxe.org/cmd/initrd, using `initrd=initrd.magic` is the recommended way to do this since it allows for individually downloaded files to be injected into the constructed initrd image.

mcb30, 2023-02-14:
That said, there would be some value in adding support for the autodetectable `LINUX_EFI_INITRD_MEDIA_GUID`, and associating it with the existing `initrd.magic` file.  This would allow the kernel to pick up the initrd even if the command line was missing `initrd=initrd.magic`.