ipxe/ipxe #896: missing .note.GNU-stack section implies executable stack

stappersg, 2023-02-15:
While compiling on a fairly recent system, I got:
```text
  [VERSION] bin/version.ipxe.dsk.o
  [LD] bin/ipxe.dsk.tmp
ld: warning: patch_cf.o: missing .note.GNU-stack section implies executable stack
ld: NOTE: This behaviour is deprecated and will be removed in a future version of the linker
ld: warning: bin/ipxe.dsk.tmp has a LOAD segment with RWX permissions
ld: warning: bin/ipxe.dsk.tmp has a LOAD segment with RWX permissions
ld: warning: bin/ipxe.dsk.tmp has a LOAD segment with RWX permissions
ld: warning: bin/ipxe.dsk.tmp has a LOAD segment with RWX permissions
```
Reporting it, for having the _NOTE: This behaviour is deprecated and will be removed in a future version of the linker_ documented.

mcb30, 2023-02-15:
The identical issue was recently fixed for ARM32 in commit https://github.com/ipxe/ipxe/commit/fcfb70bfb

You should be able to fix the warnings by adding
```
        .section ".note.GNU-stack", "", @progbits
```
immediately before `.text` in the affected files.

I look forward to the PR. :slightly_smiling_face:

stappersg, 2023-02-15:
On Wed, Feb 15, 2023 at 02:55:15PM -0800, Michael Brown wrote:
> 
> I look forward to the PR. :slightly_smiling_face: 
> 

Result of my first attempt:

     arch/x86/core/patch_cf.S:34: Fatal error: bad .section directive: want a,w,x,M,S,G,T in string

of my second attempt:

     arch/x86/core/patch_cf.S:34: Warning: setting incorrect section attributes for .note.GNU-stack

Unlikely there will be soon another attempt from me        :-/

stappersg, 2023-02-15:
Yeah, it getting late ...
```text
$ git diff
diff --git a/src/arch/x86/core/patch_cf.S b/src/arch/x86/core/patch_cf.S
index 4365563f..6b490a08 100644
--- a/src/arch/x86/core/patch_cf.S
+++ b/src/arch/x86/core/patch_cf.S
@@ -23,6 +23,7 @@
 FILE_LICENCE ( GPL2_OR_LATER_OR_UBDL )
 
        .text
+       .section ".note.GNU-stack", "", @progbits
        .arch i386
        .code16
 
```

mcb30, 2023-02-15:
> Yeah, it getting late ...
> 
> ```
> $ git diff
> diff --git a/src/arch/x86/core/patch_cf.S b/src/arch/x86/core/patch_cf.S
> index 4365563f..6b490a08 100644
> --- a/src/arch/x86/core/patch_cf.S
> +++ b/src/arch/x86/core/patch_cf.S
> @@ -23,6 +23,7 @@
>  FILE_LICENCE ( GPL2_OR_LATER_OR_UBDL )
>  
>         .text
> +       .section ".note.GNU-stack", "", @progbits
>         .arch i386
>         .code16
>  
> ```

Immediately **before** `.text`, but otherwise that should work.

mcb30, 2023-02-15:
This builds successfully for me:

```patch
diff --git a/src/arch/x86/core/patch_cf.S b/src/arch/x86/core/patch_cf.S
index 4365563fe..63730c3fd 100644
--- a/src/arch/x86/core/patch_cf.S
+++ b/src/arch/x86/core/patch_cf.S
@@ -22,6 +22,7 @@
 
 FILE_LICENCE ( GPL2_OR_LATER_OR_UBDL )
 
+       .section ".note.GNU-stack", "", @progbits
        .text
        .arch i386
        .code16
```

stappersg, 2023-02-15:
way beyond bedtime, but there is https://github.com/ipxe/ipxe/pull/897   the

>   Unlikely there will be soon another attempt from me        :-/

was wrong  ...

stappersg, 2023-02-18:
> I look forward to the PR. slightly_smiling_face

There is now #899

stappersg, 2023-02-27:
> There is now merge request #899

Now closing this issue.

closed 2023-02-27 by stappersg